limbo018/DREAMPlace #97: DREAMPlace + OpenLane

RustamC, 2022-11-11:
Hi, @limbo018 !

I’m comparing DREAMPlace and RePlAce at a global placement stage. I've embedded DREAMPlace into the **OpenLane** design flow[^1][^2] and got some experimental results[^3]. DREAMPlace is much faster than RePlAce but inferior in other criteria: final wire length, number of vias, power consumption. 

Could you please advise me if it is possible to change the DREAMPlace's input parameters[^4] to improve the results?

By the way, I've also tried to run legalization step but in all test cases it has failed with the same error:
```
[ASSERT ] /dreamplace/DREAMPlace/dreamplace/ops/abacus_legalize/../abacus_legalize/src/abacus_legalize_cpu.h:201: bool DreamPlace::abacusPlaceRowCPU(const T*, const T*, const T*, const T*, T*, T, T, T, int, int, int*, DreamPlace::AbacusCluster<T>*, int) [with T = float]: Assertion `std::abs(node_size_y[row_nodes[i]] - row_height) < 1e-6' failed
```

[^1]: [How to build OpenLane with DREAMPlace](https://github.com/RustamC/OpenLane/blob/dreamplace_work/HOW_TO_DREAM.md)
[^2]: [DREAMPlace run script](https://github.com/RustamC/OpenLane/blob/dreamplace_work/scripts/openroad/gpld.tcl)
[^3]: [DREAMPlace vs RePlAce comparison](https://github.com/limbo018/DREAMPlace/files/9991488/Github_Compare_DREAM_vs_RePlace.xlsx)
[^4]: [JSON configuration file template for DREAMPlace](https://github.com/RustamC/OpenLane/blob/dreamplace_work/scripts/config/dreamplace_conf.py)

limbo018, 2022-11-12:
Hi, can you provide a testcase that DREAMPlace failed? 

Btw, it looks like you are comparing routed wirelength and other metrics after routing. By default, DREAMPlace does not turn on routability optimization. If you want to make fair comparison, you need to turn on ```routability_opt_flag``` on in JSON and integrate OpenLane router to the placer to provide congestion feedback, which RePlAce has already adapted to.

RustamC, 2022-11-12:
Testcase: [salsa20.tar.gz](https://github.com/limbo018/DREAMPlace/files/9994519/salsa20.tar.gz)
Files in archive:
* 7-global.def - input DEF
* merged.nom.lef - input LEF
* dreamplace.json - DREAMPlace config
* 7-global.log - log

limbo018, 2022-11-13:
Hi, I have obtained the attached placement solution after fixing several bugs. Could you help check what the result is after routing? I was not able to install OpenLane due to unstable connection to Github. 
[7-global.gp.def.gz](https://github.com/limbo018/DREAMPlace/files/9996690/7-global.gp.def.gz)

RustamC, 2022-11-14:
I've checked and added new results into the table: [Github_Compare_DREAM_vs_RePlace.xlsx](https://github.com/limbo018/DREAMPlace/files/10004044/Github_Compare_DREAM_vs_RePlace.xlsx)

In short, the results have improved and are much closer to RePlace's results.

limbo018, 2022-11-15:
Thanks. You may checkout ```develop``` branch for the latest commits (https://github.com/limbo018/DREAMPlace/commit/d2bf2373adbf76271b60d7b72e5468ca3bbe268e and https://github.com/limbo018/DREAMPlace/commit/58e47729c7a341f3c1de1edca9918b06c0aed500). The commits  fix the following major issues: 
1. The LEF file has two site definitions. The parser took the wrong one as the core sites. 
2. Setting large number of bins in JSON could lead to divergence. A heuristic to derive number of bins is added. 

I recommend to choose the following json as the template. 
1. Setting ```num_bins_x``` and ```num_bins_y``` to 0 allows the number of bins derived by internal method. 
2. Initial ```density_weight``` can be between 1e-5 to 8e-5. 
3. ```RePlAce_UPPER_PCOF``` can be between 1.02 to 1.05. Smaller value may provide better wirelength, but more iterations to converge. If you reduce this value, you may at the same time increase ```iteration```. 
4. ```result_dir``` is where the program dumps the output. 
5. You are free to switch ```legalize_flag``` or ```detailed_place_flag``` to enable legalization and detailed placement. They are expected to work on skywater130 now. 

```
{                    
    "def_input": def_input_file,
    "lef_input": lef_input_file,
    "gpu": 0,
    "num_bins_x": 0,                                           
    "num_bins_y": 0,
    "global_place_stages": [
        {
            "iteration": 1000,
            "learning_rate": 0.01,
            "wirelength": "weighted_average",
            "optimizer": "nesterov"
        }
    ],
    "target_density": target_density,
    "density_weight": 8e-5,
    "gamma": 4,
    "random_seed": 1000,
    "ignore_net_degree": 100,
    "enable_fillers": 1,
    "gp_noise_ratio": 0.025,
    "global_place_flag": 1,
    "legalize_flag": 1,
    "detailed_place_flag": 1,
    "detailed_place_engine": "",
    "detailed_place_command": "",
    "stop_overflow": 0.1,
    "result_dir": "results",
    "dtype": "float32",
    "plot_flag": 0,
    "random_center_init_flag": 1,
    "sort_nets_by_degree": 0,
    "num_threads": 4,
    "deterministic_flag": 0,
    "routability_opt_flag": 0,
    "RePlAce_UPPER_PCOF" : 1.05
}
```

I would like to see the updated results for other benchmarks as well. Please let me know if you encounter any other issues. I do not expect obvious differences between DREAMPlace and RePlAce.

RustamC, 2022-11-16:
Thank you very much, @limbo018!

RustamC, 2022-12-12:
Hi, @limbo018 !
I tried your params and got these results: https://docs.google.com/spreadsheets/d/1gi-b7iLdYJDGwBONCR1Lpul8FfGo6NyH41v0S9yj-tA/edit#gid=0
The only difference is that I didn't turn on detailed placement and let RePlAce did it*.

In summary:
1) For circuits with less than 9,500 gates I got an error in "Global Routing Resizer Timing Optimization" step: `Routing congestion too high`
2) For circuits with less than 12,000 gates I got an error right after global placement in "Placement Resizer Design Optimizations" step: `Detailed placement failed`. There was also `aes_core` circuit with ~19,000 gates where I got the same error.

_*Replacing RePlAce in the detailed placement stage is a bit of a non-trivial task as it used in many other stages._

limbo018, 2022-12-12:
1. The detailed placement in RePlAce is actually legalization, so you do not need to re-run it if you enable the legalization DREAMPlace. 
2. Could you send the LEF/DEF files dumped of ```usb, blabla, y_huff, aes_core``` by RePlAce? I can try on DREAMPlace. I doubt there might still be some problems when running these cases.

RustamC, 2022-12-12:
1. For my tests I ran DREAMPlace only in global stage to get a meaningful comparison. I'm not sure if disabling legalization in later stages (like resizer optimizations) will have a positive impact on the final results.
2. Results after RePlace (after global placement): [replace.tar.gz](https://github.com/limbo018/DREAMPlace/files/10207622/replace.tar.gz)

limbo018, 2022-12-12:
1. Only running DREAMPlace in global placement is fine. For your information, if you dump the global placement results to DEF, it converts the floating point positions into integers, which may lose the global positions of cells. If you directly feed the positions to RePlAce in python, then it should be fine. 
2. I will test these cases on my side. Thank you for the files.

limbo018, 2022-12-14:
Hi @RustamC , I have tested the cases on my side and located the problem in parsing the locations of IO pins from DEF. 
The current DEF parser in DREAMPlace only works with following syntax with direct definition of pin locations. 
```
   - bs + NET bs
      + DIRECTION INPUT
      + FIXED ( 118720 55440 ) N
      + LAYER metal3 ( 0 0 ) ( 10 10 ) ;
```
The DEF files of OpenLane cases use ```PORT``` to define pin locations, which causes the placer to set all IO pins at (0, 0). 
```
    - clk + NET clk + DIRECTION INPUT + USE SIGNAL
      + PORT
        + LAYER met3 ( -2000 -300 ) ( 2000 300 )
        + PLACED ( 1198000 962540 ) N ;
```

I have made a hot fix to the problem and generated the results after legalization (I made a guess to the ```target_density``` for each case by roughly matching the HPWL with RePlAce). Could you help check whether they work in the OpenLane flow? 

[openlane_cases_dreamplace.tar.gz](https://github.com/limbo018/DREAMPlace/files/10223604/openlane_cases_dreamplace.tar.gz)

I still need further regression tests to make sure the fix does not affect other cases before pushing the commits.

RustamC, 2022-12-15:
Hi, @limbo018! I've checked the [results](https://docs.google.com/spreadsheets/d/1gi-b7iLdYJDGwBONCR1Lpul8FfGo6NyH41v0S9yj-tA/edit#gid=0) and they worked in the OpenLane flow.

limbo018, 2022-12-15:
Great! You can pull and checkout the commit (https://github.com/limbo018/DREAMPlace/commit/f3beae432e72403e2700ec1c0c85302430163e2c) for the latest version with the IO pin parsing problem fixed.

RustamC, 2022-12-20:
Hi, @limbo018 ! I've tried last `develop` version and updated the [results](https://docs.google.com/spreadsheets/d/1gi-b7iLdYJDGwBONCR1Lpul8FfGo6NyH41v0S9yj-tA/edit#gid=0) (last 3 tables). Everything now works fine.

However, I've also tried to place simple [inverter example](https://github.com/limbo018/DREAMPlace/files/10269096/inverter.zip) containing just 1 inverter and got an error:
```
[ERROR]: during executing openroad script /openlane/scripts/openroad/gpld.tcl
[ERROR]: Log: designs/inverter/runs/dream/logs/placement/7-global.log
[ERROR]: Last 10 lines:
  File "/build/dreamplace/Placer.py", line 39, in place
    placedb(params)
  File "/build/dreamplace/PlaceDB.py", line 625, in __call__
    self.initialize(params)
  File "/build/dreamplace/PlaceDB.py", line 873, in initialize
    self.num_filler_nodes = int(round(self.total_filler_node_area / (filler_size_x * filler_size_y)))
ValueError: cannot convert float NaN to integer
```

limbo018, 2022-12-21:
Hi @RustamC , can you also provide the LEF/DEF files for the inverter design? Probably there are some corner cases not considered.

RustamC, 2022-12-21:
Sorry, forgot about the LEF file: [inverter.zip](https://github.com/limbo018/DREAMPlace/files/10275241/inverter.zip)

limbo018, 2022-12-22:
Hi @RustamC , the commit (https://github.com/limbo018/DREAMPlace/commit/9a280f182279b1806e559325052abddd81549df1) in the ```develop``` branch has fixed the bug when computing the filler stats with fewer than 2 movable cells.

limbo018, 2022-12-22:
Btw, I have to mention that when there are too few movable cells (e.g., < 100), the electrostatic algorithm in global placement may not achieve good solution quality, since the algorithm assumes the statistical distribution of cells is representative. Placers based on the electrostatic algorithm usually have the same behavior, including RePlAce.

mumu157, 2023-06-20:
Hi @RustamC , I'm currently working on something similar. I understand that you have already integrated DREAMPlace into OpenLane, and I have a few questions regarding the file formats involved. In the OpenLane flow, I know that the data flow utilizes the .odb format. However, to utilize DREAMPlace, I believe I need to work with the .def/.lef format instead. Could you please confirm if my understanding is correct? Specifically, I would like to know if it is necessary to manually replace the .odb format with the .def/.lef format for compatibility with DREAMPlace. Thank you for your assistance!

RustamC, 2023-06-20:
Hi, @mumu157 ! Your understanding is correct. When I recently tried to integrate DREAMPlace into a newer version of OpenLane, I faced the same problem. Right now, from my point of view, the easiest & the fastest way to integrate DREAMPlace is to create a script for the OpenLane interactive flow (using [-file](https://github.com/The-OpenROAD-Project/OpenLane/discussions/1256#discussion-4288757) option), replacing `run_floorplan` & `run_placement` commands. 
In the script you will have to manually write LEF/DEF files, run DREAMPlace and load the resulting DEF file back. 
The final Tcl script will look like this:
```tcl
prep -design <design> -tag <tag> -config <config> -overwrite
run_synthesis
% here you run DREAMPlace, instead of run_floorplan, run_placement
run_cts
...
```

mumu157, 2023-06-21:
Really thanks @RustamC ,  I will try.