limbo018/DREAMPlace #104: Weird Block Orientation

TAKE72K, 2023-01-09:
I modify ispd2015benchmark to see how rotation of block affects std cell placement, but something weird happened.
mgc_matrix_mult_a for example,
[image]

I modify orientations of blocks as following:
```
  - h0 h0
    + FIXED ( 50655 10370 ) W ;
  - h1 h1
    + FIXED ( 55655 488000 ) E ;
  - h2 h2
    + FIXED ( 53455 1078000 ) E ;
  - h3 h3
    + FIXED ( 950020 312000 ) W ;
  - h4 h4
    + FIXED ( 938925 1136000 ) S ;
```
But the result is identical to original def:
[image]

Dose Dreamplace currently support block orientation feature in DEF/LEF?

limbo018, 2023-01-09:
Hi, thank you for reporting the problem. Could you provide the LEF/DEF files for us to debug? Probably there is a bug when processing the orientations.

TAKE72K, 2023-01-10:
Thanks for your reply.
The attachment contains modified def/lef from ispd2015, also the config file I used to run Dreamplace, and the result it produced. Seems like it still considers the blocks on the original positions.
[mgc_matrix_mult_a_rotate.zip](https://github.com/limbo018/DREAMPlace/files/10378485/mgc_matrix_mult_a_rotate.zip)

[image]

limbo018, 2023-01-11:
Hi, the orientation support has been added in the ```develop``` branch (commit https://github.com/limbo018/DREAMPlace/commit/c34c3497ea733b2dba7fcffd4cac5ac85b7cb7e3). Please pull the latest code in the ```develop``` branch and check if it works. 

The result I got from your testcase is as following. 

[image]

There are some overlaps between fixed macros and some macros go out of the boundary. 

Please let me know if there are any further issues. I will merge the code to the ```master``` branch if it works fine.

TAKE72K, 2023-01-11:
Hi,
thank you so much!
I have tested it, and it works fine.
This issue can be closed as completed.

closed 2023-01-17 by TAKE72K