limbo018/DREAMPlace #105: Met error when I run " python dreamplace/Placer.py test/ispd2005/adaptec1.json"

ybiao-he, 2023-01-18:
When I run ` python dreamplace/Placer.py test/ispd2005/adaptec1.json`, it shows 

[ASSERT ] /home/test/DREAMPlace/dreamplace/ops/independent_set_matching/../independent_set_matching/src/shuffle.cuh:50: DreamPlace::Shuffler<T, V>::Shuffler(size_t, T*, int) [with T = int; V = unsigned int; size_t = long unsigned int]: Assertion `curandCreateGenerator(&m_gen, CURAND_RNG_PSEUDO_DEFAULT) == CURAND_STATUS_SUCCESS' failed
Aborted (core dumped)

I don't have any idea why this happens. Before this errors, everything seems good, here is the screenshot of partial running log before this errors appear.

 
[image]

limbo018, 2023-01-18:
Hi, what's your GPU and what's your CUDA version?

ybiao-he, 2023-01-18:
My GPU is NVIDIA GeForce RTX 3090 and cuda version is 11.6

limbo018, 2023-01-19:
Then, hardware should not be a problem. I'm not sure whether the problem comes from CUDA version, as I cannot reproduce on my side. I'd like to make sure if other parts work fine. Could you comment out line 980-985 in ```BasicPlace.py``` (https://github.com/limbo018/DREAMPlace/blob/5ecb1dc0059f2457100d21a9d73af1ca5f497e68/dreamplace/BasicPlace.py#L980), re-install, and re-run the case?

If everything else works, then we can conclude the problem comes from the curand package.

ybiao-he, 2023-01-19:
Thanks for your suggestion. Dr. Lin. Yes, no errors when I comment out those lines.

limbo018, 2023-01-19:
Thank you! One more thing to confirm. Do you make sure to have GPU memory available when running the program? You may take a look by ```nvidia-smi``` when running the problem. In general, 2-3GB is enough for ```adapec1```.

limbo018, 2023-01-19:
> When I run ` python dreamplace/Placer.py test/ispd2005/adaptec1.json`, it shows
>
> [ASSERT ] [...]

Btw, did you modify the benchmark or the code? I took a detailed look at your run. The HPWL is much worse than the run on my side after global placement. At the end of global placement, the HPWL on my side is 7.0253E+07, but your screenshot shows 7.919E+07. Just be curious on what happened. 
```
[INFO   ] DREAMPlace - iteration  609, ( 609,  0,  0), Obj 6.720478E+07, DensityWeight 5.377610E-02, HPWL 7.024102E+07, Overflow 7.246860E-02, MaxDensity 1.868E+00, gamma 1.462454E+01, time 0.469ms                                                               
[INFO   ] DREAMPlace - full step 4.130 ms                                                                                         
[INFO   ] DREAMPlace - optimizer step 3.177 ms                                                                                    
[INFO   ] DREAMPlace - iteration  610, ( 610,  0,  0), Obj 6.723882E+07, DensityWeight 5.641763E-02, HPWL 7.024737E+07, Overflow 7.089222E-02, MaxDensity 1.854E+00, gamma 1.450288E+01, time 0.483ms                                                               
[INFO   ] DREAMPlace - full step 4.227 ms                                                                                         
[INFO   ] DREAMPlace - optimizer step 3.096 ms                                                                                    
[INFO   ] DREAMPlace - iteration  611, ( 611,  0,  0), Obj 6.727194E+07, DensityWeight 5.918607E-02, HPWL 7.025366E+07, Overflow 6.941244E-02, MaxDensity 1.857E+00, gamma 1.438637E+01, time 0.480ms
```

ybiao-he, 2023-01-19:
No, I did not modify the code or benchmark.
I just re-run it, the HPWL value is  7.027909E+07 for adaptec1.json. 

I also checked the screenshot result, it should be for the adaptec2 case. Sorry for making it confusing.

3258859094, 2023-03-19:
Prof.Lin,I met another error when I run "python dreamplace/Placer.py test/ispd2005/adaptec1.json".
[image]
Please tell me what's wrong,I searched it and find no solution,my cuda is available actually.
my GPU is NVIDIA GTX750 and CUDA version is 11.7

limbo018, 2023-03-19:
You need to check your CMake log to see whether it finds CUDA correctly.