limbo018/DREAMPlace #106: Meet problems when building

b224hisl, 2023-02-04:
I use the docker container by executing `docker pull limbo018/dreamplace:cuda`, but I met some problems when I'm building it in the root directory. here is the error log:
`[ 94%] Building CXX object dreamplace/ops/place_io/CMakeFiles/place_io_cpp.dir/src/PlaceDB.cpp.o
cd /DREAMPlace/build/dreamplace/ops/place_io && /usr/bin/c++ -DDRAWPLACE=1 -DENABLE_CUDA=0 -DTORCH_EXTENSION_NAME=place_io_cpp -DTORCH_VERSION_MAJOR=1 -DTORCH_VERSION_MINOR=7 -D_GLIBCXX_USE_CXX11_ABI=0 -Dplace_io_cpp_EXPORTS -I/DREAMPlace/dreamplace/ops/place_io/.. -I/DREAMPlace/thirdparty/Limbo -I/usr/include/cairo -isystem /DREAMPlace/thirdparty/pybind11/include -isystem /opt/conda/include/python3.8 -isystem /opt/conda/lib/python3.8/site-packages/torch/include -isystem /opt/conda/lib/python3.8/site-packages/torch/include/torch/csrc/api/include -O3 -DNDEBUG -fPIC -fvisibility=hidden -flto -fno-fat-lto-objects -D_GLIBCXX_USE_CXX11_ABI=0 -fopenmp -std=gnu++14 -MD -MT dreamplace/ops/place_io/CMakeFiles/place_io_cpp.dir/src/PlaceDB.cpp.o -MF CMakeFiles/place_io_cpp.dir/src/PlaceDB.cpp.o.d -o CMakeFiles/place_io_cpp.dir/src/PlaceDB.cpp.o -c /DREAMPlace/dreamplace/ops/place_io/src/PlaceDB.cpp
/DREAMPlace/dreamplace/ops/place_io/src/PlaceDB.cpp: In member function 'virtual void DreamPlace::PlaceDB::add_def_pin(const DefParser::Pin&)':
/DREAMPlace/dreamplace/ops/place_io/src/PlaceDB.cpp:325:17: error: 'const struct DefParser::Pin' has no member named 'vPinPort'
   } else if (!p.vPinPort.empty()) {
                 ^~~~~~~~
/DREAMPlace/dreamplace/ops/place_io/src/PlaceDB.cpp:327:35: error: 'const struct DefParser::Pin' has no member named 'vPinPort'
     for (std::size_t i = 0; i < p.vPinPort.size(); ++i) {
                                   ^~~~~~~~
/DREAMPlace/dreamplace/ops/place_io/src/PlaceDB.cpp:329:14: error: 'const struct DefParser::Pin' has no member named 'vPinPort'
       if (!p.vPinPort[i].vLayer.empty()) {
              ^~~~~~~~
/DREAMPlace/dreamplace/ops/place_io/src/PlaceDB.cpp:360:17: error: 'const struct DefParser::Pin' has no member named 'vPinPort'
   } else if (!p.vPinPort.empty()) {
                 ^~~~~~~~
/DREAMPlace/dreamplace/ops/place_io/src/PlaceDB.cpp:361:22: error: 'const struct DefParser::Pin' has no member named 'vPinPort'
     node.setOrient(p.vPinPort.front().orient);
                      ^~~~~~~~
/DREAMPlace/dreamplace/ops/place_io/src/PlaceDB.cpp:390:19: error: 'const struct DefParser::Pin' has no member named 'vPinPort'
     } else if (!p.vPinPort.empty()) {
                   ^~~~~~~~
/DREAMPlace/dreamplace/ops/place_io/src/PlaceDB.cpp:391:37: error: 'const struct DefParser::Pin' has no member named 'vPinPort'
       for (std::size_t i = 0; i < p.vPinPort.size(); ++i) {
                                     ^~~~~~~~
/DREAMPlace/dreamplace/ops/place_io/src/PlaceDB.cpp:392:20: error: 'PinPort' is not a member of 'DefParser'
         DefParser::PinPort const& pport = p.vPinPort[i];
                    ^~~~~~~
/DREAMPlace/dreamplace/ops/place_io/src/PlaceDB.cpp:393:37: error: 'pport' was not declared in this scope
         for (std::size_t j = 0; j < pport.vLayer.size(); ++j) {
                                     ^~~~~
/DREAMPlace/dreamplace/ops/place_io/src/PlaceDB.cpp:393:37: note: suggested alternative: 'qsort'
         for (std::size_t j = 0; j < pport.vLayer.size(); ++j) {
                                     ^~~~~
                                     qsort
/DREAMPlace/dreamplace/ops/place_io/src/PlaceDB.cpp:486:17: error: 'const struct DefParser::Pin' has no member named 'vPinPort'
   } else if (!p.vPinPort.empty()) {
                 ^~~~~~~~
/DREAMPlace/dreamplace/ops/place_io/src/PlaceDB.cpp:487:35: error: 'const struct DefParser::Pin' has no member named 'vPinPort'
     for (std::size_t i = 0; i < p.vPinPort.size(); ++i) {
                                   ^~~~~~~~
/DREAMPlace/dreamplace/ops/place_io/src/PlaceDB.cpp:491:18: error: 'PinPort' is not a member of 'DefParser'
       DefParser::PinPort const& pport = p.vPinPort[i];
                  ^~~~~~~
/DREAMPlace/dreamplace/ops/place_io/src/PlaceDB.cpp:492:35: error: 'pport' was not declared in this scope
       for (std::size_t j = 0; j < pport.vLayer.size(); ++j) {
                                   ^~~~~
/DREAMPlace/dreamplace/ops/place_io/src/PlaceDB.cpp:492:35: note: suggested alternative: 'qsort'
       for (std::size_t j = 0; j < pport.vLayer.size(); ++j) {
                                   ^~~~~
                                   qsort
dreamplace/ops/place_io/CMakeFiles/place_io_cpp.dir/build.make:190: recipe for target 'dreamplace/ops/place_io/CMakeFiles/place_io_cpp.dir/src/PlaceDB.cpp.o' failed
make[2]: *** [dreamplace/ops/place_io/CMakeFiles/place_io_cpp.dir/src/PlaceDB.cpp.o] Error 1
make[2]: Leaving directory '/DREAMPlace/build'
CMakeFiles/Makefile2:2785: recipe for target 'dreamplace/ops/place_io/CMakeFiles/place_io_cpp.dir/all' failed
make[1]: *** [dreamplace/ops/place_io/CMakeFiles/place_io_cpp.dir/all] Error 2
make[1]: Leaving directory '/DREAMPlace/build'
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
`
Could anyone helps with that? thanks a lot!

TAKE72K, 2023-02-07:
I met the same error, set thirdparty/Limbo to check out its master solved this in my case.

limbo018, 2023-02-07:
@b224hisl Did you run ```git submodule update``` after ```git pull```? Probably the submodules in the thirdparty folder have not been updated to the latest version. You may check if there are any out-dated contents with ```git status```.