limbo018/DREAMPlace #109: HPWL in DREAMPlace isn't as same as the HPWL in NTUPlace3

BWbwchen, 2023-03-02:
Hi Prof. @limbo018,

I ran the adaptec1 with NTUPlace3 for legalization and detailed placement, and I found a weird mismatch situation in HPWL for DREAMPlace and NTUPlace3.

I ran the adaptec1 two times and pasted the last iteration info below. This is what I got:

### First time(With GPU):
>  [INFO   ] DREAMPlace - iteration  606, ( 606,  0,  0), Obj 6.729493E+07, DensityWeight 5.879883E-02, HPWL 7.028460E+07, Overflow 6.993332E-02, MaxDensity 1.770E+00, gamma 1.442712E+01, time 3.145ms

and The  DATABASE SUMMARIES of NTUPlace3:
>  Pin-to-Pin HPWL: 70700034 (7.07e+07)

### Second time(Without GPU):
>  [INFO   ] DREAMPlace - iteration  607, ( 607,  0,  0), Obj 6.732603E+07, DensityWeight 6.167179E-02, HPWL 7.029075E+07, Overflow 6.856339E-02, MaxDensity 1.789E+00, gamma 1.432333E+01, time 11.774ms

and The  DATABASE SUMMARIES of NTUPlace3:
>  Pin-to-Pin HPWL: 70705390 (7.07054e+07)

### Why
We can notice that both of the cases have a mismatch in the HPWL result, but I don't know why.

Could you help me with this weird situation? Thanks for your help!

limbo018, 2023-03-03:
DREAMPlace ignores nets whose pins all belong to the same cells, so the HPWL reported is smaller than NTUplace3. Usually we use NTUplace3 to report the HPWL in the papers for consistency.

BWbwchen, 2023-03-03:
Hi Prof. @limbo018,

Thanks for your help. Your answer helps me a lot.

I am wondering where can I find the ignored nets code?
I didn't find any relative code in place_io/ directory, but only find some code about calculating the number of ignored nets:
https://github.com/limbo018/DREAMPlace/blob/b31e8afa60775bfefdcbb52c0926c6167ec394f6/dreamplace/ops/place_io/src/PlaceDB.cpp#L919-L950

It seems that the result of that code(`m_vNetIgnoreFlag`) didn't export to the python...
So I am wondering where can I find the ignored nets code?

Thanks for your help!!

limbo018, 2023-03-03:
Here is the related code block where only one pin is kept if a net has multiple pins belonging to the same node. 

https://github.com/limbo018/DREAMPlace/blob/b31e8afa60775bfefdcbb52c0926c6167ec394f6/dreamplace/ops/place_io/src/PlaceDB.cpp#L904

Thanks
Yibo

BWbwchen, 2023-03-04:
Thanks for your help. 
I understand its mechanism now. It helps me a lot!!

closed 2023-03-04 by BWbwchen