limbo018/DREAMPlace #113: when I make after cmake,have 92 errors

gjhhust, 2023-03-17:
[image]
according to the image,please hlep me,thanks

gjhhust, 2023-03-17:
> Can you provide more information about your system, pytorch version, cuda version, and gcc version? The problem usually comes from incompatible compiling environments.

Thank your reply！！This problem does not exist on my new device, but there are new problems: My new System: Ubuntu 18, Torch 1.17.0, CUDA: 10.2, GCC: 7.5.0,GPU Capacity：6.1  。and this problem has been fixed after I changed the machine, when i run“python dreamplace/Placer.py test/ispd2005/adaptec1.json”，the new error is

And I looked up the issue and found two similar problems, one is not running in the root directory, and one is remake, but I tried and failed, what do you suggest

limbo018, 2023-03-17:
It looks like you have solved the compiling problem. When running the program, you should run from the **install** direction, not the root direction, as mentioned in the readme.

limbo018, 2023-03-17:
What problem? Can you upload the screenshot? I don't see any figures in your reply.

gjhhust, 2023-03-17:
Oh!I'm sorry, I found that just failed to upload，here：
[image]

gjhhust, 2023-03-17:
yes, they are all environment set up under conda with Python 3.8.3.

limbo018, 2023-03-17:
The problem probably comes from the linking with PyTorch, but I don't have much clue with the current information if you make sure that you compiled and ran the program under the same environment. I assume your PyTorch version is 1.7, as no version 1.17 has ever been released. 

I can share my environment for testing. 
```
Ubuntu 18.04
Python 3.8.10
gcc 7.5.0
PyTorch 1.7.1
CUDA 11.5
```
Your current problem comes from CPU version, so it has nothing to do with CUDA yet.

limbo018, 2023-03-17:
Can you try only using the following command for building?
```
rm -r * # remove everything in build
cmake .. -DCMAKE_INSTALL_PREFIX=../../dreamplace_
```

gjhhust, 2023-03-17:
The good news is that link succeeded, but unfortunately the data was out of bounds when i run:
[image]
i try debug code：
[image]

limbo018, 2023-03-17:
Hi, 

You may try two directions: 
1. Switch to PyTorch 1.7, as it has been tested already 
2. Switch to the develop branch where I have merged a pull request which claims to fix the problem with PyTorch 2.0.

gjhhust, 2023-03-18:
Thank you! All the mistakes have been solved, thank you very much for your help, I wish you good health, Happy Life ~

closed 2023-03-18 by gjhhust