limbo018/DREAMPlace #115: ValueError: setting an array element with a sequence. The requested array has an inhomogeneous shape after 1 dimensions. The detected shape was (211447,) + inhomogeneous part.

manili, 2023-03-18:
Hi,
I'm not able to pass the following error:

```

========================================================
                       DREAMPlace
            Yibo Lin (http://yibolin.com)
   David Z. Pan (http://users.ece.utexas.edu/~dpan)
========================================================
[INFO   ] DREAMPlace - parameters = {'aux_input': 'benchmarks/ispd2005/adaptec1/adaptec1.aux', 'lef_input': '', 'def_input': '', 'verilog_input': '', 'gpu': 0, 'num_bins_x': 512, 'num_bins_y': 512, 'global_place_stages': [{'num_bins_x': 512, 'num_bins_y': 512, 'iteration': 1000, 'learning_rate': 0.01, 'wirelength': 'weighted_average', 'optimizer': 'nesterov', 'Llambda_density_weight_iteration': 1, 'Lsub_iteration': 1}], 'target_density': 1.0, 'density_weight': 8e-05, 'random_seed': 1000, 'result_dir': 'results', 'scale_factor': 1.0, 'shift_factor': [0.0, 0.0], 'ignore_net_degree': 100, 'gp_noise_ratio': 0.025, 'enable_fillers': 1, 'global_place_flag': 1, 'legalize_flag': 1, 'detailed_place_flag': 1, 'stop_overflow': 0.07, 'dtype': 'float32', 'detailed_place_engine': '', 'detailed_place_command': '', 'plot_flag': 0, 'RePlAce_ref_hpwl': 350000, 'RePlAce_LOWER_PCOF': 0.95, 'RePlAce_UPPER_PCOF': 1.05, 'gamma': 4.0, 'RePlAce_skip_energy_flag': 0, 'random_center_init_flag': 1, 'sort_nets_by_degree': 0, 'num_threads': 8, 'dump_global_place_solution_flag': 0, 'dump_legalize_solution_flag': 0, 'routability_opt_flag': 0, 'route_num_bins_x': 512, 'route_num_bins_y': 512, 'node_area_adjust_overflow': 0.15, 'max_num_area_adjust': 3, 'adjust_nctugr_area_flag': 0, 'adjust_rudy_area_flag': 1, 'adjust_pin_area_flag': 1, 'area_adjust_stop_ratio': 0.01, 'route_area_adjust_stop_ratio': 0.01, 'pin_area_adjust_stop_ratio': 0.05, 'unit_horizontal_capacity': 1.5625, 'unit_vertical_capacity': 1.45, 'unit_pin_capacity': 0.058, 'max_route_opt_adjust_rate': 2.0, 'route_opt_adjust_exponent': 2.0, 'pin_stretch_ratio': 1.414213562, 'max_pin_opt_adjust_rate': 1.5, 'deterministic_flag': 0}
[WARNING] DEF input file not specified, cannot output DEF file; set to DEFSIMPLE
[INFO   ] lef_input = 
[INFO   ] def_input = 
[INFO   ] verilog_input = 
[INFO   ] bookshelf_aux_input = benchmarks/ispd2005/adaptec1/adaptec1.aux
[INFO   ] bookshelf_pl_input = 
[INFO   ] def_size_input = 
[INFO   ] def_output = -out.def
[INFO   ] rpt_output = 
[INFO   ] target_util = 0
[INFO   ] max_displace = 0
[INFO   ] bin size = (10, 10) #rows
[INFO   ] sbin size = (5, 5) #rows
[INFO   ] bin_space_threshold = 0.2
[INFO   ] abu = 2,10 5,5 10,2 20,1 
[INFO   ] def_ignore_cells = 
[INFO   ] macro_obs_aware_layers = 
[INFO   ] enable_place = true
[INFO   ] enable_legalize = true
[INFO   ] evaluate_overlap = false
[INFO   ] move_multi_row_cell = true
[INFO   ] align_power_line = true
[INFO   ] cluster_cell = false
[INFO   ] sort_nets_by_degree = false
[INFO   ] file_format = BOOKSHELF
[INFO   ] max_iters = 6
[WARNING] no DEF file specified
[WARNING] no Verilog file specified
[INFO   ] reading benchmarks/ispd2005/adaptec1/adaptec1.aux
[WARNING] no additional Bookshelf .pl file specified
[WARNING] 20285 nets with 24352 pins from same nodes
[WARNING] 4210 nets should be ignored due to not enough pins
[INFO   ] sort nodes in the order of movable and fixed
[INFO   ] Group cells for fence regions
[INFO   ] Construct 0 groups
[INFO   ] Fence region groups done
[DEBUG  ] num_terminals 543, numFixed 543, numPlaceBlockages 0, num_terminal_NIs 0
[INFO   ] -- Converting Node Orientation --
[INFO   ] ---------------------------------
[INFO   ] -------- Area Statistics --------
[INFO   ] fixed area overlap = 4.91641e+07
[INFO   ] fixed area total = 6.4094e+07
[INFO   ] space area = 6.50265e+07
[INFO   ] ---------------------------------
Traceback (most recent call last):
  File "dreamplace/Placer.py", line 174, in <module>
    place(params)
  File "dreamplace/Placer.py", line 39, in place
    placedb(params)
  File "/home/manili/Desktop/DREAMPlace/build/install/dreamplace/PlaceDB.py", line 624, in __call__
    self.read(params)
  File "/home/manili/Desktop/DREAMPlace/build/install/dreamplace/PlaceDB.py", line 469, in read
    self.initialize_from_rawdb(params)
  File "/home/manili/Desktop/DREAMPlace/build/install/dreamplace/PlaceDB.py", line 571, in initialize_from_rawdb
    self.node2pin_map = np.array(self.node2pin_map)
ValueError: setting an array element with a sequence. The requested array has an inhomogeneous shape after 1 dimensions. The detected shape was (211447,) + inhomogeneous part.

```

I'm trying to compile the code from scratch on my local (so not using the Docker image) and tried every possible configurations. After running code on my CPU I always get the same error. Any ideas?

limbo018, 2023-03-19:
It looks like a numpy error. What's your numpy version?

manili, 2023-03-19:
Hi @limbo018 
I tried two different numpy versions:
- 1.15.4
- 1.24.2 (latest)

limbo018, 2023-03-19:
That's weird. The numpy version on my side is 1.23. Can you try ```print(self.node2pin_map)``` before line 571 of PlaceDB.py and see what you get? You can directly insert code there. It's python.

manili, 2023-03-19:
Hi
Thanks a lot for the quick response. I'm currently trying `develop` branch. I'll come back to you ASAP.
P.S. sorry didn't get what you mean by:

> You can directly insert code there. It's python.

manili, 2023-03-19:
Hi @limbo018 ,
I did what you suggested. I'll attach the output. Please note that here is the command I'm using:
[output.txt](https://github.com/limbo018/DREAMPlace/files/11010621/output.txt)

```
python3.10 dreamplace/Placer.py test/ispd2005/adaptec1_cpu.json
```

manili, 2023-03-19:
@limbo018 
It's really odd that after commenting L571 and L576 of `PlaceDB.py` everything's working fine! How can I verify the correctness of my result?

limbo018, 2023-03-19:
> 

You are right. Actually ```node2pin_map``` and ```net2pin_map``` in ```PlaceDB.py``` are only used for debugging. They are not used elsewhere. So the placement results should be correct. 

But I'm still curious why converting python list to numpy array causes error. At least the conversion works fine on python 3.7 and 3.8.

limbo018, 2023-03-19:
> Hi @limbo018 , Thanks a lot for the quick response. I did what you suggested and here is the [output](https://github.com/limbo018/DREAMPlace/files/11010621/output.txt). Please note that I'm currently using `develop` branch and here is the command I'm trying to run:
> 
> ```
> python3.10 dreamplace/Placer.py test/ispd2005/adaptec1_cpu.json
> ```

Can you try running the following code in your python 3.10 to see whether any error occurs? 

```
import numpy as np 

test_list = [[1], [2, 3], [4, 5, 6]]
for i in range(len(test_list)):
    test_list[i] = np.array(test_list[i], dtype=np.int32)

test_list2 = np.array(test_list) # if this line complains error, can you try next line?
test_list3 = np.array(test_list, dtype=object) 
```

Can you give feedbacks on whether ```test_list2``` and ```test_list3``` result in any errors?

manili, 2023-03-19:
Hi @limbo018 
Thanks a lot for the reply. As you might guess:

1. test_list2 got the same error -> ValueError: setting an array element with a sequence. The requested array has an inhomogeneous shape after 1 dimensions. The detected shape was (3,) + inhomogeneous part.
2. test_list3 just works fine.

I received exactly the same result for both python3.10 and python3.8

limbo018, 2023-03-19:
This experiment figures the problem out. You can change line 571 and 576 in ```PlaceDB.py``` to following, respectively. 
```
self.node2pin_map = np.array(self.node2pin_map, dtype=object)
self.net2pin_map = np.array(self.net2pin_map, dtype=object)
```

On my side, python 3.8 only gives deprecated warning about these lines, so I did not fix the issue earlier.

manili, 2023-03-19:
Which subversion of 3.8 are you using? I'm on 3.8.15

limbo018, 2023-03-19:
I'm on Python 3.8.10

manili, 2023-03-19:
Got it!
Thanks a lot for the support.

closed 2023-03-19 by manili