limbo018/DREAMPlace #119: Met problems when run unit test

Yixin-Gong, 2023-03-28:
**The problem follows:**
```text
eric@eric-System-Product-Name:~/Placer/DREAMPlace/install$ python3 unittest/ops/hpwl_unittest.py
E
======================================================================
ERROR: test_hpwlRandom (__main__.HPWLOpTest)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/eric/Placer/DREAMPlace/install/unittest/ops/hpwl_unittest.py", line 41, in test_hpwlRandom
    net2pin_map = np.array([np.array([0, 4]), np.array([1, 2, 3])])
ValueError: setting an array element with a sequence. The requested array has an inhomogeneous shape after 1 dimensions. The detected shape was (2,) + inhomogeneous part.

----------------------------------------------------------------------
Ran 1 test in 0.000s

FAILED (errors=1)
```
   Prof.Lin,I can run the program now, howerver, when I run torch unittest, I met this problem. I think it's because of my *numpy version*, I met the familiar problem when I run the program, and learned how to fix it from another closed issue.
Please tell me how to edit python file to cope with this one.
Must I checkout to develop branch or change my numpy version?

Yixin-Gong, 2023-03-28:
sorry to bother u,I fix it now, Change line 41 in DREAMPlace/install/unittest/ops/hpwl_unittest.py" to
```python3
net2pin_map = np.array([np.array([0, 4]), np.array([1, 2, 3])],dtype=object)
```
can fix it.

closed 2023-03-28 by Yixin-Gong

limbo018, 2023-03-28:
> sorry to bother u,I fix it now, Change line 41 in DREAMPlace/install/unittest/ops/hpwl_unittest.py" to
> 
> ```python
> net2pin_map = np.array([np.array([0, 4]), np.array([1, 2, 3])],dtype=object)
> ```
> 
> can fix it.

It would be great if you can create a pull request, so we can merge it in the next release.

Yixin-Gong, 2023-03-29:
> It would be great if you can create a pull request, so we can merge it in the next release.

Thanks for your support, Prof.Lin. I create a pull request.Thanks♪(･ω･)ﾉ