limbo018/DREAMPlace #121: Error of [dispatch_histogram.cuh] when making

wangzhen0518, 2023-04-01:
System Info
```
OS: wsl ubuntu22.04
cuda: cuda_12.1.r12.1/compiler.32415258_0
branch: develop
```

I met errors when making.

```bash
/usr/local/bin/cmake -E rm -f /home/wangzhen/Project/DREAMPlace/build/dreamplace/ops/density_map/CMakeFiles/density_map_cuda_cuda_tmp.dir/src/./density_map_cuda_cuda_tmp_generated_density_map_cuda_kernel.cu.o
-- Generating dependency file: /home/wangzhen/Project/DREAMPlace/build/dreamplace/ops/density_map/CMakeFiles/density_map_cuda_cuda_tmp.dir/src/density_map_cuda_cuda_tmp_generated_density_map_cuda_kernel.cu.o.NVCC-depend
/usr/local/cuda/bin/nvcc -M -D__CUDACC__ /home/wangzhen/Project/DREAMPlace/dreamplace/ops/density_map/src/density_map_cuda_kernel.cu -o /home/wangzhen/Project/DREAMPlace/build/dreamplace/ops/density_map/CMakeFiles/density_map_cuda_cuda_tmp.dir/src/density_map_cuda_cuda_tmp_generated_density_map_cuda_kernel.cu.o.NVCC-depend -m64 -DTORCH_EXTENSION_NAME=density_map_cuda -DTORCH_MAJOR_VERSION= -DTORCH_MINOR_VERSION= -DENABLE_CUDA=1 -D_GLIBCXX_USE_CXX11_ABI=0 -Xcompiler ,\"-g\" -gencode arch=compute_60,code=sm_60 -gencode arch=compute_61,code=sm_61 -gencode arch=compute_70,code=sm_70 -gencode arch=compute_75,code=sm_75 -gencode arch=compute_80,code=sm_80 -gencode arch=compute_86,code=sm_86 --compiler-options -fPIC -DNVCC -I/usr/local/cuda/include -I/home/wangzhen/Project/DREAMPlace/dreamplace/ops/density_map/.. -I/home/wangzhen/miniconda3/envs/ml/lib/python3.10/site-packages/torch/include -I/home/wangzhen/miniconda3/envs/ml/lib/python3.10/site-packages/torch/include/torch/csrc/api/include -I/home/wangzhen/Project/DREAMPlace/thirdparty/pybind11/include -I/home/wangzhen/miniconda3/envs/ml/include/python3.10
/usr/local/cuda/include/cub/device/dispatch/dispatch_histogram.cuh(246): error: name followed by "::" must be a class or namespace name
        using CommonT = typename cuda::std::common_type<LevelT, SampleT>::type;
                                       ^
/usr/local/cuda/include/cub/device/dispatch/dispatch_histogram.cuh(246): error: expected a ";"
        using CommonT = typename cuda::std::common_type<LevelT, SampleT>::type;
                                                              ^
/usr/local/cuda/include/cub/device/dispatch/dispatch_histogram.cuh(247): error: name followed by "::" must be a class or namespace name
        static_assert(cuda::std::is_convertible<CommonT, int>::value,
                            ^
/usr/local/cuda/include/cub/device/dispatch/dispatch_histogram.cuh(247): error: expected a string literal
        static_assert(cuda::std::is_convertible<CommonT, int>::value,
                                                         ^
/usr/local/cuda/include/cub/device/dispatch/dispatch_histogram.cuh(250): error: name followed by "::" must be a class or namespace name
        static_assert(cuda::std::is_trivially_copyable<CommonT>::value,
                            ^
/usr/local/cuda/include/cub/device/dispatch/dispatch_histogram.cuh(250): error: the global scope has no "value"
        static_assert(cuda::std::is_trivially_copyable<CommonT>::value,
                                                                 ^
/usr/local/cuda/include/cub/device/dispatch/dispatch_histogram.cuh(277): error: name followed by "::" must be a class or namespace name
                            cuda::std::true_type )
                                  ^
/usr/local/cuda/include/cub/device/dispatch/dispatch_histogram.cuh(291): error: name followed by "::" must be a class or namespace name
                            cuda::std::false_type )
                                  ^
/usr/local/cuda/include/cub/device/dispatch/dispatch_histogram.cuh(288): error: invalid redeclaration of member function template "DreamPlace::cub::DispatchHistogram<NUM_CHANNELS, NUM_ACTIVE_CHANNELS, SampleIteratorT, CounterT, LevelT, OffsetT>::ScaleTransform::ScaleT DreamPlace::cub::DispatchHistogram<NUM_CHANNELS, NUM_ACTIVE_CHANNELS, SampleIteratorT, CounterT, LevelT, OffsetT>::ScaleTransform::ComputeScale(int, T, T, <error-type>)" (declared at line 274)
        ScaleT ComputeScale(int num_levels,
               ^
/usr/local/cuda/include/cub/device/dispatch/dispatch_histogram.cuh(354): error: name followed by "::" must be a class or namespace name
                       cuda::std::true_type )
                             ^
/usr/local/cuda/include/cub/device/dispatch/dispatch_histogram.cuh(364): error: name followed by "::" must be a class or namespace name
                       cuda::std::false_type )
                             ^
/usr/local/cuda/include/cub/device/dispatch/dispatch_histogram.cuh(361): error: invalid redeclaration of member function template "int DreamPlace::cub::DispatchHistogram<NUM_CHANNELS, NUM_ACTIVE_CHANNELS, SampleIteratorT, CounterT, LevelT, OffsetT>::ScaleTransform::ComputeBin(T, T, DreamPlace::cub::DispatchHistogram<NUM_CHANNELS, NUM_ACTIVE_CHANNELS, SampleIteratorT, CounterT, LevelT, OffsetT>::ScaleTransform::ScaleT, <error-type>)" (declared at line 351)
        int ComputeBin(T sample,
            ^
/usr/local/cuda/include/cub/device/dispatch/dispatch_histogram.cuh(308): error: name followed by "::" must be a class or namespace name
                                    cuda::std::is_floating_point<T>{});
                                          ^
/usr/local/cuda/include/cub/device/dispatch/dispatch_histogram.cuh(308): error: expected an expression
                                    cuda::std::is_floating_point<T>{});
                                                                   ^
/usr/local/cuda/include/cub/device/dispatch/dispatch_histogram.cuh(317): error: name followed by "::" must be a class or namespace name
          { return this->ComputeScale(num_levels, __half2float(max_level), __half2float(min_level), cuda::std::true_type{}); }
                                                                                                          ^

/usr/local/cuda/include/cub/device/dispatch/dispatch_histogram.cuh(317): error: expected a ")"
          { return this->ComputeScale(num_levels, __half2float(max_level), __half2float(min_level), cuda::std::true_type{}); }
                                                                                                                        ^
/usr/local/cuda/include/cub/device/dispatch/dispatch_histogram.cuh(377): error: name followed by "::" must be a class or namespace name
                                  cuda::std::is_floating_point<T>{});
                                        ^
/usr/local/cuda/include/cub/device/dispatch/dispatch_histogram.cuh(377): error: expected an expression
                                  cuda::std::is_floating_point<T>{});
                                                                 ^
18 errors detected in the compilation of "/home/wangzhen/Project/DREAMPlace/dreamplace/ops/pin_pos/src/pin_pos_cuda_segment_kernel.cu".
-- Removing /home/wangzhen/Project/DREAMPlace/build/dreamplace/ops/pin_pos/CMakeFiles/pin_pos_cuda_segment_cuda_tmp.dir/src/./pin_pos_cuda_segment_cuda_tmp_generated_pin_pos_cuda_segment_kernel.cu.o
[... 32 lines cut ...]
                 from /usr/local/cuda/include/thrust/iterator/iterator_facade.h:37,
                 from /home/wangzhen/Project/DREAMPlace/thirdparty/cub/cub/device/../iterator/arg_index_input_iterator.cuh:48,
                 from /home/wangzhen/Project/DREAMPlace/thirdparty/cub/cub/device/device_reduce.cuh:41,
                 from /home/wangzhen/Project/DREAMPlace/thirdparty/cub/cub/cub.cuh:53,
                 from /home/wangzhen/Project/DREAMPlace/dreamplace/ops/pin_pos/../utility/src/utils_cub.cuh:18,
                 from /home/wangzhen/Project/DREAMPlace/dreamplace/ops/pin_pos/src/pin_pos_cuda_segment_kernel.cu:6:
/usr/local/cuda/include/thrust/system/cuda/config.h:116:2: error: #error The version of CUB in your include path is not compatible with this release of Thrust. CUB is now included in the CUDA Toolkit, so you no longer need to use your own checkout of CUB. Define THRUST_IGNORE_CUB_VERSION_CHECK to ignore this.
  116 | #error The version of CUB in your include path is not compatible with this release of Thrust. CUB is now included in the CUDA Toolkit, so you no longer need to use your own checkout of CUB. Define THRUST_IGNORE_CUB_VERSION_CHECK to ignore this.
      |  ^~~~~
CMake Error at pin_pos_cuda_segment_cuda_tmp_generated_pin_pos_cuda_segment_kernel.cu.o.Debug.cmake:220 (message):
  Error generating
[... cut ...]
make[2]: Leaving directory '/home/wangzhen/Project/DREAMPlace/build'
make[1]: *** [CMakeFiles/Makefile2:3183: dreamplace/ops/pin_pos/CMakeFiles/pin_pos_cuda_segment_cuda_tmp.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```

**It seemed that I must use library files to compile this.**

I found some similar issues.
https://github.com/limbo018/DREAMPlace/issues/34, https://github.com/limbo018/DREAMPlace/pull/41

But I could not find solutions from these issues. How can I deal with this problem?

limbo018, 2023-04-01:
Hi, thanks for pointing out the problem. I haven't tested on CUDA 12. Most of our machines use CUDA 11. 

1. I don't think you can switch to thirdparty/cub, as CUDA has integrated cub since CUDA 11. The version in thirdparty/cub is only for CUDA 10 or earlier versions. 

2. Probably the API for cub in CUDA 12 has been changed. Actually, the implementation of pin_pos_cuda_segment is not used, so you do not need to compile it. You can try commenting out ```line 30-34``` and ```line 38``` in ```dreamplace/ops/pin_pos/CMakeLists.txt``` to see if the compiling error is resolved.

wangzhen0518, 2023-04-01:
Thank you! I met some new errors after commenting the codes in `dreamplace/ops/pin_pos/CMakeLists.txt`. The  errors were all the same as before.

```bash
/usr/local/cuda/include/cub/device/dispatch/dispatch_histogram.cuh(246): error: name followed by "::" must be a class or namespace name
        using CommonT = typename cuda::std::common_type<LevelT, SampleT>::type;
                                       ^
......
```

First, I met an error when compiling `global_swap_cuda_kernel.cu`, so I commented `line 44-55` in `dreamplace/ops/global_swap/CMakeLists.txt` as following.

```cmake
# if(TORCH_ENABLE_CUDA)
# add_pytorch_extension(${TARGET_NAME}_cuda 
#   src/global_swap_cuda.cpp
#   src/global_swap_cuda_kernel.cu
#   EXTRA_INCLUDE_DIRS ${INCLUDE_DIRS}
#   EXTRA_LINK_LIBRARIES ${LINK_LIBS}
#   EXTRA_DEFINITIONS DRAWPLACE=${DRAWPLACE})

# install(TARGETS 
#   ${TARGET_NAME}_cuda 
#   DESTINATION dreamplace/ops/${OP_NAME})
# endif(TORCH_ENABLE_CUDA)
```

After that, I encountered another error when compiling `global_swap_cuda_kernel.cu`, so I commented `line 48-60` in file `dreamplace/ops/independent_set_matching/CMakeLists.txt` as following.

```cmake
# if(TORCH_ENABLE_CUDA)
# set(CUDA_SEPARABLE_COMPILATION ON)
# add_pytorch_extension(${TARGET_NAME}_cuda 
#   src/independent_set_matching_cuda.cpp
#   src/independent_set_matching_cuda_kernel.cu
#   EXTRA_INCLUDE_DIRS ${INCLUDE_DIRS}
#   EXTRA_LINK_LIBRARIES ${LINK_LIBS} ${CUDA_curand_LIBRARY}
#   EXTRA_DEFINITIONS DRAWPLACE=${DRAWPLACE})
# install(TARGETS 
#   ${TARGET_NAME}_cuda 
#   DESTINATION dreamplace/ops/${OP_NAME})
# set(CUDA_SEPARABLE_COMPILATION OFF)
# endif(TORCH_ENABLE_CUDA)
```

Then, I encountered another error when compiling `k_reorder_cuda_kernel.cu`... I did not think I could solve the problem, so I **uninstalled cuda12.1 and installed cuda11.8. Finally, no more errors occurred.**

I guess the problem was caused by `using xxx` in `.cpp` or `.cu`. If the authors could adapt DreamPlace to cuda12.x+ in the future, I would be very grateful to them.

closed 2023-04-06 by wangzhen0518