limbo018/DREAMPlace #123: make过程发生的错误

AOrigin, 2023-04-06:
您好~我在make的时候发生错误，请问这个问题如何解决，已经摸索了很久但是还是找不到问题所在？（我用的无docker方式创建）
/disc1/aqy/AQY/EDA/DREAMPlace/dreamplace/ops/place_io/src/Enums.cpp:67:89: error: must #include <typeinfo> before using typeid
         dreamplacePrint(kWARN, "%s::%s unknown enum type %s, set to UNKNOWN\n", typeid(*this).name(), __func__, s.c_str());
                                                                                         ^
/disc1/aqy/AQY/EDA/DREAMPlace/dreamplace/ops/place_io/src/Enums.cpp: In member function ‘virtual DreamPlace::PlaceStatus::enum_type DreamPlace::PlaceStatus::str2Enum(const string&) const’:
/disc1/aqy/AQY/EDA/DREAMPlace/dreamplace/ops/place_io/src/Enums.cpp:113:89: error: must #include <typeinfo> before using typeid
         dreamplacePrint(kWARN, "%s::%s unknown enum type %s, set to UNKNOWN\n", typeid(*this).name(), __func__, s.c_str());
                                                                                         ^
/disc1/aqy/AQY/EDA/DREAMPlace/dreamplace/ops/place_io/src/Enums.cpp: In member function ‘virtual DreamPlace::MultiRowAttr::enum_type DreamPlace::MultiRowAttr::str2Enum(const string&) const’:
/disc1/aqy/AQY/EDA/DREAMPlace/dreamplace/ops/place_io/src/Enums.cpp:159:89: error: must #include <typeinfo> before using typeid
         dreamplacePrint(kWARN, "%s::%s unknown enum type %s, set to UNKNOWN\n", typeid(*this).name(), __func__, s.c_str());
                                                                                         ^
/disc1/aqy/AQY/EDA/DREAMPlace/dreamplace/ops/place_io/src/Enums.cpp: In member function ‘virtual DreamPlace::SignalDirect::enum_type DreamPlace::SignalDirect::str2Enum(const string&) const’:
/disc1/aqy/AQY/EDA/DREAMPlace/dreamplace/ops/place_io/src/Enums.cpp:205:89: error: must #include <typeinfo> before using typeid
         dreamplacePrint(kWARN, "%s::%s unknown enum type %s, set to UNKNOWN\n", typeid(*this).name(), __func__, s.c_str());
                                                                                         ^
/disc1/aqy/AQY/EDA/DREAMPlace/dreamplace/ops/place_io/src/Enums.cpp: In member function ‘virtual DreamPlace::PlanarDirect::enum_type DreamPlace::PlanarDirect::str2Enum(const string&) const’:
/disc1/aqy/AQY/EDA/DREAMPlace/dreamplace/ops/place_io/src/Enums.cpp:247:89: error: must #include <typeinfo> before using typeid
         dreamplacePrint(kWARN, "%s::%s unknown enum type %s, set to UNKNOWN\n", typeid(*this).name(), __func__, s.c_str());
                                                                                         ^
/disc1/aqy/AQY/EDA/DREAMPlace/dreamplace/ops/place_io/src/Enums.cpp: In member function ‘virtual DreamPlace::ReportFlag::enum_type DreamPlace::ReportFlag::str2Enum(const string&) const’:
/disc1/aqy/AQY/EDA/DREAMPlace/dreamplace/ops/place_io/src/Enums.cpp:297:89: error: must #include <typeinfo> before using typeid
         dreamplacePrint(kWARN, "%s::%s unknown enum type %s, set to UNKNOWN\n", typeid(*this).name(), __func__, s.c_str());
                                                                                         ^
/disc1/aqy/AQY/EDA/DREAMPlace/dreamplace/ops/place_io/src/Enums.cpp: In member function ‘virtual DreamPlace::RowPlaceSolver::enum_type DreamPlace::RowPlaceSolver::str2Enum(const string&) const’:
/disc1/aqy/AQY/EDA/DREAMPlace/dreamplace/ops/place_io/src/Enums.cpp:350:89: error: must #include <typeinfo> before using typeid
         dreamplacePrint(kWARN, "%s::%s unknown enum type %s, set to UNKNOWN\n", typeid(*this).name(), __func__, s.c_str());
                                                                                         ^
/disc1/aqy/AQY/EDA/DREAMPlace/dreamplace/ops/place_io/src/Enums.cpp: In member function ‘virtual DreamPlace::MinCostFlowSolver::enum_type DreamPlace::MinCostFlowSolver::str2Enum(const string&) const’:
/disc1/aqy/AQY/EDA/DREAMPlace/dreamplace/ops/place_io/src/Enums.cpp:395:89: error: must #include <typeinfo> before using typeid
         dreamplacePrint(kWARN, "%s::%s unknown enum type %s, set to UNKNOWN\n", typeid(*this).name(), __func__, s.c_str());
                                                                                         ^
/disc1/aqy/AQY/EDA/DREAMPlace/dreamplace/ops/place_io/src/Enums.cpp: In member function ‘virtual DreamPlace::RegionAssignSolver::enum_type DreamPlace::RegionAssignSolver::str2Enum(const string&) const’:
/disc1/aqy/AQY/EDA/DREAMPlace/dreamplace/ops/place_io/src/Enums.cpp:436:89: error: must #include <typeinfo> before using typeid
         dreamplacePrint(kWARN, "%s::%s unknown enum type %s, set to UNKNOWN\n", typeid(*this).name(), __func__, s.c_str());
                                                                                         ^
/disc1/aqy/AQY/EDA/DREAMPlace/dreamplace/ops/place_io/src/Enums.cpp: In member function ‘virtual DreamPlace::GlobalMoveEffort::enum_type DreamPlace::GlobalMoveEffort::str2Enum(const string&) const’:
/disc1/aqy/AQY/EDA/DREAMPlace/dreamplace/ops/place_io/src/Enums.cpp:481:89: error: must #include <typeinfo> before using typeid
         dreamplacePrint(kWARN, "%s::%s unknown enum type %s, set to UNKNOWN\n", typeid(*this).name(), __func__, s.c_str());
                                                                                         ^
/disc1/aqy/AQY/EDA/DREAMPlace/dreamplace/ops/place_io/src/Enums.cpp: In member function ‘virtual DreamPlace::GlobalMoveAlgo::enum_type DreamPlace::GlobalMoveAlgo::str2Enum(const string&) const’:
/disc1/aqy/AQY/EDA/DREAMPlace/dreamplace/ops/place_io/src/Enums.cpp:522:89: error: must #include <typeinfo> before using typeid
         dreamplacePrint(kWARN, "%s::%s unknown enum type %s, set to UNKNOWN\n", typeid(*this).name(), __func__, s.c_str());
                                                                                         ^
/disc1/aqy/AQY/EDA/DREAMPlace/dreamplace/ops/place_io/src/Enums.cpp: In member function ‘virtual DreamPlace::RegionType::enum_type DreamPlace::RegionType::str2Enum(const string&) const’:
/disc1/aqy/AQY/EDA/DREAMPlace/dreamplace/ops/place_io/src/Enums.cpp:563:89: error: must #include <typeinfo> before using typeid
         dreamplacePrint(kWARN, "%s::%s unknown enum type %s, set to UNKNOWN\n", typeid(*this).name(), __func__, s.c_str());
                                                                                         ^
dreamplace/ops/place_io/CMakeFiles/place_io_cpp.dir/build.make:94: recipe for target 'dreamplace/ops/place_io/CMakeFiles/place_io_cpp.dir/src/Enums.cpp.o' failed
make[2]: *** [dreamplace/ops/place_io/CMakeFiles/place_io_cpp.dir/src/Enums.cpp.o] Error 1
make[2]: Leaving directory '/disc1/aqy/AQY/EDA/DREAMPlace/build'
CMakeFiles/Makefile2:3032: recipe for target 'dreamplace/ops/place_io/CMakeFiles/place_io_cpp.dir/all' failed
make[1]: *** [dreamplace/ops/place_io/CMakeFiles/place_io_cpp.dir/all] Error 2
make[1]: Leaving directory '/disc1/aqy/AQY/EDA/DREAMPlace/build'
Makefile:135: recipe for target 'all' failed
make: *** [all] Error 2

limbo018, 2023-04-06:
Hi, I think your problem has been solved in this pull request. I have merged it into ```develop``` branch. You may try switching to develop branch and have test. It needs to ```#include <typeinfo>``` in Enums.cpp.

AOrigin, 2023-04-06:
谢谢老师，问题已经解决。下次我再认真搜索看看问答区。尽量不占用您宝贵的时间。真的非常感谢了