limbo018/DREAMPlace #125: when i processiong global placement, an Error occur,how to fix this problem?

imcamino, 2023-04-07:
[image]

limbo018, 2023-04-08:
Hi, you may check out ```develop``` branch. The problem comes from higher python version. We have fixed it. 

https://github.com/limbo018/DREAMPlace/blob/8e9f599084e0658f81a1e29706594e55a8060a82/dreamplace/PlaceDB.py#L576

imcamino, 2023-04-08:
Thanks for your help. i already change to dev branch. but the problem still exists.

imcamino, 2023-04-08:
[image]
I think the Exception is the critical problem instead of the warnning

limbo018, 2023-04-08:
It looks like you have changed the code? (you wrap it in the subprocess) Does it original version of code run?

imcamino, 2023-04-09:
Yep.i just wrap it in a Entry script。 Actually It works for ISPD case，but fail to run other case come from open source。

limbo018, 2023-04-09:
If you can share the LEF/DEF files of your case, I can help take a look. There might some problem when reading the design.

imcamino, 2023-04-09:
[test_case.zip](https://github.com/limbo018/DREAMPlace/files/11185711/test_case.zip)
There is one of my case.

imcamino, 2023-04-12:
is there any problem in this design?

limbo018, 2023-04-12:
I will take a look during the weekend. Quite busy this week. 

                       
Original

 

 
&mdash;

imcamino, 2023-04-12:
Ok,  I appreciate your time.

limbo018, 2023-04-14:
Hi, I just took a look at your DEF file. The ```COMPONENTS``` section is missing.

imcamino, 2023-04-14:
Thanks for your reply. i will write the def out with unplaced stdcell & try.