limbo018/DREAMPlace #129: save .nets file

zerzerzerz, 2023-05-27:
Thanks for your excellent works!
I want to save `.nets` file with function `placedb.write_nets(params, nets_file_path)`, but there are something strange like
```
NetDegree : 8 b'n7'
	b'o3' b'OUTPUT' : 2 3
	b'o22446' b'INPUT' : 12 4
	b'o22500' b'INPUT' : 11 0
	b'o22569' b'INPUT' : 12 7
	b'o22605' b'INPUT' : 12 4
	b'o22643' b'INPUT' : 12 4
	b'o22646' b'INPUT' : 4 11
	b'o22356' b'INPUT' : 12 4
```
, where many `b'` are in this files.

Could you add `.decode()` like
```python
  def write_nets(self, params, net_file):
      """
      @brief write .net file
      @param params parameters
      @param net_file .net file
      """
      tt = time.time()
      logging.info("writing to %s" % (net_file))
      content = "UCLA nets 1.0\n"
      content += "\nNumNets : %d" % (len(self.net2pin_map))
      content += "\nNumPins : %d" % (len(self.pin2net_map))
      content += "\n"

      for net_id in range(len(self.net2pin_map)):
          pins = self.net2pin_map[net_id]
          content += "\nNetDegree : %d %s" % (len(pins), self.net_names[net_id].decode())
          for pin_id in pins:
              content += "\n\t%s %s : %d %d" % (self.node_names[self.pin2node_map[pin_id]].decode(), self.pin_direct[pin_id].decode(), self.pin_offset_x[pin_id]/params.scale_factor, self.pin_offset_y[pin_id]/params.scale_factor)

      with open(net_file, "w") as f:
          f.write(content)
      logging.info("write_nets takes %.3f seconds" % (time.time()-tt))
```
Again, thanks for your excellent works!

limbo018, 2023-06-01:
Thank you for the suggestions! Will add it soon. It would also be great if you can create a pull request so I can merge it.

zerzerzerz, 2023-06-01:
> Thank you for the suggestions! Will add it soon. It would also be great if you can create a pull request so I can merge it.

OK and I have already created a PR. Thanks for your excellent works!