limbo018/DREAMPlace #136: meet error when making

Moshibing, 2023-06-15:
/home/wu/Project_ICCAD/DREAMPlace-master/build/dreamplace/ops/place_io && /usr/bin/c++ -DDRAWPLACE=0 -DENABLE_CUDA=1 -DTORCH_EXTENSION_NAME=place_io_cpp -DTORCH_VERSION_MAJOR=2 -DTORCH_VERSION_MINOR=0 -D_GLIBCXX_USE_CXX11_ABI=0 -Dplace_io_cpp_EXPORTS -I/home/wu/Project_ICCAD/DREAMPlace-master/thirdparty/flute -I/home/wu/Project_ICCAD/DREAMPlace-master/thirdparty/OpenTimer -I/home/wu/Project_ICCAD/DREAMPlace-master/dreamplace/ops/place_io/.. -I/home/wu/Project_ICCAD/DREAMPlace-master/thirdparty/Limbo -isystem /usr/include/python3.8 -isystem /home/wu/Project_ICCAD/DREAMPlace-master/thirdparty/pybind11/include -isystem /usr/local/lib/python3.8/dist-packages/torch/include -isystem /usr/local/lib/python3.8/dist-packages/torch/include/torch/csrc/api/include -O3 -DNDEBUG -fPIC -fvisibility=hidden -flto -fno-fat-lto-objects -D_GLIBCXX_USE_CXX11_ABI=0 -fopenmp -std=gnu++17 -MD -MT dreamplace/ops/place_io/CMakeFiles/place_io_cpp.dir/src/PlaceDB.cpp.o -MF CMakeFiles/place_io_cpp.dir/src/PlaceDB.cpp.o.d -o CMakeFiles/place_io_cpp.dir/src/PlaceDB.cpp.o -c /home/wu/Project_ICCAD/DREAMPlace-master/dreamplace/ops/place_io/src/PlaceDB.cpp
/home/wu/Project_ICCAD/DREAMPlace-master/dreamplace/ops/place_io/src/PlaceDB.cpp: In member function ‘virtual void DreamPlace::PlaceDB::add_def_pin(const DefParser::Pin&)’:
/home/wu/Project_ICCAD/DREAMPlace-master/dreamplace/ops/place_io/src/PlaceDB.cpp:325:17: error: ‘const struct DefParser::Pin’ has no member named ‘vPinPort’
  325 |   } else if (!p.vPinPort.empty()) {
      |                 ^~~~~~~~
/home/wu/Project_ICCAD/DREAMPlace-master/dreamplace/ops/place_io/src/PlaceDB.cpp:327:35: error: ‘const struct DefParser::Pin’ has no member named ‘vPinPort’
  327 |     for (std::size_t i = 0; i < p.vPinPort.size(); ++i) {
      |                                   ^~~~~~~~
/home/wu/Project_ICCAD/DREAMPlace-master/dreamplace/ops/place_io/src/PlaceDB.cpp:329:14: error: ‘const struct DefParser::Pin’ has no member named ‘vPinPort’
  329 |       if (!p.vPinPort[i].vLayer.empty()) {
      |              ^~~~~~~~
/home/wu/Project_ICCAD/DREAMPlace-master/dreamplace/ops/place_io/src/PlaceDB.cpp:360:17: error: ‘const struct DefParser::Pin’ has no member named ‘vPinPort’
  360 |   } else if (!p.vPinPort.empty()) {
      |                 ^~~~~~~~
/home/wu/Project_ICCAD/DREAMPlace-master/dreamplace/ops/place_io/src/PlaceDB.cpp:361:22: error: ‘const struct DefParser::Pin’ has no member named ‘vPinPort’
  361 |     node.setOrient(p.vPinPort.front().orient);
      |                      ^~~~~~~~
/home/wu/Project_ICCAD/DREAMPlace-master/dreamplace/ops/place_io/src/PlaceDB.cpp:390:19: error: ‘const struct DefParser::Pin’ has no member named ‘vPinPort’
  390 |     } else if (!p.vPinPort.empty()) {
      |                   ^~~~~~~~
/home/wu/Project_ICCAD/DREAMPlace-master/dreamplace/ops/place_io/src/PlaceDB.cpp:391:37: error: ‘const struct DefParser::Pin’ has no member named ‘vPinPort’
  391 |       for (std::size_t i = 0; i < p.vPinPort.size(); ++i) {
      |                                     ^~~~~~~~
/home/wu/Project_ICCAD/DREAMPlace-master/dreamplace/ops/place_io/src/PlaceDB.cpp:392:20: error: ‘PinPort’ is not a member of ‘DefParser’
  392 |         DefParser::PinPort const& pport = p.vPinPort[i];
      |                    ^~~~~~~
/home/wu/Project_ICCAD/DREAMPlace-master/dreamplace/ops/place_io/src/PlaceDB.cpp:393:37: error: ‘pport’ was not declared in this scope
  393 |         for (std::size_t j = 0; j < pport.vLayer.size(); ++j) {
      |                                     ^~~~~
/home/wu/Project_ICCAD/DREAMPlace-master/dreamplace/ops/place_io/src/PlaceDB.cpp:486:17: error: ‘const struct DefParser::Pin’ has no member named ‘vPinPort’
  486 |   } else if (!p.vPinPort.empty()) {
      |                 ^~~~~~~~
/home/wu/Project_ICCAD/DREAMPlace-master/dreamplace/ops/place_io/src/PlaceDB.cpp:487:35: error: ‘const struct DefParser::Pin’ has no member named ‘vPinPort’
  487 |     for (std::size_t i = 0; i < p.vPinPort.size(); ++i) {
      |                                   ^~~~~~~~
/home/wu/Project_ICCAD/DREAMPlace-master/dreamplace/ops/place_io/src/PlaceDB.cpp:491:18: error: ‘PinPort’ is not a member of ‘DefParser’
  491 |       DefParser::PinPort const& pport = p.vPinPort[i];
      |                  ^~~~~~~
/home/wu/Project_ICCAD/DREAMPlace-master/dreamplace/ops/place_io/src/PlaceDB.cpp:492:35: error: ‘pport’ was not declared in this scope
  492 |       for (std::size_t j = 0; j < pport.vLayer.size(); ++j) {
      |                                   ^~~~~
make[2]: *** [dreamplace/ops/place_io/CMakeFiles/place_io_cpp.dir/build.make:191：dreamplace/ops/place_io/CMakeFiles/place_io_cpp.dir/src/PlaceDB.cpp.o] 错误 1
make[2]: 离开目录“/home/wu/Project_ICCAD/DREAMPlace-master/build”
make[1]: *** [CMakeFiles/Makefile2:3948：dreamplace/ops/place_io/CMakeFiles/place_io_cpp.dir/all] 错误 2
make[1]: 离开目录“/home/wu/Project_ICCAD/DREAMPlace-master/build”
make: *** [Makefile:139：all] 错误 2

I need some help please.

limbo018, 2023-06-15:
Do you pull and update the submodules? 
```
git submodule --init
git submodule --update 
```

Moshibing, 2023-06-16:
Yes, problem solved, thank you!

kiloGrand, 2024-08-12:
i have received error when making:
`CMake Error at pin_pos_cuda_segment_cuda_tmp_generated_pin_pos_cuda_segment_kernel.cu.o.Release.cmake:280 (message):
  Error generating file
  ~/DREAMPlace/build/dreamplace/ops/pin_pos/CMakeFiles/pin_pos_cuda_segment_cuda_tmp.dir/src/./pin_pos_cuda_segment_cuda_tmp_generated_pin_pos_cuda_segment_kernel.cu.o
`

limbo018, 2024-08-12:
Hi, please check your CUDA version. We recommend cuda 11.5. Newer versions have compatibility issues.

kiloGrand, 2024-08-12:
After I changed the version of Coda to 11.7 on ubuntu 22.04, recompiled it, the following error occurred：
`In file included from ~/DREAMPlace/thirdparty/OpenTimer/unittest/utility.cpp:3:
~/DREAMPlace/thirdparty/OpenTimer/unittest/doctest.h:4008:47: error: size of array ‘altStackMem’ is not an integral constant-expression
 4008 |         static char             altStackMem[4 * SIGSTKSZ];
      |                                               ^
make[2]: *** [thirdparty/OpenTimer/CMakeFiles/util.dir/build.make:79：thirdparty/OpenTimer/CMakeFiles/util.dir/unittest/utility.cpp.o] 错误 1`

limbo018, 2024-08-12:
Please see the following issue to resolve the error. 

https://github.com/limbo018/DREAMPlace/issues/133#issuecomment-1780502026