limbo018/DREAMPlace #138: Issue with enable timing_opt_flag

mumu157, 2023-06-24:
I encountered an issue when I enabled the timing_opt_flag feature. The following error occurred:
```
W 63232 23-06-24 13:00:26 timer.cpp:603] can't insert PO 0 (already existed)
I 63232 23-06-24 13:00:26 verilog.cpp:21] added verilog module "ibex_core" [gates:15650]
E 63232 23-06-24 13:00:26 sdc.cpp:67] set_input_delay: port "boot_addr_i[0]" not found
E 63232 23-06-24 13:00:26 sdc.cpp:67] set_input_delay: port "boot_addr_i[10]" not found
```

However, I have found the definition of the ports in the netlist:
` input [31:0] boot_addr_i; `

I would appreciate any suggestions or help regarding this issue. Thank you!

limbo018, 2023-06-25:
The placer relies on OpenTimer to provide timing analysis and feedback. It looks like OpenTimer's parser reports error with your benchmark.

ZhuLinsen, 2023-08-07:
@limbo018  Dr. Lin, I apologize for replying so late. I have another question: If I use the BOOKshelf format, can I still use timing_opt? I've looked at the code, and many of the processes seem to be tied to reading Verilog and DEF/LEF files.

```
[WARNING] IO pin not found: clk_i
[WARNING] IO pin not found: debug_req_i
[WARNING] IO pin not found: ipi_i
[WARNING] IO pin not found: rst_ni
[WARNING] IO pin not found: time_irq_i
[WARNING] IO pin not found: axi_req_o
[WARNING] IO pin not found: axi_resp_i
[WARNING] IO pin not found: boot_addr_i
[WARNING] IO pin not found: hart_id_i
[WARNING] IO pin not found: irq_i
[ASSERT ] /DREAMPlace/dreamplace/ops/place_io/src/PlaceDB.cpp:756: virtual void DreamPlace::PlaceDB::verilog_net_declare_cbk(const string&, const VerilogParser::Range&): Assertion `range.low == range.high' failed: do not support bus yet
```

ZhuLinsen, 2023-08-07:
@limbo018  Dr. Lin, I apologize for replying so late. I have another question: If I use the BOOKshelf format, can I still use timing_opt? I've looked at the code, and many of the processes seem to be tied to reading Verilog and DEF/LEF files.

limbo018, 2023-08-07:
I'm afraid not, as OpenTimer needs to read Verilog and Liberty as input for timing analysis. Although DREAMPlace can deal with bookshelf format, OpenTimer still relies on information in Verilog and Liberty.

closed 2023-11-16 by mumu157