limbo018/DREAMPlace #141: meet cub error when making

Mathew0553, 2023-07-03:
Dear professor，i have received error when making:
```
/usr/include/cub/util_namespace.cuh:46:2: error: #error CUB requires a definition of CUB_NS_QUALIFIER when CUB_NS_PREFIX/POSTFIX are defined.
   46 | #error CUB requires a definition of CUB_NS_QUALIFIER when CUB_NS_PREFIX/POSTFIX are defined.
      |  ^~~~~
```
although this error can be search on the google , i tried almost each kinds of method, but still doesn't work for me,
maybe it should change somewhere in `util_namespace.cuh` ?

please kindly help! it is important for me,thanks

limbo018, 2023-07-04:
It looks like you have a separate cub installation under ```/usr/include```. What's your CUDA version and where is it installed?

Mathew0553, 2023-07-04:
> It looks like you have a separate cub installation under `/usr/include`
 yes,i'm just one of the student users in my laboratory，cub is installed on `/usr/include ` but not used the cub in my conda list

>What's your CUDA version and where is it installed?
CUDA version is 11.7 and it is installed on `/usr/local`
how can i fix it?
thanks for reply!

limbo018, 2023-07-04:
CUDA 11.7 has already integrated cub. So the other cub under ```/usr/include``` conflicts with it. You might need to try 
1. Move the other cub somewhere else so that it is not included. 
2. Exclude ```/usr/include``` from your ```PATH``` environment, but this may cause other problems.

limbo018, 2023-07-17:
Hi, are you work with the latest code? I checked the header file https://github.com/limbo018/DREAMPlace/blob/master/dreamplace/ops/utility/src/utils_cub.cuh and found `CUB_NS_QUALIFIER` is already defined.

Mathew0553, 2023-07-18:
Thanks for your patients
okay, i was told to used DREAMPlace3.0.0，
Is there a significant difference between the two versions, and can the latest code replace version 3.0.0？
Besides, i tried DREAMPlace latest code and meet another Unsolvable torch problem when cmake ..
[image]
i got this with the same environment..

if lastest code is different than previous，i would like to fix the version 3.0.0 bug. Do you have any good suggestions？
@limbo018

limbo018, 2023-07-18:
The code on master branch is version 4.0.0 which supports timing-driven optimization (off by default). If you want to fix the problem with version 3.0.0, you can follow the contents in https://github.com/limbo018/DREAMPlace/blob/master/dreamplace/ops/utility/src/utils_cub.cuh modify where cub is included.

limbo018, 2023-07-18:
For the error with version 4.0.0, you didn't pull the submodules. 
```
git submodules init
git submodules update
```

Mathew0553, 2023-07-18:
> For the error with version 4.0.0, you didn't pull the submodules.
> 
> ```
> git submodules init
> git submodules update
> ```

i execute `git clone --recursive https://github.com/limbo018/DREAMPlace.git` but sill got the same  torch error , it can not found the correct torch and python.
[image]

limbo018, 2023-07-18:
Can you try `-DPython_EXECUTABLE`? Another issue reports that the new FindPython in cmake has changed the API to case-sensitive.

Mathew0553, 2023-07-18:
i would like to know can i download version 3.0 through `git clone --recursive`?

Mathew0553, 2023-07-18:
Sincerely thank you for all your help and patience. I can build the latest code now. Wish u best !@limbo018

limbo018, 2023-07-18:
> i would like to know can i download version 3.0 through `git clone --recursive`?

You can try following command according a tutorial for git (https://devconnected.com/how-to-checkout-git-tags/)
```
git checkout tags/3.0.0
```

Mathew0553, 2023-07-27:
> git checkout tags/3.0.0
Hi, professor Lin , I have try your advice and correctly get the 3.0.0version code and its submodules, but still got the cub error,i think the thirdparty's cub has not changed between 4.0 and 3.0. 
But never mind , currently i successfully using the 4.0 version.
Now , i have another question: 
[image]
when i wanna check `place_io_cpp.pydb`’s content, find out if it provides standard cell position（including node_x ,node_y,node_name and so on）. i found that `make install` step created the `place_io.cpp.so` executable file but not the cpp file. When I used to build, I also generated executable file instead of cpp file. how do i understand that?
i checked <[#39](https://github.com/limbo018/DREAMPlace/issues/39)> But it doesn't seem like the same issue.
Thanks for reply.

limbo018, 2023-07-28:
You might have misunderstanding about what `make` and `make install` do. It might be helpful to take a look at the basics of compiling and linking process. 
https://www.cs.cmu.edu/~213/lectures/15-linking.pdf

In a CMake building system (refer to CMake docs), `make` does the real compiling and linking, while `make install` only copies files to where you want to install.

Mathew0553, 2024-09-15:
> > i would like to know can i download version 3.0 through `git clone --recursive`?
> 
> You can try following command according a tutorial for git (https://devconnected.com/how-to-checkout-git-tags/)
> 
> ```
> git checkout tags/3.0.0
> ```

dear limbo professors,
sorry for bothering you,  i would like to ask you for device :recenlty i do my experiment , i must use DREAMPlace-3.0.0,  but still meet  the  below torch problem when i cmake,  i can fix it by myself , could you please help me  fix it?
[image]