limbo018/DREAMPlace #142: In 4.0 how to generate flute LUT

kouyichi, 2023-07-11:
in dreamplace/ops/timing/src/timing_cpp.cpp line139-144

`  using Point2i = ::DreamPlace::Point<int>;
  dreamplacePrint(kINFO, "launch rc tree construction...\n");
  flute::readLUT(
      "thirdparty/flute/lut.ICCAD2015/POWV9.dat",
      "thirdparty/flute/lut.ICCAD2015/POST9.dat");
  auto beg = std::chrono::steady_clock::now();
`
flute use a look up table that already made. If I want to use the 4.0 in something out of iccad2015, how can I generate the look up table?

enzoleo, 2023-07-11:
They come from the source code of `flute`. It should also work in other benchmarks.

kouyichi, 2023-07-13:
> They come from the source code of `flute`. It should also work in other benchmarks.

Sorry, I'm not very familiar with the flute code. I just saw there is a readLUT and not a writeLUT function; where can I find the function that generates a LUT from some file?

limbo018, 2023-07-13:
You'd better read the original FLUTE paper to get the idea how it works before looking into the code. Its LUT is quite general and does not requires re-generation for different designs. 

https://ieeexplore.ieee.org/document/4358497

                       
Original

closed 2023-07-26 by kouyichi