limbo018/DREAMPlace #143: The Error .so file When 'make install' over

LmMax0, 2023-07-17:
When I used the Conda (Python 3.9) environment, I compiled the project

`Cmake- DCMAKE_ INSTALL_ PREFIX=/home/mini liang/science/DREAMPlace/install - DPYTHON_ EXECUTABLE=[my conda Python URL]`

But in the final generated cpython so file version, it is place_ Io_ Cpp.cpython-311-x86_ 64 Linux gnu so (Python 3.11)

How should I modify the compilation options or make changes to make the so file conform to the Python version in my Conda environment?

Thank you for your reply.

limbo018, 2023-07-17:
It is related to pybind11 to detect the python version. You may check its document about how to specify custom python version. 

https://pybind11.readthedocs.io/en/stable/faq.html?highlight=cmake#cmake-doesn-t-detect-the-right-python-version

LmMax0, 2023-07-18:
> It is related to pybind11 to detect the python version. You may check its document about how to specify custom python version.
> 
> https://pybind11.readthedocs.io/en/stable/faq.html?highlight=cmake#cmake-doesn-t-detect-the-right-python-version

Thank you very much for your prompt reply！

I used `cmake .. -DCMAKE_INSTALL_PREFIX=/home/minliang/science/DREAMPlace/install  -DPython_EXECUTABLE=/home/minliang/.conda/envs/DeepPlace/bin/python` to rerun, then it works.

The new FindPython we maybe use `Python_EXECUTABLE` instead (`PYTHON` -> `Python`) which I find in https://github.com/pybind/pybind11/issues/2661

I wish you good health and success in your work

limbo018, 2023-07-18:
Thank you for the information. I have updated the readme to `Python_EXECUTABLE`.

closed 2023-07-20 by LmMax0