limbo018/DREAMPlace #144: Have no "flute.h" when make install

LmMax0, 2023-07-19:
[image]

I have uncommented the line of code `add_subdirectory(rmst_wl)`.

However, during the subsequent construction process, there were issues as shown in the above images. I have tried many methods myself, but none of them have been effective. I'm sorry, because I have a lack of knowledge in cmake.

I would like to ask how I can solve this problem, as this module of DREAMPlace has been referenced in other open source projects, and I cannot avoid this problem.

Thank you for your guidance！

limbo018, 2023-07-19:
Compile problem fixed in https://github.com/limbo018/DREAMPlace/commit/477ea7a0b4e479c8bfe425e9499946c6bfd5525b

You can try the `develop` branch.

LmMax0, 2023-07-20:
It works. Amazing!
Thank you very much!

closed 2023-07-20 by LmMax0