limbo018/DREAMPlace #145: question about the macros

Icamd, 2023-07-21:
Dear professor Lin:
    Thank you for your code! When I was using the DreamPlace to solve the placement of the ispd2005 benchmarks, I find that all the macros are already fixed and cannot be moved. But there are a lot of papers that optimize the position of macros using reinforcement learning and claim to have good results. Does the DreamPlace algorithm optimize the location of macros? Or is it just optimizing standard cells? And are those fixed macros in the ispd2005 benchmark "expert data" that have eliminated overlap? And do you how to generate the placement pictures/ is there any program that visualizes the Dreamplace algorithm? Thank you professor!

limbo018, 2023-07-21:
You can modify the files in the ispd2005 benchmarks to make FIXED nodes movable. DREAMPlace supports mixed-size placement.

Icamd, 2023-07-21:
@limbo018 Thanks for your reply! Actually I am trying to train a reinforcement learning model based on offline training dataset with no overlap. Are those positions of fixed macros good enough, or does DreamPlace work better when macros are movable and achieve shorter hpwl? And how to make FIXED nodes movable ( just delete the "/FIXED" sign in benchmark )? Thank you!

Icamd, 2023-07-25:
@limbo018 Thank you! How to set FIXED nodes movable ? And how to generate the pictures of DREAMPlace result (is there any open source drawing program that visualizes Dreamplace)? Thank you!

Icamd, 2023-07-25:
@limbo018 Thank you! Could you please tell me how to set FIXED nodes movable in DREAMPlace? And how to generate the pictures of DREAMPlace result (is there any open source drawing program that visualizes DREAMPlace)? Thank you! @limbo018

Icamd, 2023-07-25:
@limbo018   Thank you! Could you please tell me how to set FIXED nodes movable in DREAMPlace? And how to generate the pictures of DREAMPlace result ( is there any open-source drawing program that visualizes DREAMPlace )? Thank you!

limbo018, 2023-07-25:
Please check `dreamplace/params.json` for all the options. You can set `plot_flag` in JSON to 1 to enable plotting the results.

limbo018, 2023-07-25:
> 

If you don't want to change the input files, you may need to modify the code in `PlaceDB.py` to make fixed nodes movable.

limbo018, 2023-07-26:
Hi, it is not recommended to change all fixed nodes to movable, which can make the optimization unstable. If you don't know how to choose movable macros, I recommend to you the `mms` benchmarks. Previous papers used it to study macro placement. 

https://disk.pku.edu.cn:443/link/2F11552C2A8AA27B24CC28EB8ECE8CE1
Valid Until: 2023-12-31 23:59

Icamd, 2023-07-26:
@limbo018 Thank you professor! I wish you all the best in your scientific research!

Icamd, 2023-08-06:
@limbo018 Hi Professor, sorry to bother you again. I have encountered new problems in generating pictures of the result. I set plot_flag to 1, but it appears that the program only visualizes part of macros （it seems to be 1/10 of the total macros）. How can I visualize all macros and standard cells? Thank you!

Icamd, 2023-11-04:
Dear professor:
I have another question about DREAMPlace. Is it possible to adjust some parameters in order to achieve optimal HPWL with ISPD05 mms benchmarks? (for example, by adding weights to wirelength or increase the gradient descent iteration number?) Thank you!

closed 2023-11-04 by Icamd