limbo018/DREAMPlace #148: meet an error when parsing  the Bookshelf format

mumu157, 2023-08-04:
Dear Dr. Lin,  I hope this message finds you well. I have encountered two questions while working with the DREAMPlace tool.  All designs are contained in the files below
[designs.zip](https://github.com/limbo018/DREAMPlace/files/12257981/designs.zip)

1、**When attempting to parse the Bookshelf file, I encountered a syntax error**:
```[WARNING] DEF input file not specified, cannot output DEF file; set to DEFSIMPLE
[INFO   ] lef_input =
[INFO   ] def_input =
[INFO   ] verilog_input =
[INFO   ] bookshelf_aux_input = place_iop_pad0_ISPD11_mapped.aux
[INFO   ] bookshelf_pl_input =
[INFO   ] def_size_input =
[INFO   ] def_output = -out.def
[INFO   ] rpt_output =
[INFO   ] target_util = 0
[INFO   ] max_displace = 0
[INFO   ] bin size = (10, 10) #rows
[INFO   ] sbin size = (5, 5) #rows
[INFO   ] bin_space_threshold = 0.2
[INFO   ] abu = 2,10 5,5 10,2 20,1
[INFO   ] def_ignore_cells =
[INFO   ] macro_obs_aware_layers =
[INFO   ] enable_place = true
[INFO   ] enable_legalize = true
[INFO   ] evaluate_overlap = false
[INFO   ] move_multi_row_cell = true
[INFO   ] align_power_line = true
[INFO   ] cluster_cell = false
[INFO   ] sort_nets_by_degree = false
[INFO   ] file_format = BOOKSHELF
[INFO   ] max_iters = 6
[WARNING] no DEF file specified
[WARNING] no Verilog file specified
[INFO   ] reading place_iop_pad0_ISPD11_mapped.aux
place_iop_pad0_ISPD11_mapped.scl:12.23-24: syntax error, unexpected string
[ERROR  ] Bookshelf file parsing failed: place_iop_pad0_ISPD11_mapped.aux
[ASSERT ] /DREAMPlace/dreamplace/ops/place_io/src/place_io.cpp:149: DreamPlace::PlaceDB DreamPlace::place_io_forward(const pybind11::list&): Assertion `flag' failed: failed to read input Bookshelf files
```

2、**Error with routability_opt_flag**:
Has anyone tried using DEF/LEF files as input with the `routability_opt_flag` enabled? I encountered the following error when I enabled the routability_opt_flag: When I debugged, I found that `PlanarDirectEnum::HORIZONTAL.empty()` is true and `unit_horizontal_capacities` is None. My design should contain routing information.
```
[WARNING] db.routingCapacity(PlanarDirectEnum::HORIZONTAL).empty(): 1 
[INFO   ] -- Converting Node Orientation --
[INFO   ] FN -> N: 2761 nodes
[INFO   ] FS -> N: 4555 nodes
[INFO   ] S -> N: 2680 nodes
[INFO   ] ---------------------------------
[INFO   ] -------- Area Statistics --------
[INFO   ] fixed area overlap = 3.13004e+12
[INFO   ] fixed area total = 4.80673e+12
[INFO   ] space area = 5.12362e+12
[INFO   ] ---------------------------------
[DEBUG  ] DREAMPlace - ==================================== core debug info =======================================
[DEBUG  ] DREAMPlace - num_routing_grids_x 0, num_routing_grids_y 0
[DEBUG  ] DREAMPlace - pydb.unit_horizontal_capacities []
[DEBUG  ] DREAMPlace - pydb.unit_vertical_capacities []
[DEBUG  ] DREAMPlace - enter else mode
[INFO   ] DREAMPlace - set shift_factor = (20140, 25200), as original row bbox = (20140, 25200, 2.89598e+06, 2.8952e+06)
[INFO   ] DREAMPlace - set scale_factor = 0.00263158, as site_width = 380
[INFO   ] DREAMPlace - shift coordinate system by (20140, 25200), scale coordinate system by 0.00263158
[INFO   ] DREAMPlace - 
================================= Benchmark Statistics =================================
#nodes = 155574, #terminals = 15110, # terminal_NIs = 495, #movable = 139969, #nets = 171092
die area = (0, 0, 7568, 7552.63) 5.71583e+07
row height = 7.36842, site width = 1
#pins = 574677, #movable_pins = 566394
total_movable_node_area = 7.94437e+06, total_fixed_node_area = 3.32876e+07, total_space_area = 3.54821e+07
utilization = 0.223898, target_density = 0.3
use placeable_area = 3.54821e+07 to compute fillers
total_filler_node_area = 2.70027e+06, #fillers = 49951, filler sizes = 7.33646x7.36842
num_bins = 512x512, bin sizes = 2.00603x2.00195
================================== routing information =================================
routing grids (800, 415)
routing grid sizes (9.46, 18.1991)
routing capacity H/V (909.956, 548.68) per tile
========================================================================================
[INFO   ] DREAMPlace - reading benchmark takes 7.27476 seconds
[INFO   ] DREAMPlace - reading database takes 7.27 seconds
[DEBUG  ] DREAMPlace - node_x = [2709.0764  4303.908   3570.9475  ...  -52.63158  -52.63158  -52.63158]
[DEBUG  ] DREAMPlace - prepare init_pos takes 0.01 seconds
[DEBUG  ] DREAMPlace - build pos takes 0.00 seconds
[INFO   ] DREAMPlace - unit_pin_capacity = 0.0293115
[WARNING] DREAMPlace - net weights are all the same, ignored
[DEBUG  ] DREAMPlace - build data_collections takes 0.17 seconds
[DEBUG  ] DREAMPlace - build op_collections takes 0.00 seconds
[DEBUG  ] DREAMPlace - build BasicPlace ops takes 0.00 seconds
[INFO   ] DREAMPlace - non-linear placement initialization takes 0.19 seconds
[INFO   ] DREAMPlace - Global placement: 512x512 bins by default
[INFO   ] DREAMPlace - 512x512 bins #bins 512x512, bin sizes 2.00603x2.00195, max_num_bins = 4, padding = 0
[INFO   ] dreamplace.ops.electric_potential.electric_potential - regard 0 cells as movable macros in global placement
[WARNING] DREAMPlace - quadratic density penalty is disabled, density weight update is forced to be based on HPWL
placedb.unit_horizontal_capacities: None
placedb.routing_grid_size_y: 18.199112238427393
placedb.unit_vertical_capacities: None
placedb.routing_grid_size_x: 9.46
Traceback (most recent call last):
  File "dreamplace/Placer.py", line 197, in <module>
    place(params)
  File "dreamplace/Placer.py", line 65, in place
    metrics = placer(params,  placedb)
  File "/DREAMPlace/dreamplace/NonLinearPlace.py", line 80, in __call__
    model = PlaceObj.PlaceObj(
  File "/DREAMPlace/dreamplace/PlaceObj.py", line 251, in __init__
    self.op_collections.nctugr_congestion_map_op = self.build_nctugr_congestion_map(
  File "/DREAMPlace/dreamplace/PlaceObj.py", line 1007, in build_nctugr_congestion_map
    placedb.unit_horizontal_capacities *
TypeError: unsupported operand type(s) for *: 'NoneType' and 'float'
```

Thank you for your time and assistance. I appreciate your guidance in resolving these issues.

limbo018, 2023-08-04:
For the second question, the routability opt mode is enabled for DAC 2012 benchmarks, which contains an additional *.route file to set routing capacity. You'd better check it out first.

ZhuLinsen, 2023-08-07:
@limbo018 Thank you, Dr.Lin. 
- The solution for the fist question is now available, and I have successfully run the code.
- Regarding for the second question, I utilize the same design and merely converted the format between Lef/Def and Bookshelf. Futhermore, the Bookshelf format works well,  but there seems to be an issue with the DEF/LEF format conversion, indicating that there might be some mistakes
- Once again, thank you for your response and suggestions.

limbo018, 2023-08-07:
Can you share your converted LEF/DEF files from DAC 2012 benchmarks?

mumu157, 2023-08-11:
Prof. LIn,
- I don't have the LEF/DEF files from DAC 2012 benchmarks. I only tested the `Ariane133` from def/lef to Bookshelf format, as shown in the attached file.
The previous question was: ``` I utilize the same design and merely converted the format between Lef/Def and Bookshelf. Futhermore, the Bookshelf format works well, but there seems to be an issue with the DEF/LEF format conversion, indicating that there might be some mistakes```

- Please allow me to ask another question:
In the .scl，the `Siteorient` property has two options. Should I set them both to 1?
```
  Siteorient     :    R0 
  Siteorient     :    MX
```
Thank you!

mumu157, 2023-08-11:
Of course, the files are in the 'question2' directory. You can find them at this [link](https://github.com/limbo018/DREAMPlace/files/12257981/designs.zip)
I have checked the code, and it appears that route information in the DEF/LEF files has not been verified. For example, the m_vRoutingCapacity field might be empty."

closed 2023-11-16 by mumu157