limbo018/DREAMPlace #150: FENCE Regions

udaymallappa, 2023-08-22:
I don't find examples for setting up FENCE region constraints. Can you help me ?

limbo018, 2023-08-23:
FENCE region is defined in DEF files. You may refer to ispd 2015 benchmarks for how to define region constraints. 

https://www.ispd.cc/contests/15/web/benchmark.html

udaymallappa, 2023-08-23:
Thanks for your quick response. I am using a bookshelf format for my testcase. Is there an easier way to define FENCE constraints, without having to using the DEF format ?

limbo018, 2023-08-23:
Bookshelf format does not support FENCE region. You can consider convert bookshelf to LEF/DEF. There was an old tool to do that. You might be able to find other tools as well. 

http://vlsicad.eecs.umich.edu/BK/PlaceUtils/

chiux, 2023-10-17:
Hi,

I would like to ask if DreamPlace support other types of region constraints such as GUIDE, DEFAULT defined in DEF.
Thanks!

Best regards,
Yang

Constwelve, 2023-11-09:
DREAMPLace does not support region constraints other than FENCE, we will support more region constraints in the future.