limbo018/DREAMPlace #152: meet error when making

Yixin-Gong, 2023-10-15:
```bash
[ 80%] Building CXX object dreamplace/ops/place_io/CMakeFiles/place_io_cpp.dir/src/PyPlaceDB.cpp.o
cd /home/eric/Desktop/DREAMPlace/build/dreamplace/ops/place_io && /usr/bin/c++ -DDRAWPLACE=1 -DENABLE_CUDA=0 -DTORCH_EXTENSION_NAME=place_io_cpp -DTORCH_VERSION_MAJOR=2 -DTORCH_VERSION_MINOR=1 -D_GLIBCXX_USE_CXX11_ABI=0 -Dplace_io_cpp_EXPORTS -I/home/eric/Desktop/DREAMPlace/thirdparty/flute -I/home/eric/Desktop/DREAMPlace/thirdparty/OpenTimer -I/home/eric/Desktop/DREAMPlace/dreamplace/ops/place_io/.. -I/home/eric/Desktop/DREAMPlace/thirdparty/Limbo -I/usr/include/cairo -isystem /usr/include/python3.10 -isystem /home/eric/Desktop/DREAMPlace/thirdparty/pybind11/include -isystem /home/eric/.local/lib/python3.10/site-packages/torch/include -isystem /home/eric/.local/lib/python3.10/site-packages/torch/include/torch/csrc/api/include -O3 -DNDEBUG -fPIC -fvisibility=hidden -flto -fno-fat-lto-objects -D_GLIBCXX_USE_CXX11_ABI=0 -fopenmp -std=gnu++17 -MD -MT dreamplace/ops/place_io/CMakeFiles/place_io_cpp.dir/src/PyPlaceDB.cpp.o -MF CMakeFiles/place_io_cpp.dir/src/PyPlaceDB.cpp.o.d -o CMakeFiles/place_io_cpp.dir/src/PyPlaceDB.cpp.o -c /home/eric/Desktop/DREAMPlace/dreamplace/ops/place_io/src/PyPlaceDB.cpp
In file included from /home/eric/.local/lib/python3.10/site-packages/torch/include/torch/csrc/jit/ir/ir.h:18,
                 from /home/eric/.local/lib/python3.10/site-packages/torch/include/torch/csrc/jit/api/function_impl.h:4,
                 from /home/eric/.local/lib/python3.10/site-packages/torch/include/torch/csrc/jit/api/method.h:7,
                 from /home/eric/.local/lib/python3.10/site-packages/torch/include/torch/csrc/jit/api/object.h:6,
                 from /home/eric/.local/lib/python3.10/site-packages/torch/include/torch/csrc/jit/api/module.h:4,
                 from /home/eric/.local/lib/python3.10/site-packages/torch/include/torch/csrc/api/include/torch/serialize/input-archive.h:6,
                 from /home/eric/.local/lib/python3.10/site-packages/torch/include/torch/csrc/api/include/torch/serialize/archive.h:3,
                 from /home/eric/.local/lib/python3.10/site-packages/torch/include/torch/csrc/api/include/torch/data/samplers/serialize.h:4,
                 from /home/eric/.local/lib/python3.10/site-packages/torch/include/torch/csrc/api/include/torch/data/samplers.h:8,
                 from /home/eric/.local/lib/python3.10/site-packages/torch/include/torch/csrc/api/include/torch/data/datasets/chunk.h:7,
                 from /home/eric/.local/lib/python3.10/site-packages/torch/include/torch/csrc/api/include/torch/data/datasets.h:4,
                 from /home/eric/.local/lib/python3.10/site-packages/torch/include/torch/csrc/api/include/torch/data.h:4,
                 from /home/eric/.local/lib/python3.10/site-packages/torch/include/torch/csrc/api/include/torch/all.h:9,
                 from /home/eric/.local/lib/python3.10/site-packages/torch/include/torch/extension.h:5,
                 from /home/eric/Desktop/DREAMPlace/dreamplace/ops/place_io/../utility/src/torch.h:13,
                 from /home/eric/Desktop/DREAMPlace/dreamplace/ops/place_io/src/PyPlaceDB.h:18,
                 from /home/eric/Desktop/DREAMPlace/dreamplace/ops/place_io/src/PyPlaceDB.cpp:8:
/home/eric/.local/lib/python3.10/site-packages/torch/include/ATen/core/interned_strings.h:354:1: error: macro "assert_msg" requires 2 arguments, but only 1 given
  354 | FORALL_NS_SYMBOLS(DEFINE_SYMBOL)
      | ^~~~~~~~~~~~~~~
In file included from /home/eric/Desktop/DREAMPlace/thirdparty/Limbo/limbo/programoptions/ProgramOptions.h:21,
                 from /home/eric/Desktop/DREAMPlace/dreamplace/ops/place_io/src/Params.h:14,
                 from /home/eric/Desktop/DREAMPlace/dreamplace/ops/place_io/src/PlaceDB.h:26,
                 from /home/eric/Desktop/DREAMPlace/dreamplace/ops/place_io/src/PyPlaceDB.h:16,
                 from /home/eric/Desktop/DREAMPlace/dreamplace/ops/place_io/src/PyPlaceDB.cpp:8:
/home/eric/Desktop/DREAMPlace/thirdparty/Limbo/limbo/preprocessor/AssertMsg.h:34: note: macro "assert_msg" defined here
   34 | #define assert_msg(condition, message) \
      | 
make[2]: *** [dreamplace/ops/place_io/CMakeFiles/place_io_cpp.dir/build.make:233: dreamplace/ops/place_io/CMakeFiles/place_io_cpp.dir/src/PyPlaceDB.cpp.o] Error 1
make[2]: Leaving directory '/home/eric/Desktop/DREAMPlace/build'
make[1]: *** [CMakeFiles/Makefile2:3802: dreamplace/ops/place_io/CMakeFiles/place_io_cpp.dir/all] Error 2
make[1]: Leaving directory '/home/eric/Desktop/DREAMPlace/build'
make: *** [Makefile:139: all] Error 2
```

limbo018, 2023-10-15:
Hi, this problem with PyTorch 2.0 or later has been resolved in the ```develop``` branch.

Yixin-Gong, 2023-10-15:
OK,Thanks.

closed 2023-10-17 by Yixin-Gong