limbo018/DREAMPlace #153: run without gpu

mavoyang, 2023-10-17:
How can I configure CMAKE to make Dreamplacer run without GPU???

To work with other projects, I need to build version 2.2.0 of Dreamplacer. 
I first built the Docker container, and I can find Cuda 10.0 in it.
Then, if I run the Docker container with GPU, I can successfully 'make'. But running any benchmark with 'placer.py' gets stuck at '[INFO] DREAMPlace - move cells to the center of the layout with random noise.'
[image]

This may be due to the incompatibility between the old version of Dreamplacer and the current CUDA.
So, I tried running the Docker container without GPU, but running any benchmark with 'placer.py' results in an error 'Found no NVIDIA driver on your system' after 'move cells to the center of the layout with random noise.'
[image]

How should I configure CMake to build a Dreamplacer that doesn't use CUDA at all?" Or how can I fix it to run with Cuda?

One more thing, for the newest Dreamplacer, it runs with GPU well when I use the provided docker.

limbo018, 2023-10-17:
You can modify the ```gpu``` parameter in the ```JSON``` input file to 0 for CPU.

mavoyang, 2023-10-17:
Ha, it's really a rookie question. Thanks!

closed 2023-10-17 by mavoyang