limbo018/DREAMPlace #155: Question about minimum wirelength

Icamd, 2023-11-04:
Dear professor:
I have another question about DREAMPlace. Is it possible to adjust some parameters in order to achieve optimal HPWL with ISPD05 mms benchmarks? (For example, by adding weights to wirelength or increase the gradient descent iteration number? Or should I set some other macros to be movable in mms benchmarks to shorten the HPWL?) Thank you!

Icamd, 2023-11-07:
And could you please tell me how to run ICCAD04 dataset using DREAMPlace? Thank you!

Constwelve, 2023-11-07:
> Dear professor Lin: Thank you for your code! I have a question about DREAMPlace wirelength. Is it possible to adjust some parameters in order to achieve optimal HPWL with ISPD05 mms benchmarks? (For example, by adding weights to wirelength or increase the gradient descent iteration number? Or should I set some other macros to be movable in mms benchmarks to shorten the HPWL?) Thank you!
By adjusting the parameters, a better wirelength can be achieved. You can refer to NVIDIA's work AutoDMP. They use a multi-objective Bayesian optimizer to adjust the input parameters, resulting in better macro placement.

Constwelve, 2023-11-07:
By adjusting the parameters, a better wirelength can be achieved. You can refer to NVIDIA's work AutoDMP. They use a multi-objective Bayesian optimizer to adjust input parameters, resulting in better macro placement.

> Dear professor Lin: Thank you for your code! I have a question about DREAMPlace wirelength. Is it possible to adjust some parameters in order to achieve optimal HPWL with ISPD05 mms benchmarks? (For [...]

Constwelve, 2023-11-07:
For Bookshelf format, you can refer to the .json files in test/ispd2005. To run your design, replace the aux_input  in .json to yours like  
```
"aux_input" : "path/to/your/aux_file", 
``` 

Similarly, for LEF/DEF format, you can refer to the .json files in test/ispd2015/lefdef and replace the lef_input, def_input  in .json to yours like  
```
"lef_input" : ["path/to/your/lef_file_0", "path/to/your/lef_file_0", ...],
"def_input" : "path/to/your/def_file",
```

> And could you please tell me how to run ICCAD04 dataset using DREAMPlace? Thank you!

Icamd, 2023-11-07:
> For Bookshelf format, you can refer to the .json files in test/ispd2005. To run your design, replace the aux_input in .json to yours like
> 
> ```
> "aux_input" : "path/to/your/aux_file", 
> ```
> 
> Similarly, for LEF/DEF format, you can refer to the .json files in test/ispd2015/lefdef and replace the lef_input, def_input in .json to yours like
> 
> ```
> "lef_input" : ["path/to/your/lef_file_0", "path/to/your/lef_file_1", ...],
> "def_input" : "path/to/your/def_file",
> ```
> 
> > And could you please tell me how to run ICCAD04 dataset using DREAMPlace? Thank you!

Thank you, but there is still a problem with ICCAD04 datasets. An error reports that"failed to read input Bookshelf files":
`[INFO   ] reading /home/proj/EDA-AI-main/benchmarks/ICCAD04/ibm03/ibm03.aux
/home/proj/EDA-AI-main/benchmarks/ICCAD04/ibm03/ibm03.nets:32.1: syntax error, unexpected end of line, expecting ':'
[ERROR  ] Bookshelf file parsing failed: /home/proj/EDA-AI-main/benchmarks/ICCAD04/ibm03/ibm03.aux
[ASSERT ] /home/proj/EDA-AI-main/DREAMPlace/dreamplace/ops/place_io/src/place_io.cpp:149: DreamPlace::PlaceDB DreamPlace::place_io_forward(const pybind11::list&): Assertion `flag' failed: failed to read input Bookshelf files
Aborted (core dumped)`
Instead of replacing the "aux_input", do you have the json file for ICCAD04 dataset(such as ibm01.json, ibm02.json, ......, ibm18.json)? Thank you!

Constwelve, 2023-11-07:
You are right. I find our parser crashed when parsering iccad2004 benchmark. 
For bookshelf format, there are some pins without offsets, like `p 198` in the following `net 2`
```
NetDegree : 2  net0
      p198 B 
     a8117 B  : 48.7308 -40
```
I try to modify it to 
```
NetDegree : 2  net0
      p198 B  : 0 0
     a8117 B  : 48.7308 -40
```
Also, I replace the `1.77636e-15` to `0`, because I find our parser cannot recognize scientific notation.
Besides, in the `.wts` file, the weights are adding to nodes, but it should be adding to nets, so I remove all weights given in `.wts`.
And now DREAMPlace can read it.
```
[INFO   ] write SolutionFileFormat.BOOKSHELF takes 0.007 seconds
[INFO   ] writing placement to results/ibm01/plot/iter0466.png
[INFO   ] plotting to results/ibm01/plot/iter0466.png takes 0.228 seconds
[INFO   ] non-linear placement takes 9.94 seconds
[INFO   ] writing to results/ibm01/ibm01.nonlinear.pl
[INFO   ] writing placement to results/ibm01/ibm01.nonlinear.pl
[INFO   ] write SolutionFileFormat.BOOKSHELF takes 0.007 seconds
[WARNING] External detailed placement engine thirdparty/ntuplace3 or aux file NOT found
[INFO   ] placement takes 11.751 seconds
```

For LEF/DEF format, I haven't been able to fix it yet.

> Thank you, but there is still a problem with ICCAD04 datasets. An error reports that"failed to read input Bookshelf files": `[INFO ] reading /home/proj/EDA-AI-main/benchmarks/ICCAD04/ibm03/ibm03.aux [...]

Icamd, 2023-11-08:
Thank you! And do you know how to set the DREAMPlace to start optimization of movable macros from other initial positions, instead of optimizing all movable macros from the center of the chip canvas? @Constwelve

Constwelve, 2023-11-08:
To use the position given in .pl as initial position, you can set the `random_center_init_flag` to 0.

Icamd, 2023-11-08:
Thank you! Does the random_center_init_flag includes the initial position of both macros and standard cells?

> To use the position given in .pl as initial position, you can set the `random_center_init_flag` to 0.

Constwelve, 2023-11-08:
Yes.

Icamd, 2023-11-08:
Thank you! I wish you all the best in your research! > Yes.

Icamd, 2023-11-09:
Sorry to bother you again, I have another problem about DREAMPlace. When I set the random_center_init_flag to 0 and optimize the placement base on previous results in ISPD05 mms, I notice that the HPWL has a large increase(sometimes the HPWL even doubles when I optimize the placement base on previous results such as adaptec02.) For example, the initial HPWL result of adaptec02 is around 7.7*10^7, but when I optimize the placement base on this initial result, the HPWL reaches above 1.25*10^8, and only output 4 PNG(iter0000.png, iter0004.png, iter0005.png, iter0006.png) Do you know the reason why the HPWL increases when start optimizing from previous results? Thank you!

Constwelve, 2023-11-09:
I guess the placer may mistakenly detect divergence and stop the global placement. Could you provide the log?

Icamd, 2023-11-09:
Of course, in the first file, I use the DREAMPlace to run the adaptec2-mms dataset, and the random_center_init_flag=1, and the initial result is around 7.7e7; In the second file, I just replace the adaptec2.pl with the initial result "adaptec2.gp.pl" and set the random_center_init_flag to 0. You can see that in line 213, the HPWL is equal to the initial result, but in line 278 the HPWL reaches 1.65e8. Also, the iter0000.png and iter0006.png don't look very different, which shouldn't leads to such a large gap of HPWL between two results. Thank you! @Constwelve 
[adaptec2-mms-initial result-HPWL=7.7e7.txt](https://github.com/limbo018/DREAMPlace/files/13305726/adaptec2-mms-initial.result-HPWL.7.7e7.txt)
[adaptec2-mms-second result-HPWL=1.25e8.txt](https://github.com/limbo018/DREAMPlace/files/13305729/adaptec2-mms-second.result-HPWL.1.25e8.txt)
[image]
[image]

Constwelve, 2023-11-09:
The placer mistakenly detected divergence because the overflow of initial placement is 0,
```
[INFO   ] DREAMPlace - full step 18.218 ms
[WARNING] DREAMPlace - Divergence detected: overflow increases too much than best overflow (8.5199 > 0.0022)
[ERROR  ] DREAMPlace - possible DIVERGENCE detected, roll back to the best position recorded
[INFO   ] DREAMPlace - optimizer step 16.930 ms
``` 
You can give a larger threshold or turn off the divergence check at https://github.com/limbo018/DREAMPlace/blob/f59ebfa4831573d3db38642f4b2d2e2108bd6e3e/dreamplace/NonLinearPlace.py#L477C1-L477C1

Icamd, 2023-11-09:
Thank you! I set a larger threshold and the problem is solved: ) But I still have another question: when I delete all "/FIXED" signs for macros in dataset, the program meets another error "Segmentation fault (core dumped)". Do you know what is causing this problem?(There appears to be no overflow with GPU memory) Here is the log of adaptec3_mms, I removed all the \FIXED sign in adaptec3.pl, and set the random_center_init_flag to 0 so that I can optimize from previous placement results. But the program ends unexpectedly. Thank you again!
[Segmentation_fault.txt](https://github.com/limbo018/DREAMPlace/files/13308205/Segmentation_fault.txt)

Constwelve, 2023-11-11:
If you use MMS benchmark, the /FIXED nodes are IO pads and their sizes are all set to be (0, 0). Thus, there is no need to delete /FIXED and nodes with (0, 0) sizes may result in errors in detailed placement.

Icamd, 2023-11-11:
Thank you! But, in the event that these nodes are able to move, is it possible to achieve a lower HPWL?
> If you use MMS benchmark, the /FIXED nodes are IO pads and their sizes are all set to be (0, 0). Thus, there is no need to delete /FIXED and movable nodes with (0, 0) sizes may result in errors in detailed placement.

Constwelve, 2023-11-11:
Yes. But MMS is not designed for IO pads placement, IO pads have non-zero sizes in original ISPD2005 benchmark. So if you want to make IO pads movable, you can delete all /FIXED in ISPD2005 benchmark. We've tried that DREAMPlace won't report Segmentation fault in this case.

Icamd, 2023-11-11:
Wow! Thank you, thank you! My problems are solved! I wish you all the best in your research!
> Yes. But MMS is not designed for IO pads placement, IO pads have non-zero sizes in original ISPD2005 benchmark. So if you want to make IO pads movable, you can delete all /FIXED in ISPD2005 benchmark. We've tried that DREAMPlace won't report Segmentation fault in this case.

closed 2023-11-11 by Icamd