limbo018/DREAMPlace #158: Question about the IO pins parsed by the LEF/DEF parser

ZhuLinsen, 2023-11-16:
Dear Professor, 
I hvae two question about IO pins(num_terminal_NIs) parsed by LEF/DEF parser. Below is a sample PIN from DEF file:
```
PINS 54 ;
    - clk + NET clk + DIRECTION INPUT + USE SIGNAL
      + PORT
        + LAYER metal5 ( -140 -140 ) ( 140 140 )
        + PLACED ( 140 519820 ) N ;
```

1. In the DEF file, IO pins have a size, and the size of `clk `above is 280 * 280 . Howerve, why are the IO pins sizes all 0 in `node_x `and `node_y` (PyPlaceDB.cpp)?
2. Why use the center coordinates rather than left-lower coordinates to represent the location of IO pins?

If there are any bugs, I would appreciate the opportunity to fix them.

limbo018, 2023-11-16:
Thank you for pointing this out. IO pins are considered as terminal NI, which do not impose any overlap constraints to placement. Originally, I simply set the sizes to 0 after reading them. If you need the information of their sizes, I commented out the resizing code. 
https://github.com/limbo018/DREAMPlace/blob/5d655047329e7b106f2befdbbcf023b687c3bfd6/dreamplace/ops/place_io/src/PlaceDB.cpp#L507

ZhuLinsen, 2023-11-16:
Thank you for your response.I comprehend.

closed 2023-11-16 by ZhuLinsen