limbo018/DREAMPlace #161: Segmentation Fault when enabling timing_opt in 4.0

xiaosinju, 2023-11-29:
I am encountering the same issue across multiple deployed machines by the following line: 

`python dreamplace/Placer.py test/iccad2015.ot/superblue3.json`

A segmentation fault occurs with the `malloc` function on line 97 in `flute.cpp`. When we increase the size of the space allocated by `malloc`, this error disappears, only to be replaced by an erroneous reading of the `kk` value, 182934, on lines 91-92, exceeding the upper limit of `MGROUP`. This issue has a chance of not occurring randomly, but another segmentation fault (which has not yet been precisely located) will appear in `timing_cpp.cpp`. 

Such issue is really abnormal for the `flute` module is independent, and if the issue is a common one, it should be encountered by anyone build and enable the `timing_opt` function.

enzoleo, 2023-11-29:
The `flute` module has been slightly changed and merged to work with our timer, but the core should remain the same (changing the C code itself is hard and unnecessary). So, I do not really think the problem comes from the source code of `flute`. However, you mentioned this issue occurred across multiple deployed machines, which seems to tell us there exists a bug :thinking:. That is confusing.

Since nobody has ever encountered this issue as I remember, and I can't really reproduce it on my machines, would you please attach your system info? Also, does it only occur on `superblue3` or every case? I tend to believe this is more likely to be related to machines.

xiaosinju, 2023-11-29:
The issue has been resolved! It appears that the problem was caused by a confusion in the deployment of different versions of the source code. We used the version 4.0.0 of the code released by Professor Lin, recompiled and ran it, and everything is running smoothly! 

I will close this issue now. Thank you so much for your help!

closed 2023-11-29 by xiaosinju