limbo018/DREAMPlace #163: Error when I build the project

zhubenchao, 2023-12-21:
Error Info:
/usr/local/cuda-12/bin/nvcc /home/dev/Open-Source/DREAMPlace/dreamplace/ops/pin_pos/src/pin_pos_cuda_segment_kernel.cu -c -o /home/dev/Open-Source/DREAMPlace/build/dreamplace/ops/pin_pos/CMakeFiles/pin_pos_cuda_segment_cuda_tmp.dir/src/./pin_pos_cuda_segment_cuda_tmp_generated_pin_pos_cuda_segment_kernel.cu.o -m64 -DTORCH_EXTENSION_NAME=pin_pos_cuda_segment -DTORCH_MAJOR_VERSION= -DTORCH_MINOR_VERSION= -DENABLE_CUDA=1 -D_GLIBCXX_USE_CXX11_ABI=1 -Xcompiler ,"-O3","-DNDEBUG" -gencode arch=compute_60,code=sm_60 -gencode arch=compute_61,code=sm_61 -gencode arch=compute_70,code=sm_70 -gencode arch=compute_75,code=sm_75 -gencode arch=compute_80,code=sm_80 --compiler-options -fPIC -DNVCC -I/usr/local/cuda-12/include -I/home/dev/Open-Source/DREAMPlace/thirdparty/flute -I/home/dev/Open-Source/DREAMPlace/thirdparty/OpenTimer -I/home/dev/Open-Source/DREAMPlace/dreamplace/ops/pin_pos/… -I/home/dev/miniconda3/envs/pytorch/lib/python3.8/site-packages/torch/include -I/home/dev/miniconda3/envs/pytorch/lib/python3.8/site-packages/torch/include/torch/csrc/api/include -I/home/dev/Open-Source/DREAMPlace/thirdparty/pybind11/include -I/home/dev/miniconda3/envs/pytorch/include/python3.8
/usr/local/cuda-12/include/cub/agent/agent_three_way_partition.cuh(78): error: name followed by “::” must be a class or namespace name
struct accumulator_pack_base_t<OffsetT, typename cuda::std::enable_if<sizeof(OffsetT) == 4>::type>
^

/usr/local/cuda-12/include/cub/agent/agent_three_way_partition.cuh(78): error: expected an identifier
struct accumulator_pack_base_t<OffsetT, typename cuda::std::enable_if<sizeof(OffsetT) == 4>::type>
^

/usr/local/cuda-12/include/cub/agent/agent_three_way_partition.cuh(78): error: expected a “>”
struct accumulator_pack_base_t<OffsetT, typename cuda::std::enable_if<sizeof(OffsetT) == 4>::type>
^

/usr/local/cuda-12/include/cub/agent/agent_three_way_partition.cuh(78): error: expected a “;”
struct accumulator_pack_base_t<OffsetT, typename cuda::std::enable_if<sizeof(OffsetT) == 4>::type>

I have tried gcc 7.5, 8.5, 9.5, and 10.5, but none of them work. Could you give some tips if you encountered a similar error?

limbo018, 2023-12-21:
It is likely that your CUDA version is too high. The widely tested version is CUDA 11.

zhubenchao, 2023-12-22:
Thank you for your reply; I have made it work, and I found that if I simply replace
#include "utility/src/namespace.h"

// include cub in a safe manner
#define CUB_NS_PREFIX namespace DREAMPLACE_NAMESPACE {
#define CUB_NS_POSTFIX }
#define CUB_NS_QUALIFIER DREAMPLACE_NAMESPACE::cub
#include "cub/cub.cuh"
#undef CUB_NS_QUALIFIER
#undef CUB_NS_POSTFIX
#undef CUB_NS_PREFIX

#endif
by #include<cub/cub.h>, then it works. So I guess the #define may conflict with cuda-12/cub.

closed 2023-12-22 by zhubenchao

Icamd, 2024-01-18:
> Thank you for your reply; I have made it work, and I found that if I simply replace #include "utility/src/namespace.h"
>
> // include cub in a safe manner #define CUB_NS_PREFIX namespace [...]

@zhubenchao I have the same problem, do you mean you replaced "#include "cub/cub.cuh"" with "#include<cub/cub.h>"? Thank you!

yihuajack, 2024-05-22:
> @zhubenchao I have the same problem, do you mean you replaced "#include "cub/cub.cuh"" with "#include<cub/cub.h>"? Thank you!

You need to replace all the code Blanchard-Zhu listed with #include<cub/cub.h>

lfy2947590638, 2025-02-02:
> Thank you for your reply; I have made it work, and I found that if I simply replace #include "utility/src/namespace.h"
>
> // include cub in a safe manner #define CUB_NS_PREFIX namespace [...]

Hi! Could you please tell me where did you do the replacement? Thanks!

limbo018, 2025-02-03:
Do you mean that you want to turn off the random initialization? There is a parameter in json called ```random_center_init_flag``` that you can turn off to disable the initialization.

lfy2947590638, 2025-02-03:
> Hi! Could you please tell me where did you do the replacement? Thanks!

Great! I have found it! DREAMPlace/dreamplace/ops/utility/src/utils_cub.cuh