limbo018/DREAMPlace #170: Error while building

IshraqAtUCF, 2024-03-04:
For the current commit `d510848` it is giving me an error while building. 

[image]

Just to reiterate my building process:

I initialized the submodules and directly tried to build without docker. 

It completes the cmake but gives error while I try to `make`.

limbo018, 2024-03-04:
What's your gcc/g++ version?

limbo018, 2024-03-04:
Btw, you can skip OpenTimer's unittest, as it is not used.

IshraqAtUCF, 2024-03-04:
It is 11.4

[image]

Also, how to skip the OpenTimer's unittest?

IshraqAtUCF, 2024-03-05:
Is this because of higher version of gcc/g++? @limbo018

closed 2024-03-23 by IshraqAtUCF