limbo018/DREAMPlace #171: dreamplace4.0  iccad2015的case

huichengyu, 2024-03-05:
In the case of ICCAD2015, this situation has appeared, what to do, is there a big guy to help solve it
[INFO   ] DREAMPlace - 
================================= Benchmark Statistics =================================
#nodes = 1219735, #terminals = 52488, # terminal_NIs = 6482, #movable = 1160765, #nets = 1224979
die area = (0, 0, 51092, 16560) 8.46084e+08
row height = 9, site width = 1
#pins = 3905321, #movable_pins = 3736022
total_movable_node_area = 1.73703e+08, total_fixed_node_area = 4.76021e+08, total_space_area = 3.79436e+08
utilization = 0.457791, target_density = 1
use placeable_area = 3.79436e+08 to compute fillers
total_filler_node_area = 2.05734e+08, #fillers = 1477782, filler sizes = 15.4687x9
num_bins = 2048x2048, bin sizes = 2.77192x0.898438
========================================================================================
[INFO   ] DREAMPlace - reading benchmark takes 50.7278 seconds
[INFO   ] DREAMPlace - reading database takes 50.73 seconds
[INFO   ] DREAMPlace - reading timer constraints takes 0.00230169 seconds
I 63232 24-03-05 17:15:04 celllib.cpp:34] loading celllib "benchmarks/iccad2015.ot/superblue3/superblue3_Early.lib"
I 63232 24-03-05 17:15:04 celllib.cpp:34] loading celllib "benchmarks/iccad2015.ot/superblue3/superblue3_Late.lib"
I 59136 24-03-05 17:15:04 verilog.cpp:14] loading netlist "benchmarks/iccad2015.ot/superblue3/superblue3.v"
Aborted

enzoleo, 2024-03-05:
The error comes from timer. It seems that the Verilog parser has encountered some problems. You can check the function `ot::vlog::read_verilog` in `thirdparty/OpenTimer/ot/verilog/verilog.cpp`, as reported in the log. I tend to believe the input Verilog file is incorrect.

nineight908, 2024-03-07:
hi bro，Can you share your ICCAD2015 benchmarks？

huichengyu, 2024-03-07:
@enzoleo The problem has been solved, I copied the dreamplace directly from the server, and when I recompile it, it works. Finding out is verilog.cpp problem. Thank you so much.

huichengyu, 2024-03-07:
@nineight908 I'm under the URL from the folder inside, and there is no problem with https://drive.google.com/file/d/1xeauwLR9lOxnYvsK2JGPSY0INQh8VuE4/view?usp=sharing

enzoleo, 2024-03-07:
@huichengyu Glad to hear that.