limbo018/DREAMPlace #174: IndexError: _Map_base::at

qkr-xoals, 2024-03-19:
[image]
It works fine with bookshelf format, but this error always occurs with the circuits I synthesized.

limbo018, 2024-03-20:
Can you share the input files that cause the issue?

qkr-xoals, 2024-03-20:
> Can you share the input files that cause the issue?

Thank you for reply. Here are the input files I used.
[input files.zip](https://github.com/limbo018/DREAMPlace/files/14661259/input.files.zip)

limbo018, 2024-03-20:
Hi, I found the problem comes from your DEF file. The site name ```unit``` is not defined in the tech LEF. 

[image]

You need to replace it with the correct site name ```FreePDK45_38x28_10R_NP_162NW_34O``` defined in the tech LEF. 

[image]

I made a new commit (https://github.com/limbo018/DREAMPlace/commit/ef3f00748f0fedb51f6083770102ea1f1f36aa2d) in the ```develop``` branch to make the error message more readable.

qkr-xoals, 2024-03-20:
> Hi, I found the problem comes from your DEF file. The site name `unit` is not defined in the tech LEF.
> 
> <img alt="image" width="833" src="https://private-user-images.githubusercontent.com/6440093/314383522-6861f882-37ba-4a0e-bb04-5d862f5aefed.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA5MTI1MDcsIm5iZiI6MTcxMDkxMjIwNywicGF0aCI6Ii82NDQwMDkzLzMxNDM4MzUyMi02ODYxZjg4Mi0zN2JhLTRhMGUtYmIwNC01ZDg2MmY1YWVmZWQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDMyMCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAzMjBUMDUyMzI3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MTZkZTZlZTMyMWM5ZGZmYjgwZDRhNjQzYzEzMDI0NjJjMWE0MzdhYTNmMDQyZGMyZTRhYzRkODNhZjk1NDhiYSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.1BbcmHle5v8JKvVOdeX_dCJ0N4-UWXfHiZpOI2j1cz8">
> You need to replace it with the correct site name `FreePDK45_38x28_10R_NP_162NW_34O` defined in the tech LEF.
> 
> <img alt="image" width="1229" src="https://private-user-images.githubusercontent.com/6440093/314383624-9e7831ef-99ce-42d5-b4ca-46be52511e75.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA5MTI1MDcsIm5iZiI6MTcxMDkxMjIwNywicGF0aCI6Ii82NDQwMDkzLzMxNDM4MzYyNC05ZTc4MzFlZi05OWNlLTQyZDUtYjRjYS00NmJlNTI1MTFlNzUucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDMyMCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAzMjBUMDUyMzI3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MTY1OTVjOTcwM2RkMmE3MDc4MDA0YmE2MWY4YmM1ZGI2MmMwMzMxYjBiMDkyODZlMDUwM2Q1ODlhZmQ1NzVhYSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.YbCRLlygfTAohCCUSmPeJcPMHusV4xvwTdMlbDx6pFc">
> I made a new commit ([ef3f007](https://github.com/limbo018/DREAMPlace/commit/ef3f00748f0fedb51f6083770102ea1f1f36aa2d)) in the `develop` branch to make the error message more readable.

I have been greatly inspired by your papers. Thank you so much for your help despite your busy schedule!