limbo018/DREAMPlace #177: Compared to dc/invc/openroad upon my cases, dreaplace has the shock congestion Morphologically

tigereatsheep, 2024-04-18:
dcg: 0.+% H + 0.+% V
innovus: 1.+% H + 1.+% V
OR: 22.+% H + 33.+% V
dreamplace: 58+% H + 38.+% V

and dreaplace' routing length is 4x dcg's result.

I think here maybe hpwl is not a good cost function?

limbo018, 2024-04-18:
Hi, by default, dreamplace does not turn on routability optimization. It looks like your case needs heavy optimization on routability.

tigereatsheep, 2024-04-18:
> Hi, by default, dreamplace does not turn on routability optimization. It looks like your case needs heavy optimization on routability.

You are right, I'm focusing on solving congestion problem in EPlace framework. Do you know better WL estimation method (better than LSE/WA)？ By the way, DreamPlace is really fast.

limbo018, 2024-04-18:
You may check section III.F in the TCAD version of DREAMPlace paper (https://yibolin.com/publications/papers/PLACE_TCAD2020_Lin.pdf). It explains a typical way to do routability optimization. The default flow of DREAMPlace does not turn on routability optimization though.

tigereatsheep, 2024-04-19:
> You may check section III.F in the TCAD version of DREAMPlace paper (https://yibolin.com/publications/papers/PLACE_TCAD2020_Lin.pdf). It explains a typical way to do routability optimization. The default flow of DREAMPlace does not turn on routability optimization though.

Thank you for your help. I would working on this problem for several weeks