limbo018/DREAMPlace #179: Legality check fails after additional sta

levy-deng, 2024-05-16:
Hi Professor Lin,

After running this on the adaptec4 benchmark with the default json settings, I'm facing some strange behavior. I'm experiencing the issues that after the legalization step, Dreamplace decides to proceed with local optimization which fails legality checks.

```
[DEBUG  ] maximum grid spacing 3.40282e+38x3.40282e+38, equivalent to 0x0 bins
[DEBUG  ] Construct 2067x1253 Hannan grids, diamond search sequence 785005
[INFO   ] Legalize movable macros with linear programming on constraint graphs
[DEBUG  ] source 1329, terminal 1330
[DEBUG  ] Original slack: TNS[X/Y] = 0/0, WNS[X/Y] = 0/0
[DEBUG  ] Adjusted slack: TNS[X/Y] = 0/0, WNS[X/Y] = 0/0
[INFO   ] Macro displacement total 0, max 0, weighted total 0, max 0
[DEBUG  ] Macro legality check PASSED
[INFO   ] Align macros to site and rows
[DEBUG  ] Macro legality check PASSED
[INFO   ] Macro legalization takes 1803.82 ms
[DEBUG  ] Standard cell legalization blank_num_bins_y = 1944
[DEBUG  ] Standard cell legalization iteration 0 with 1x1 bins
[DEBUG  ] Standard cell legalization #bin_blanks
[DEBUG  ] #bin_objs = 33016
[INFO   ] Standard cell legalization legalizeBin takes 442.112 ms
[DEBUG  ] Standard cell legalization num_unplaced_cells = 0
[DEBUG  ] Standard cell legalization blank_num_bins_y = 1944
[DEBUG  ] Standard cell legalization iteration 0 with 1x1 bins
[DEBUG  ] Standard cell legalization #bin_blanks
[DEBUG  ] #bin_objs = 33016
[INFO   ] Standard cell legalization legalizeBin takes 413.824 ms
[DEBUG  ] Standard cell legalization num_unplaced_cells = 0
[INFO   ] Greedy legalization takes 1006.6 ms
[INFO   ] Legality check takes 38.7335 ms
[DEBUG  ] average displace = 27.1563
[INFO   ] Abacus legalization takes 77.6752 ms
[INFO   ] Legality check takes 38.7362 ms
[INFO   ] DREAMPlace - legalization takes 2.972 seconds
[INFO   ] DREAMPlace - additional sta after legalization
[INFO   ] DREAMPlace - iteration  889, wHPWL 1.382333E+08, time 4.713ms
[INFO   ] DREAMPlace - Start ABCDPlace for refinement
[INFO   ] Legality check takes 40.3756 ms
[INFO   ] DREAMPlace - ABCDPlace input legal flag = 1
[DEBUG  ] 4-reorder
[DEBUG  ] compute_independent_rows takes 18.6816 ms
[INFO   ] initial hpwl = 138234080.000
[INFO   ] Iter 0 time (ms)       922.768
[INFO   ] iteration 0: hpwl 138234080.000 => 136755040.000 (imp. 1.06995%)
[INFO   ] Iter 1 time (ms)       947.715
[INFO   ] iteration 1: hpwl 138234080.000 => 136480896.000 (imp. 1.26827%)
[INFO   ] kernel         time (ms)       runs
[INFO   ] compute_reorder_hpwl   8574.5          11595600        0.000739462
[INFO   ] apply_reorder          26.6447         483150          5.51478e-05
[INFO   ] K-reorder time: 1922.07 ms
[INFO   ] Legality check takes 39.9544 ms
[INFO   ] DREAMPlace - K-Reorder legal flag = 1
[INFO   ] diamond search grid size 64, sequence length 2048
[INFO   ] initial hpwl 1.36481e+08
[DEBUG  ] maximal_independent_set_sequential
[DEBUG  ] 1275 sets, average set size 111, max set size 128
[INFO   ] iteration 0, target hpwl 1.35852e+08, delta -629328(-0.461111%), solved 1275 sets, moved 8.1793% cells, runtime 211.622 ms
[DEBUG  ] maximal_independent_set_sequential
[DEBUG  ] 1089 sets, average set size 110, max set size 128
[DEBUG  ] maximal_independent_set_sequential
[DEBUG  ] 1170 sets, average set size 109, max set size 128
[DEBUG  ] maximal_independent_set_sequential
[DEBUG  ] 1128 sets, average set size 108, max set size 128
[DEBUG  ] maximal_independent_set_sequential
[DEBUG  ] 1150 sets, average set size 110, max set size 128
[DEBUG  ] maximal_independent_set_sequential
[DEBUG  ] 1116 sets, average set size 109, max set size 128
[INFO   ] iteration 5, target hpwl 1.34739e+08, delta -1.74173e+06(-1.27617%), solved 1116 sets, moved 17.5311% cells, runtime 202.108 ms
[DEBUG  ] maximal_independent_set_sequential
[DEBUG  ] 1155 sets, average set size 109, max set size 128
[... 207 lines cut ...]
"scale_factor" : 1.0,
"ignore_net_degree" : 100,
"enable_fillers" : 1,
"gp_noise_ratio" : 0.025,
"macro_place_flag" : 1,
"macro_halo_x": 10,
"macro_halo_y": 10,
"global_place_flag" : 1,
"legalize_flag" : 1,
"detailed_place_flag" : 1,
"detailed_place_engine" : "", 
"detailed_place_command" : "", 
"stop_overflow" : 0.07, 
"dtype" : "float32", 
"plot_flag" : 0, 
"random_center_init_flag" : 1,
"sort_nets_by_degree" : 0,
"num_threads" : 8,
"deterministic_flag" : 0,
"use_bb": 1
}
```
Important things to note is that, I'm using the CPU purely.

Additionally, Using the default config for adaptec4 seems to yield whpwl of around 1.8, which is very different from the numbers shown in the table. I'm wondering if I've misconfigured the tool?

limbo018, 2024-05-20:
Hi, the table is for macro placement, where you need to modify the input files by relaxing the fixed nodes to movable. Since you have more freedom to move nodes, the HPWL is smaller. You can download the benchmarks using the link in the readme. 
[image]

For the second run with timing optimization, I don't know how you enable it, as ispd2005 benchmarks do not provide timing library, which should contain delay information. Therefore, it does not support timing optimization.

enzoleo, 2024-05-20:
@limbo018 Probably the report of that additional sta in [Line 768](https://github.com/limbo018/DREAMPlace/blob/1a7d5853c407294c68a1e703715afc0198106a27/dreamplace/NonLinearPlace.py#L768C2-L768C62) should appear after the check of `timing_opt_flag`. The current version will report this line every time after legalization even if the timing optimization is not enabled.

limbo018, 2024-05-20:
> Line 768

Good point. Fixed.