limbo018/DREAMPlace #182: About dac2012

Mayone0, 2024-06-18:
Hi Professor Lin,
When I tried to place DAC-2012 benchmark suite I met some problems, 
[image]
I know it is NCTUgr crashed on my machine.But the file generated by NCTUgr(superblue2.NCTUgr )is not what I want(superblue2.NCTUgr.ofinfo).I'm still having this issue when I have NCTUgr running normally on the server
I don't know what went wrong and look forward to your answer

limbo018, 2024-06-20:
Please check the following issue. 
https://github.com/limbo018/DREAMPlace/issues/54

It is likely that NCTUgr is not compatible with your machine/environment.

Mayone0, 2024-06-20:
> Please check the following issue. #54
> 
> It is likely that NCTUgr is not compatible with your machine/environment.

Thanks for the reply, I probably know what the problem is.

Redjade123, 2025-03-27:
> > Please check the following issue. [#54](https://github.com/limbo018/DREAMPlace/issues/54)
> > It is likely that NCTUgr is not compatible with your machine/environment.
> 
> Thanks for the reply, I probably know what the problem is.

Hi, I'm sorry to bother. I met the same problem as you. Have you solved it?