limbo018/DREAMPlace #185: Issue reading LEF file

IshraqAtUCF, 2024-06-28:
Hi, 

I was trying to read the NanGate45 lef file by modifying the params.json file. I unfortunately am getting the following error. 
[image]

This is my change for the params.json file

`"lef_input" : {
    "description" : "input LEF file",
    "default" : "~/DREAMPlace/input_lefdef/NangateOpenCellLibrary.macro.mod.lef",
    "required" : "required for LEF/DEF"
    },`

limbo018, 2024-06-29:
It complains about not able to open the path. Can you try full path for ```~/DREAMPlace/input_lefdef/NangateOpenCellLibrary.macro.mod.lef```?

IshraqAtUCF, 2024-06-29:
I tried removing the tilde `~` and that gives me a segmentation error. 
[image]

Strangely if I change this to a different .lef file (fakeram45_256x16.lef) and give ariane136_fp_placed_macros.def as the def input, it succesfully reads the .lef and gives segmentation error for the .def file. 
[image]

FYI, I am running this inside WSL and the docker container inside the WSL. I have 15gb of memory resource available with 3 gb swap. 
[image]

Also, I am trying to run the TILOS benchmark. Do you have any guidelines or documentation to run that benchmark?

limbo018, 2024-06-29:
How many lef files does Nangate45 have? You need to include all of them by putting the technology lef before the macro lef files like following, 
```
"lef_input" : ["benchmarks/ispd2015/mgc_des_perf_1/tech.lef", "benchmarks/ispd2015/mgc_des_perf_1/cells.lef"]
```

IshraqAtUCF, 2024-07-01:
There are 4 files. I did add them like the way you mentioned. But they seem to still give the segmentation error. Do you have any reasoning why this is happening?

IshraqAtUCF, 2024-07-01:
Just to add to this @limbo018, do you have any reference to the TILOS MacroPlacement designs that is working for DREAMPlace?

limbo018, 2024-07-01:
We did try TILOS benchmarks and it looked fine. Let me come back to you about whether any modifications are made.

IshraqAtUCF, 2024-07-01:
I'd highly appreciate if you could provide the files you used for the TILOS benchmark or any direction to a branch that has the implementation @limbo018

Constwelve, 2024-07-02:
I synthesized the netlist via OpenRoad and imported the design into Innovus to regenerate the .lef def file. The files generated this way work for DREAMPplace.

IshraqAtUCF, 2024-07-03:
Hi @Constwelve and @limbo018. I synthesized (Genus) and created .def file (Innovus) for the ASAP7 - Mempool of TILOS MacroPlacement. I gave the following input to the `test/ispd2005/adaptec1.json` file and removed the aux input. 

`
"lef_input" : ["lef/sram_asap7_32x256_1rw.lef", "lef/sram_asap7_64x64_1rw.lef","lef/asap7_tech_1x_201209.lef","lef/asap7sc7p5t_27_R_1x_201211.lef"],
"def_input" : "def/mempool_tile_wrap.def",
`
The .def file is generated through Innovus and the lef files were already there in MacroPlacement.

I am still getting a segmentation error. 

Here is the log file. I appreciate any sort of suggestion. 

[mempool_ASAP7.txt](https://github.com/user-attachments/files/16088136/mempool_ASAP7.txt)

limbo018, 2024-07-03:
@IshraqAtUCF Should the order be as following?
```
"lef_input" : ["lef/asap7_tech_1x_201209.lef", "lef/sram_asap7_32x256_1rw.lef", "lef/sram_asap7_64x64_1rw.lef", "lef/asap7sc7p5t_27_R_1x_201211.lef"]
```

@Constwelve Could you share your Mempool case?

IshraqAtUCF, 2024-07-04:
It gives the same segmentation error even if I change the ordering in the lef files. @limbo018 

And yes, I would appreciate if you could share your files @Constwelve

Constwelve, 2024-07-05:
Here are the files I generated, you can try it out.
https://www.dropbox.com/scl/fi/1av6bjjp82js7vfb4t7xi/TILOS_lefdef.zip?rlkey=ie7hp4nzalooccyb6rss5hnk7&st=cu03rnv5&dl=0

nineight908, 2024-10-09:
> 这是我生成的文件，您可以试用一下。 https://www.dropbox.com/scl/fi/1av6bjjp82js7vfb4t7xi/TILOS_lefdef.zip?rlkey=ie7hp4nzalooccyb6rss5hnk7&st=cu03rnv5&dl=0

Hello professor, can you please share with me the lef files of ariane133 and ariane136 of the NanGate45, I would be grateful for that @Constwelve