limbo018/DREAMPlace #186: iccad 2015 contest benchmark has only simplified tech lef but need full

MyonForCode, 2024-07-25:
Hi Dr.Lin,

  Sorry to bother, but I was trying to carry on the physical design after global placement with DREAMPlace.
  When I set the init_lef_file in Innovus, an error came out saying the tech lef is not correct.
  ICCAD 2015 contest benchmarks is what I'm using, but the [tech.lef.gz](https://www.iccad-contest.org/2015/problem_C/Benchmarks/tech.lef.gz) cannot be found anymore.
  Is there any chance that you could provide the full tech.lef.gz?
  I would appreciate it if you can help me, thank you for all your effort. 

Matt

limbo018, 2024-07-25:
Please check https://github.com/limbo018/DREAMPlace/blob/master/benchmarks/iccad2015.ot.md

This markdown document has the link for the benchmarks.

MyonForCode, 2024-07-25:
Yes, I have downloaded it. But the benchmarks doesn't include the complete tech lef used for future PD / routing research.

[image]

 The contest provided the link I mentioned earlier but cannot be found. So I would like to ask if you still have the complete tech lef provided at that time. 

 If you no longer have these files, please let me know. Thank you for your time on this issue.

MyonForCode, 2024-07-25:
This is the format of the tech lef I need (you provided tech lef like this in ISPD 2015 benchmarks but with no libs). 
[image]

This is the simplified version you provided in ICCAD 2015 benchmarks. 
[image]

limbo018, 2024-07-25:
We do not have the full tech file either. Maybe you can check with the contest chairs to see whether they have a copy.

MyonForCode, 2024-07-25:
Thank you, Dr. Lin. I will contact with them. If I get the full version, I will post the link here. 

Thanks again.