limbo018/DREAMPlace #188: Assert issue

Mathew0553, 2024-09-17:
@limbo018 
hi Professors Limbo:
i meet the below assert problem when i use DREAMPlace 3.0 
[image]

at first i try to replace abacus_legalize file folder with DREAMPlace 4.0's , but it will meet another fimiliar assert error at abacus_legalize_cpu.h:94
and next i try to annotate the assert code , it still show the same assert error

how can i fix it or skip it? thanks for reply!

Mathew0553, 2024-09-17:
> You can disable abacus by commenting out line 735-739 in BasicPlace.py (
> 
> https://github.com/limbo018/DREAMPlace/blob/61d54291b5d11cb45ebeb0cfb6281a247c128cfa/dreamplace/BasicPlace.py#L734
> 
> ), and replacing line 734 with `pos3 = pos2`.
> Remember to go to the build directory and `make install` again.

@limbo018 
i try  follwer your steps,it occuer new import mistake, i think i cannot just replace BasicPlace.py as DREAMPlace 4.0 but the other files use by DREAMPlace 3.0; 
could you please offer the 3.0‘s solutions?
[image]

limbo018, 2024-09-17:
Hey, you can simply checkout 3.0 using git, if you just want that version. All versions are available. 

https://github.com/limbo018/DREAMPlace/tags

Mathew0553, 2024-09-17:
> Hey, you can simply checkout 3.0 using git, if you just want that version. All versions are available.
> 
> https://github.com/limbo018/DREAMPlace/tags

yes,professor,i git the checkout tags/3.0.0 and deploy it , is show the unavoidable assert as you can see in first pic, do you think i  can just simply commenting out the assertmsg to skip it? or i commenting out the fimiliar line you gave in BasicPlace.py in 3.0 version? thanks for patient!
[image]

limbo018, 2024-09-17:
Does you case contain fence regions? If no fence regions, you only need to replace line 629 in your screenshot with ```return pos2```. 

If your case contains fence regions, there is another place that called abacus legalization in this file. You also need to disable that.

Mathew0553, 2024-09-17:
> Does you case contain fence regions? If no fence regions, you only need to replace line 629 in your screenshot with `return pos2`.
> 
> If your case contains fence regions, there is another place that calls abacus legalization in this file. You also need to disable that.

@limbo018 
i just need to get the base result first, dont case the fence regions.
i tried two methods: 1. replace line 629 as return pos2.  2. simly comment out assermsg 
they seemly got the next  same assert, which mean i think the second method is work,  
[image]
maybe i can comment out all the follow assert? it may lead to catastrophic errors or obtain basic results? do you think its work?

limbo018, 2024-09-17:
You didn't get what I mean. 
1. You need to disable the entire abacus legalization instead of just commenting out the assertion part. 
2. If you just want to get basic results, you can even disable detailed placement by setting ```detailed_place_flag``` to 0 in the json configuration file. 
3. There are two places that will invoke abacus legalization in BasicPlace.py (one for cases with fence regions and the other for cases without fence regions). "Comment out" all of them to disable abacus (I assume you can read python code and be able to make minor modifications).

limbo018, 2024-09-21:
Hi, please checkout commit https://github.com/limbo018/DREAMPlace/commit/cd4ecff71cff86400965307a0cf483b791fa0808 . I added a flag ```abacus_legalize_flag``` to control whether turn on/off abacus legalization. I recommend you to pull the latest ```develop``` branch and modify the json file a bit like following: 

```
"global_place_flag" : 1,
"legalize_flag" : 1,
"abacus_legalize_flag" : 0,
"detailed_place_flag" : 0,
```

You are mostly likely to resolve your issues by turning off abacus legalization and detailed placement. If you can share your testcases, we can also take look at why abacus legalization or detailed placement fails.

Mathew0553, 2024-09-23:
> Hi, please checkout commit [cd4ecff](https://github.com/limbo018/DREAMPlace/commit/cd4ecff71cff86400965307a0cf483b791fa0808) . I added a flag `abacus_legalize_flag` to control whether turn on/off abacus legalization. I recommend you to pull the latest `develop` branch and modify the json file a bit like following:
> 
> ```
> "global_place_flag" : 1,
> "legalize_flag" : 1,
> "abacus_legalize_flag" : 0,
> "detailed_place_flag" : 0,
> ```
> 
> You are mostly likely to resolve your issues by turning off abacus legalization and detailed placement. If you can share your testcases, we can also take a look at why abacus legalization or detailed placement fails.

Thanks for your help so much Professor lim, i have resolved my issues by reinstalled DREAMPlace3.0，but i still be grateful for your patient and guidance too.