limbo018/DREAMPlace #189: Can it run without GPU?

mingtianbo, 2024-09-24:
Dear Dr. Limbo: 

Today I tried to use **Docker** to build in a local virtual machine and started **Docker under Linux without GPU**, but the following error was still reported when running the benchmark. How can I run this project without GPU?

Thanks so much!

[image]

limbo018, 2024-09-25:
Please read the README file. You can configure the JSON file by setting ```gpu``` flag to 0. 

[image]

mingtianbo, 2024-09-25:
I apologize for my oversight and truly appreciate your understanding.