limbo018/DREAMPlace #190: error  about "config_cbc_default.h"

JeffreyzzZ0, 2024-10-13:
[image]

Dear Prof.Lin @limbo018 ,  when I try to build DP, I met this error, it seems that the dependency lemon can not be built, I checked the file  "config_cbc_default.h" and it does not exist , and I tried to rebuild it but still failed, 
my System Information:
Operating System: CentOS Linux release 7.9.2009 (Core)
Compiler Version: 11.2.0 (GCC) 
[image]

I expect the compiler to correctly find and use the config_cbc_default.h file so that the project can be compiled successfully.

Thank you for your help!

limbo018, 2024-10-13:
I think cbc is optional for lemon. DREAMPlace does not need cbc either. You do not need to build it.

JeffreyzzZ0, 2024-10-13:
> I think cbc is optional for lemon. DREAMPlace does not need cbc either. You do not need to build it.

It seems that cbc is built-in for lemon and   lemon has no option to avoid building it. When I building DREAMPlace, I just follow the guide and change noting settings.

limbo018, 2024-10-13:
Is it because you have coin/cbc installed in your ```/usr/local``` directory and lemon detects it? You may modify the ```CMakeLists.txt``` in lemon's directory to avoid building cbc.

JeffreyzzZ0, 2024-10-13:
> Is it because you have coin/cbc installed in your `/usr/local` directory and lemon detects it? You may modify the `CMakeLists.txt` in lemon's directory to avoid building cbc.

[image]
and it is the same for clp

limbo018, 2024-10-13:
The problem comes from ```/usr/local/include/coin```, not the DREAMPlace directory. So I guess there is some incompatibility between lemon and the version of ```coin``` you installed.

JeffreyzzZ0, 2024-10-16:
> The problem comes from `/usr/local/include/coin`, not the DREAMPlace directory. So I guess there is some incompatibility between lemon and the version of `coin` you installed.

Thank you, Pro Lin! I avoided building Lemon and instead created a link to the separately installed one, and DREAMPlace successfully built and work!

closed 2024-10-16 by JeffreyzzZ0

c01dwater, 2024-11-04:
> > 问题出在`/usr/local/include/coin`，而不是 DREAMPlace 目录。所以我猜想 lemon 和您安装的 版本之间存在一些不兼容性`coin`。
> 
> 谢谢 Pro Lin！我避免构建 Lemon，而是创建了指向单独安装的 Lemon 的链接，DREAMPlace 成功构建并运行！

您好，我也遇到了同样的问题，能否详细说明解决步骤呢？