limbo018/DREAMPlace #191: About legality check

HongYuLou, 2024-10-15:
Hi Professor Lin,
Sorry to bother you. I noticed that there doesn’t seem to be any mention of power line alignment in the legality check section. Does DREAMPlace currently not support mixed-row-height placement?

limbo018, 2024-10-15:
Hi, you are right. The parser does not extract the power line information of cells and rows, so it cannot check power line alignment. If you know the VDD/GND configurations of rows/cells, you can add this check easily.

HongYuLou, 2024-10-15:
> Hi, you are right. The parser does not extract the power line information of cells and rows, so it cannot check power line alignment. If you know the VDD/GND configurations of rows/cells, you can add this check easily.

Thank you so much for your helpful response! 

I have two follow-up questions:

It seems that the legalization and detailed placement algorithms in DREAMPlace currently do not fully support mixed-row-height designs. In this case, would you recommend switching to a different algorithm to handle mixed-row-height placement effectively?

As far as I can tell, there doesn’t appear to be any open-source code available that simultaneously handles mixed-row-height and fence region constraints in detailed placement(not legalization). Is this correct?

Thanks again for your time and insights!

limbo018, 2024-10-15:
1. Current implementation of greedy legalization in DREAMPlace can handle mixed-row-height cells with fence regions. It just does not consider power line alignment, which is not difficult to satisfy in an algorithmic perspective. You can try and compare it with other algorithms (if available) to see the quality of results. 
2. I haven't done thorough survey on detailed placement. As far as I know, very few work on detailed placement is open-source.

HongYuLou, 2024-10-16:
> 1. Current implementation of greedy legalization in DREAMPlace can handle mixed-row-height cells with fence regions. It just does not consider power line alignment, which is not difficult to satisfy [...]
Thank you so much for your detailed response and the valuable insights! I really appreciate the clarification regarding the mixed-row-height cells and fence regions, and I'll try to incorporate power line alignment into the algorithm . Your guidance has been incredibly helpful. I'll go ahead and close this issue. Thanks again for your time and support!

closed 2024-10-16 by HongYuLou