limbo018/DREAMPlace #193: TIME-DRIVEN

hemingxiao-3, 2024-10-21:
Hello Professor Lin, I would like to know whether the Verilog netlist must be used as input to enable the time-driven mode.
Because I found that my Verilog cannot be parsed correctly. I tried to adjust my Verilog to adapt to the parser, but it still reports an error: some pins or nets cannot be found in the timer. I hope you can answer my doubts.

limbo018, 2024-10-21:
The timing analysis is done by OpenTimer, which requires to take Verilog as input. You can build OpenTimer separately and feed your Verilog to it to see whether your Verilog works.

hemingxiao-3, 2024-10-21:
Thanks, I'll try it.

hemingxiao-3, 2024-10-21:
[ASSERT ] /path-to/DREAMPlace/dreamplace/ops/timing/src/timing_cpp.cpp:164: int DreamPlace::timingCppLauncher(ot::Timer&, const T*, const T*, const std::vector<std::basic_string<char> >&, const std::vector<std::basic_string<char> >&, const int*, const int*, const int*, const T*, const T*, T, T, double, int, int, int) [with T = float]: Assertion `net_iter != timer.nets().end()' failed: could not find net name _07519_ in timer

Aborted (core dumped)
I have been troubled by this kind of problem, is this also because of OpenTimer?

limbo018, 2024-10-21:
Can you check the Verilog file or DEF file to see whether you have a net named ```07519```? If not, it would be hard to debug remotely. Could you share your case with us?

hemingxiao-3, 2024-10-22:
I'm not sure about this, maybe we can do it via email?

limbo018, 2024-10-22:
Sure. You can email me the case.

JeffreyzzZ0, 2024-11-04:
> I'm not sure about this, maybe we can do it via email?

hello, I met the same problem ,have you solve it?

nineight908, 2024-11-05:
Hello Professor Lin, I also encountered this problem, I found the network information that timing.cpp can't find in the def file, it is located in the place where the net information is defined in the def file, how can I solve this kind of problem？

limbo018, 2024-11-05:
It may be difficult to debug OpenTimer, as it is not developed by us. I would suggest you to follow the steps: 
1. Try ICCAD 2015 benchmarks to see if the cases run correctly. 
2. If ICCAD 2015 benchmarks work, then compare the difference between your verilog, liberty, and other input files with those of ICCAD 2015 benchmarks and check what are the differences. 
3. Try removing the different syntax you found to see if it can pass OpenTimer's check. 

With the above procedure, you should be able to locate which syntax or part of the input files cause the problem.