limbo018/DREAMPlace #194: Reading issue with lef for NanGate45

nineight908, 2024-10-22:
Hello professor, when I try to use NanGate45's lef file as input lef, I get a Segmentation fault (core dumped) error.
[image]
Here are my lef files and def files：
[image]
Looking forward to hearing from you, thank you.

limbo018, 2024-10-22:
Can you share your case? You can upload here or email me.

Constwelve, 2024-10-23:
The previous version would crash when processing def pins with ports. This issue has now been resolved in the latest commit on the develop branch.

limbo018, 2024-10-23:
Please check commit https://github.com/limbo018/DREAMPlace/commit/daaa03fe644a80922eedb6842fdd1fef34cdd555

nineight908, 2024-10-24:
@limbo018 @Constwelve Hello professors, thank you for taking time out of your busy schedule to answer my questions, but I still have problems in the experiment, how can I solve them?
[image]

Constwelve, 2024-10-24:
Too many movable macros (123692) caused the macro legalizer to fail to find a legal solution. If you prefer not to change the algorithm, I suggest merging some macros or fixing them manually.

nineight908, 2024-10-29:
I've tried to pin a part of the components in the def file, but I still get the above problem, is there any other solution?