limbo018/DREAMPlace #195: bug while building

Gelisi, 2024-10-22:
cd /home/spcen/project/FSL/EDA/DREAMPlace/build/thirdparty/OpenTimer && /usr/bin/c++ -D_GLIBCXX_USE_CXX11_ABI=0 -I/home/spcen/project/FSL/EDA/DREAMPlace/thirdparty/flute -I/home/spcen/project/FSL/EDA/DREAMPlace/thirdparty/OpenTimer -fPIC -O3 -DNDEBUG -std=c++17 -MD -MT thirdparty/OpenTimer/CMakeFiles/util.dir/unittest/utility.cpp.o -MF CMakeFiles/util.dir/unittest/utility.cpp.o.d -o CMakeFiles/util.dir/unittest/utility.cpp.o -c /home/spcen/project/FSL/EDA/DREAMPlace/thirdparty/OpenTimer/unittest/utility.cpp
In file included from /home/spcen/project/FSL/EDA/DREAMPlace/thirdparty/OpenTimer/unittest/utility.cpp:3:
/home/spcen/project/FSL/EDA/DREAMPlace/thirdparty/OpenTimer/unittest/doctest.h:4008:47: error: size of array ‘altStackMem’ is not an integral constant-expression
 4008 |         static char             altStackMem[4 * SIGSTKSZ];
      |                                               ^
make[2]: *** [thirdparty/OpenTimer/CMakeFiles/util.dir/build.make:79: thirdparty/OpenTimer/CMakeFiles/util.dir/unittest/utility.cpp.o] Error 1
make[2]: Leaving directory '/home/spcen/project/FSL/EDA/DREAMPlace/build'
make[1]: *** [CMakeFiles/Makefile2:3819: thirdparty/OpenTimer/CMakeFiles/util.dir/all] Error 2
make[1]: Leaving directory '/home/spcen/project/FSL/EDA/DREAMPlace/build'
make: *** [Makefile:139: all] Error 2

Please teach me how to fix it. thanks

limbo018, 2024-10-22:
You may check this issue for the solution. 

https://github.com/limbo018/DREAMPlace/issues/133

Gelisi, 2024-10-23:
I have solve this problem, but another one is here.

[100%] Linking CXX executable abacus_unittest
cd /home/spcen/project/FSL/EDA/DREAMPlace/build/unittest/ops/abacus_legalize_unittest && /home/spcen/.local/lib/python3.10/site-packages/cmake/data/bin/cmake -E cmake_link_script CMakeFiles/abacus_unittest.dir/link.txt --verbose=1
/usr/bin/c++ -O3 -DNDEBUG CMakeFiles/abacus_unittest.dir/abacus_unittest.cpp.o -o abacus_unittest  -Wl,-rpath,/usr/local/cuda/lib64:/home/spcen/.local/lib/python3.10/site-packages/torch/lib: ../../../dreamplace/ops/utility/libutility_cxx.a /usr/local/cuda/lib64/libcudart.so /home/spcen/.local/lib/python3.10/site-packages/torch/lib/libtorch.so /home/spcen/.local/lib/python3.10/site-packages/torch/lib/libc10.so /home/spcen/.local/lib/python3.10/site-packages/torch/lib/libtorch_cpu.so /home/spcen/.local/lib/python3.10/site-packages/torch/lib/libc10_cuda.so /home/spcen/.local/lib/python3.10/site-packages/torch/lib/libtorch_cuda.so 
/usr/bin/ld: /usr/local/MATLAB/MATLAB_Runtime/v94/bin/glnxa64/libnccl.so.2: undefined reference to `cudaSetupArgument'
/usr/bin/ld: /home/spcen/.local/lib/python3.10/site-packages/torch/lib/libtorch_cuda.so: undefined reference to `ncclRecv'
/usr/bin/ld: /home/spcen/.local/lib/python3.10/site-packages/torch/lib/libtorch_cuda.so: undefined reference to `ncclCommAbort'
/usr/bin/ld: /home/spcen/.local/lib/python3.10/site-packages/torch/lib/libtorch_cuda.so: undefined reference to `ncclRedOpDestroy'
/usr/bin/ld: /home/spcen/.local/lib/python3.10/site-packages/torch/lib/libtorch_cuda.so: undefined reference to `ncclRedOpCreatePreMulSum'
/usr/bin/ld: /home/spcen/.local/lib/python3.10/site-packages/torch/lib/libtorch_cuda.so: undefined reference to `ncclGetVersion'
/usr/bin/ld: /home/spcen/.local/lib/python3.10/site-packages/torch/lib/libtorch_cuda.so: undefined reference to `ncclSend'
/usr/bin/ld: /home/spcen/.local/lib/python3.10/site-packages/torch/lib/libtorch_cuda.so: undefined reference to `ncclBroadcast'
/usr/bin/ld: /home/spcen/.local/lib/python3.10/site-packages/torch/lib/libtorch_cuda.so: undefined reference to `ncclCommGetAsyncError'
/usr/bin/ld: /home/spcen/.local/lib/python3.10/site-packages/torch/lib/libtorch_cuda.so: undefined reference to `ncclGetLastError'
/usr/bin/ld: /usr/local/MATLAB/MATLAB_Runtime/v94/bin/glnxa64/libnccl.so.2: undefined reference to `cudaLaunch'
collect2: error: ld returned 1 exit status
make[2]: *** [unittest/ops/abacus_legalize_unittest/CMakeFiles/abacus_unittest.dir/build.make:107: unittest/ops/abacus_legalize_unittest/abacus_unittest] Error 1
make[2]: Leaving directory '/home/spcen/project/FSL/EDA/DREAMPlace/build'
make[1]: *** [CMakeFiles/Makefile2:6111: unittest/ops/abacus_legalize_unittest/CMakeFiles/abacus_unittest.dir/all] Error 2
make[1]: Leaving directory '/home/spcen/project/FSL/EDA/DREAMPlace/build'
make: *** [Makefile:139: all] Error 2

near 100%

limbo018, 2024-10-23:
It looks like your libtorch_cuda wrongly links to the libnccl.so under MATLAB rather than cuda. Can you avoid ```/usr/local/MATLAB``` directory to be searched in ```$PATH```?

Gelisi, 2024-10-24:
Morning professor! Sincerely thank you for your help. I have run the demo and I get the placement of the demo adaptec1.log.  (1) Is My dreamplace  4.1.0 ? But the project mentioned the HPWL is much lower than mine. Waiting for your help. It's my first time to do this topic. Sorry to bother you
[adaptec1.log](https://github.com/user-attachments/files/17500048/adaptec1.log)

limbo018, 2024-10-24:
The log looks OK. You can use ```git status``` to see which commit the code is. The tool does not report the version number at runtime. 

Which baseline are you comparing too? If you are comparing with some numbers in published papers, they may use other tools to report HPWL, like NTUplace3 or NTUplace4dr.

Gelisi, 2024-10-24:
[image]
This result in your readme? I just git the code days before. I think it's the newest one?

Gelisi, 2024-10-24:
But the result in log is about [INFO   ] K-reorder time: 888.724 ms
[INFO   ] Legality check takes 48.2037 ms
[INFO   ] DREAMPlace - K-Reorder legal flag = 1
[INFO   ] DREAMPlace - detailed placement takes 4.658 seconds
[INFO   ] DREAMPlace - iteration  611, wHPWL 7.277018E+07, time 0.908ms
[INFO   ] DREAMPlace - non-linear placement takes 17.55 seconds
[INFO   ] DREAMPlace - writing to results/adaptec1/adaptec1.gp.pl
[INFO   ] writing placement to results/adaptec1/adaptec1.gp.pl
[INFO   ] DREAMPlace - write SolutionFileFormat.BOOKSHELF takes 0.327 seconds
[INFO   ] DREAMPlace - placement takes 33.508 seconds

limbo018, 2024-10-24:
You need to download the modified ISPD2005 benchmarks with all fixed macros and IO pads regarded as movable.

Gelisi, 2024-10-24:
[image]
so download in this way is not correct?
How can I download the modified ISPD2005 benchmarks with all fixed macros and IO pads regarded as movable.

limbo018, 2024-10-24:
See https://github.com/limbo018/DREAMPlace/issues/195#issuecomment-2434175165

You need to modify the case a little bit.

Gelisi, 2024-10-24:
where can I modify the benchmarks by removing the FIXED and FIXED_NI keywords…

limbo018, 2024-10-24:
Open the input ```.pl, .nodes``` files with any text editors.

Gelisi, 2024-10-24:
Here, do we need to remove the keyword "fixed" from all files ending with ".pl"? And in the "nodes" section, only the keyword "terminal" is present, so it doesn't need to be touched, right?

Gelisi, 2024-10-24:
[image]

there are many .pl files

Gelisi, 2024-10-24:
[INFO   ] Kernel time: 968.246 ms
[INFO   ] K-reorder time: 1211.18 ms
[INFO   ] DREAMPlace - K-Reorder legal flag = 1
[INFO   ] DREAMPlace - detailed placement takes 11.330 seconds
[INFO   ] Legality check takes 48.5227 ms
[INFO   ] DREAMPlace - iteration  637, wHPWL 1.011426E+08, time 1.338ms
[INFO   ] writing placement to results/adaptec1/plot/iter0638.png
[WARNING] filename = results/adaptec1/plot/iter0638.png, width = 800, height = 799, file format = 3 not used, as DRAWPLACE not enabled
[I] plotting to results/adaptec1/plot/iter0638.png takes 10.976 seconds
[INFO   ] DREAMPlace - plotting to results/adaptec1/plot/iter0638.png takes 10.991 seconds
[INFO   ] DREAMPlace - non-linear placement takes 154.05 seconds
[INFO   ] DREAMPlace - writing to results/adaptec1/adaptec1.gp.pl
[INFO   ] writing placement to results/adaptec1/adaptec1.gp.pl
[INFO   ] DREAMPlace - write SolutionFileFormat.BOOKSHELF takes 0.340 seconds
[INFO   ] DREAMPlace - placement takes 169.947 seconds 

Professor I have get the result, it seems that my result is the same as Dreamplace 4.0

Constwelve, 2024-10-24:
Hi, we have uploaded the modified ISPD2005 benchmark. You can download it from [here](https://www.dropbox.com/scl/fi/01jvzui9hv0aa4krnd8lm/ispd2005free.zip?rlkey=ijwspusl9onncnsu5j4na4tqe&st=l44f3dnw&dl=0).

limbo018, 2024-10-24:
> Hi, we have uploaded the modified ISPD2005 benchmark. You can download it from [here](https://www.dropbox.com/scl/fi/01jvzui9hv0aa4krnd8lm/ispd2005free.zip?rlkey=ijwspusl9onncnsu5j4na4tqe&st=l44f3dnw&dl=0).

@Constwelve could you modify the README and add this link?

Gelisi, 2024-10-24:
But I can't open this link~~~~https://[www.dropbox.com](https://www.dropbox.com/scl/fi/01jvzui9hv0aa4krnd8lm/ispd2005free.zip?rlkey=ijwspusl9onncnsu5j4na4tqe&st=l44f3dnw&dl=0)/scl/fi/01jvzui9hv0aa4krnd8lm/ispd2005free.zip?rlkey=ijwspusl9onncnsu5j4na4tqe&st=l44f3dnw&dl=0

limbo018, 2024-10-24:
You need 科学上网 to access dropbox.

Gelisi, 2024-10-24:
已经挂了laddar了啊

Gelisi, 2024-10-24:
不然git都很卡

Constwelve, 2024-10-24:
Try [this one](https://disk.pku.edu.cn/link/AA3B46DBB859CF41B0AF35984309CB5CB3)

Gelisi, 2024-10-24:
@Constwelve  could I please have your wechat, we want to do a search on this project about the optimizer~~~~

Constwelve, 2024-10-24:
Sure. You can send me your WeChat ID by [email](chenyifan2019@pku.edu.cn).