limbo018/DREAMPlace #198: Get Location of Pins

young03600, 2024-10-27:
Hi,

I use the following files as the input of Placer.py. 
```
DEF/LEF files
Verilog file
```
I would like to know how can I retrieve all the location of pins.

Any help is appreciated.
Thanks.

limbo018, 2024-11-06:
Yes. It's the problem from precision, as floating point does not follow the commutative law in summation.

young03600, 2024-11-06:
Hi professor,

Many thanks for your reply.
I have one more last question.

Could you please help me to check whether my code to retrieve the pin locations is correct?

```python3
import sys
import os
import logging
# for consistency between python2 and python3
root_dir = os.path.dirname(os.path.dirname(os.path.abspath(__file__)))
if root_dir not in sys.path:
    sys.path.append(root_dir)
import numpy as np
import torch
import PlaceDB
import Params
from dreamplace.ops.pin_pos.pin_pos import PinPosFunction, PinPosSegmentFunction
from dreamplace.ops.hpwl.hpwl import HPWLFunction, HPWLAtomicFunction
from dreamplace.ops.pin_pos.pin_pos import PinPos
from dreamplace.ops.hpwl.hpwl import HPWL

if __name__ == '__main__':

    np.set_printoptions(threshold=np.inf, suppress=True)

    params = Params.Params()
    params.load(sys.argv[1])  # only read the DEF/LEF file in this json file
    logging.info("parameters = %s" % (params)) 
    placedb = PlaceDB.PlaceDB()
    placedb(params)

    pin_offset_x = torch.from_numpy(placedb.pin_offset_x).to(torch.float).cuda()
    pin_offset_y = torch.from_numpy(placedb.pin_offset_y).to(torch.float).cuda()
    pin2node_map = torch.from_numpy(placedb.pin2node_map).to(torch.long).cuda()
    flat_node2pin_map = torch.from_numpy(placedb.flat_node2pin_map).to(torch.int).cuda()
    flat_node2pin_start_map = torch.from_numpy(placedb.flat_node2pin_start_map).to(torch.int).cuda()

    pin_pos_instance = PinPos(
        pin_offset_x=pin_offset_x,
        pin_offset_y=pin_offset_y,
        pin2node_map=pin2node_map,
        flat_node2pin_map=flat_node2pin_map,
        flat_node2pin_start_map=flat_node2pin_start_map,
        num_physical_nodes=placedb.num_physical_nodes,
    )
    pin_pos_instance = pin_pos_instance.cuda()
    pos = torch.cat([torch.from_numpy(placedb.node_x).to(torch.float), 
                     torch.from_numpy(placedb.node_y).to(torch.float)]).cuda()
    pin_pos = pin_pos_instance(pos)
    
    # only use for checking the pin locations
    pin_pos = np.array(pin_pos.cpu())
    pin_x = pin_pos[:placedb.num_pins]
    pin_y = pin_pos[placedb.num_pins:]
    pin_pos = np.stack((pin_x, pin_y), axis=1)
```

limbo018, 2024-11-07:
Looks ok to me (roughly), but you may need to run and verify the correctness. The ```PinPos``` operator runs on CPU as well. You do not need to put it on GPU if eventually you want the results on CPU.

young03600, 2024-11-07:
Hi professor,

Thanks for your reply.

I am currently facing an issue with verifying the correctness of the pin locations.

Although I am reading the same provided DEF/LEF files, my retrieved pin locations are different from those provided by the paper's authors. I understand that they used the pin_ops operator in DREAMPlace to obtain their pin locations, whereas my approach is yielding discrepancies.

Additionally, if I would like to calculate the distances from the pin locations to the four die boundaries
```python3
left = placedb.xl()
right = placedb.xh()
bottom = placedb.yl()
top = placedb.yh()

# pin_pos.shpae : [num_pins, 2],  where 2 is x then y
relative_distances = np.array(
    [
        pin_pos[:, 0] - left,       # distance_to_left
        right - pin_pos[:, 0],      # distance_to_right
        pin_pos[:, 1] - bottom,     # distance_to_bottom
        top - pin_pos[:, 1],        # distance_to_top
    ]
).T
```
I'm seeing slight differences only in the distance_to_bottom and distance_to_top (y-axis) values, and I wonder if this is due to a precision issue.
It's my last question. Thank you for reading my long questions.