limbo018/DREAMPlace #200: I meet error: 'const struct DefParser::Pin' has no member named 'vPinPort'

meihailang, 2024-11-04:
DefParser::P in in limbo in the third-party tool really didn't find the property vPinPort

limbo018, 2024-11-04:
You need to run ```git submodule update``` to update the limbo package.

meihailang, 2024-11-04:
Thank you for the reply, this issue is solved, but the following problem arises：

/usr/local/cuda/include/cub/agent/agent_three_way_partition.cuh(87): error: name followed by "::" must be a class or namespace name
  struct accumulator_pack_base_t<OffsetT, typename cuda::std::enable_if<sizeof(OffsetT) == 4&gt;::type&gt;
                                                         ^

/usr/local/cuda/include/cub/agent/agent_three_way_partition.cuh(87): error: expected an identifier
  struct accumulator_pack_base_t<OffsetT, typename cuda::std::enable_if<sizeof(OffsetT) == 4&gt;::type&gt;
                                                                        ^

/usr/local/cuda/include/cub/agent/agent_three_way_partition.cuh(87): error: expected a "&gt;"
  struct accumulator_pack_base_t<OffsetT, typename cuda::std::enable_if<sizeof(OffsetT) == 4&gt;::type&gt;
                                                                        ^

/usr/local/cuda/include/cub/agent/agent_three_way_partition.cuh(87): error: expected a ";"
  struct accumulator_pack_base_t<OffsetT, typename cuda::std::enable_if<sizeof(OffsetT) == 4&gt;::type&gt;
                                                                                                   ^

/usr/local/cuda/include/cub/agent/agent_three_way_partition.cuh(105): error: accumulator_pack_base_t is not a template
  struct accumulator_pack_t : accumulator_pack_base_t<OffsetT&gt;
                              ^

/usr/local/cuda/include/cub/agent/agent_three_way_partition.cuh(107): error: accumulator_pack_base_t is not a template
    using base = accumulator_pack_base_t<OffsetT&gt;;
                 ^

/usr/local/cuda/include/cub/agent/agent_three_way_partition.cuh(108): error: a class or namespace qualified name is required
    using typename base::pack_t;
                   ^

/usr/local/cuda/include/cub/agent/agent_three_way_partition.cuh(108): error: expected a ";"
    using typename base::pack_t;
                       ^

/usr/local/cuda/include/cub/agent/agent_three_way_partition.cuh(110): error: identifier "pack_t" is undefined
              static void subtract(pack_t &amp;packed, OffsetT val)
                                   ^

/usr/local/cuda/include/cub/agent/agent_three_way_partition.cuh(115): error: identifier "pack_t" is undefined
              static OffsetT sum(pack_t &amp;packed)
                                 ^

/usr/local/cuda/include/cub/agent/agent_three_way_partition.cuh(120): error: identifier "pack_t" is undefined
              static pack_t zero()
                     ^

/usr/local/cuda/include/cub/agent/agent_three_way_partition.cuh(112): error: name followed by "::" must be a class or namespace name
      packed = base::pack( base::first(packed) - val, base::second(packed) - val);
               ^

/usr/local/cuda/include/cub/agent/agent_three_way_partition.cuh(112): error: name followed by "::" must be a class or namespace name
      packed = base::pack( base::first(packed) - val, base::second(packed) - val);
                           ^

/usr/local/cuda/include/cub/agent/agent_three_way_partition.cuh(112): error: name followed by "::" must be a class or namespace name
      packed = base::pack( base::first(packed) - val, base::second(packed) - val);
                                                      ^

/usr/local/cuda/include/cub/agent/agent_three_way_partition.cuh(117): error: name followed by "::" must be a class or namespace name
      return base::first(packed) + base::second(packed);
             ^

/usr/local/cuda/include/cub/agent/agent_three_way_partition.cuh(117): error: name followed by "::" must be a class or namespace name
      return base::first(packed) + base::second(packed);
                                   ^

/usr/local/cuda/include/cub/agent/agent_three_way_partition.cuh(177): error: namespace "DreamPlace::cub::CUB_200302_600_610_700_750_800_NS::detail::three_way_partition" has no member "accumulator_pack_t"
    using AccumPackHelperT = detail::three_way_partition::accumulator_pack_t<OffsetT&gt;;
                                                          ^

[... 318 lines cut ...]
                                                      ^

/usr/local/cuda/include/cub/device/dispatch/tuning/tuning_three_way_partition.cuh(153): error: the template argument list of the partial specialization includes a nontype argument whose type depends on a template parameter
  struct sm90_tuning<Input, OffsetT, input_size::_16, offset_size::_4&gt;
         ^

/usr/local/cuda/include/cub/device/dispatch/tuning/tuning_three_way_partition.cuh(158): error: identifier "BlockLoadAlgorithm" is undefined
    static constexpr BlockLoadAlgorithm load_algorithm = BLOCK_LOAD_WARP_TRANSPOSE;
                     ^

/usr/local/cuda/include/cub/device/dispatch/tuning/tuning_three_way_partition.cuh(158): error: identifier "BLOCK_LOAD_WARP_TRANSPOSE" is undefined
    static constexpr BlockLoadAlgorithm load_algorithm = BLOCK_LOAD_WARP_TRANSPOSE;
                                                         ^

Error limit reached.
100 errors detected in the compilation of "/root/WorkSpace/DREAMPlace/dreamplace/ops/pin_pos/src/pin_pos_cuda_segment_kernel.cu".
Compilation terminated.
-- Removing /root/WorkSpace/DREAMPlace/build/dreamplace/ops/pin_pos/CMakeFiles/pin_pos_cuda_segment_cuda_tmp.dir/src/./pin_pos_cuda_segment_cuda_tmp_generated_pin_pos_cuda_segment_kernel.cu.o
/usr/bin/cmake -E rm -f /root/WorkSpace/DREAMPlace/build/dreamplace/ops/pin_pos/CMakeFiles/pin_pos_cuda_segment_cuda_tmp.dir/src/./pin_pos_cuda_segment_cuda_tmp_generated_pin_pos_cuda_segment_kernel.cu.o
CMake Error at pin_pos_cuda_segment_cuda_tmp_generated_pin_pos_cuda_segment_kernel.cu.o.Release.cmake:280 (message):
  Error generating file
  /root/WorkSpace/DREAMPlace/build/dreamplace/ops/pin_pos/CMakeFiles/pin_pos_cuda_segment_cuda_tmp.dir/src/./pin_pos_cuda_segment_cuda_tmp_generated_pin_pos_cuda_segment_kernel.cu.o

梅海浪:-*
***@***.***

limbo018, 2024-11-04:
What's your CUDA version? As far as I know, CUDA 12 or later has severe back compatibility issues.

meihailang, 2024-11-04:
>  My cuda version is 12.4 and I'm trying 11.5, but 11.5 corresponds to 1.11 for pytorch, which is not the version recommended by github docs, and another issue arises
>
>
> 梅海浪:-*

梅海浪:-*
***@***.***

limbo018, 2024-11-06:
My local environment is PyTorch 1.7 with CUDA 11.5. Similar environment should work, but do downgrade CUDA to 11.X.