limbo018/DREAMPlace #204: Core dumpd error when reading lef

young03600, 2024-12-05:
Hi professor,

Sorry to bother you.

I read the LEF provided by synopsys and DEF file dumped by ICCII tool.
But I got core dumped error when reading LEF.

Sorry for the lacked information about the LEF because the file has license.
But I think I can provided some information for references.

Thanks in advance.

young03600, 2024-12-05:
Hi professor,

Thanks for the information.
After run with gdb, the following nearest 10 calls in call stack is showed:

```
Thread 1 "python3" received signal SIGSEGV, Segmentation fault.
0x00007ffff4489964 in std::string::c_str() const () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
(gdb) bt
#0  0x00007ffff4489964 in std::string::c_str() const () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#1  0x00007fff267a530d in DreamPlace::PlaceDB::lef_macro_cbk (this=0x7fffffffbf70, m=...) at /home/user/DREAMPlace/dreamplace/ops/place_io/src/PlaceDB.cpp:126
#2  0x00007fff26a758a5 in LefParser::macroCB (c=LefDefParser::lefrMacroCbkType, macro=0x555557993c00) at /home/user/DREAMPlace/thirdparty/Limbo/limbo/parsers/lef/adapt/LefDriver.cc:318
#3  0x00007fff26acab95 in LefDefParser::lefyyparse () at lef/lef.y:4139
#4  0x00007fff26a96afb in LefDefParser::lefrRead (f=0x5555578eaa60, fName=0x55555798a318 "/home/user/netlists/techlib/saed32nm_rvt_1p9m.lef", uData=0x555557985860)
    at /home/user/DREAMPlace/thirdparty/Limbo/limbo/thirdparty/lefdef/5.8/lef/lef/lefrReader.cpp:323
#5  0x00007fff26a767fa in LefParser::Driver::parse_file (this=0x7fffffffbd60, Python Exception <class 'gdb.error'> No type named class std::basic_string<char, std::char_traits<char>, std::allocator<char> >::_Rep.: 
filename=) at /home/user/DREAMPlace/thirdparty/Limbo/limbo/parsers/lef/adapt/LefDriver.cc:694
#6  0x00007fff26a768c2 in LefParser::read (db=..., Python Exception <class 'gdb.error'> No type named class std::basic_string<char, std::char_traits<char>, std::allocator<char> >::_Rep.: 
lefFile=) at /home/user/DREAMPlace/thirdparty/Limbo/limbo/parsers/lef/adapt/LefDriver.cc:776
#7  0x00007fff267fffc2 in DreamPlace::readLef (db=...) at /home/user/DREAMPlace/dreamplace/ops/place_io/src/PyPlaceDB.cpp:25
#8  0x00007fff26a674a2 in DreamPlace::place_io_forward (args=...) at /home/user/DREAMPlace/dreamplace/ops/place_io/src/place_io.cpp:132
#9  0x00007fff26a714f4 in pybind11::detail::argument_loader<pybind11::list const&>::call_impl<DreamPlace::PlaceDB, DreamPlace::PlaceDB (*&)(pybind11::list const&), 0ul, pybind11::detail::void_type>(DreamPlace::PlaceDB (*&)(pybind11::list const&), std::integer_sequence<unsigned long, 0ul>, pybind11::detail::void_type&&) && (this=0x7fffffffbf50, f=@0x5555579401e8: 0x7fff26a67292 <DreamPlace::place_io_forward(pybind11::list const&)>)
    at /home/user/DREAMPlace/thirdparty/pybind11/include/pybind11/cast.h:1443
#10 0x00007fff26a70405 in pybind11::detail::argument_loader<pybind11::list const&>::call<DreamPlace::PlaceDB, pybind11::detail::void_type, DreamPlace::PlaceDB (*&)(pybind11::list const&)>(DreamPlace::PlaceDB (*&)(pybind11::list const&)) && (this=0x7fffffffbf50, f=@0x5555579401e8: 0x7fff26a67292 <DreamPlace::place_io_forward(pybind11::list const&)>) at /home/user/DREAMPlace/thirdparty/pybind11/include/pybind11/cast.h:1412
```

Are there any further information needed?
Thanks.

limbo018, 2024-12-06:
Did you clean build folder and rerun cmake with ```-DCMAKE_BUILD_TYPE=Debug``` to enable debug mode? In debug mode, you can see which line of code crashes.

limbo018, 2024-12-06:
Can you also check whether the ```SITE``` definition is correct in your lef file and whether it is recognized by the parser in the log? I doubt some ```MACRO``` definitions do not have the core site definition.

limbo018, 2024-12-06:
> Hi professor,
> It's true that the line 127 breaks the program.
> I found that SAED32 library only have cell lef so that's why it breaks.
> Currently I only have the cell lef and .tf file.

I see. You need a tech lef (tlef) to define sites and layers.

limbo018, 2024-12-06:
I added an assertion with error message when core SITE is not found in commit (https://github.com/limbo018/DREAMPlace/commit/c0c416dbbbd1574b92dbd5b593359f67fe884cd1). It will help users to figure out missing input files/information.

young03600, 2024-12-06:
Thanks for your help.
It seems worked after I use ```write_lef``` command in ICC2 to dump tech lef information and merged them in one lef file.
So, just left it as a record for someone like me.
BTW, I'd like to know how the row bbox is computed based on the given ```DIEAREA```?

closed 2024-12-06 by young03600

limbo018, 2024-12-06:
Row bbox is computed based on the union of all ROW definitions in def, which is usually equal to or smaller than diearea.

young03600, 2024-12-10:
Hi professor,

Sorry for one more last question.

If I used the DEF file provided by someone to dump the pin location by pin_pos.py.
But I got the different result of the locations.
Could I assume that the DEF file is different? Or should I do some transformation by self to considering like the direction of macro to get the correct pin locations after using pin_pos operator?

limbo018, 2024-12-10:
I'm a bit confused by your question. You can easily verify whether the pin pos is correct by manually calculating the coordinates in DEF and scaling to DREAMPlace unit. If you confirm pin_pos does not provide the correct results, please provide an example where it triggers the error.

young03600, 2024-12-25:
Sorry professor,
I don't mean any offense with that question.
But I'm not sure how to verify the pin positions myself.
It would be very helpful if you can provide steps about how to compute the pin position from DEF file.

Many thanks for your patience.