limbo018/DREAMPlace #206: Assertion `range.low == range.high' failed: do not support bus yet

c01dwater, 2024-12-10:
Dear Prof. @limbo018. When I use DreamPlace to place a netlist synthesized by DC, the following error is reported: 

[image]

Verilog file：
[image]

I tried “set flatten_array” from DC, but it doesn't work.
How did this error occur?  What can I do to solve this problem?

limbo018, 2024-12-10:
Can you provide a sample case? Currently the database does not support to build bus, but it should not be easy to fix it.

c01dwater, 2024-12-10:
田康林
***@***.***

 林老师您好。
这是我使用yosys综合的网表，您能告诉我是哪里的，哪些语法出现了问题吗？
感谢林老师。

limbo018, 2024-12-11:
Hi, I cannot find any uploaded files. You may need to upload through github website.

c01dwater, 2024-12-11:
[mapped.v.zip](https://github.com/user-attachments/files/18089950/mapped.v.zip)

> Hi, I cannot find any uploaded files. You may need to upload through GitHub website.
Dear Prof. @limbo018. Here is our Verilog file. I appreciate your help.

limbo018, 2024-12-11:
Can you provide the full case with lef, def, and verilog? I cannot run with only verilog.

limbo018, 2024-12-11:
> Dear Prof. @limbo018. When I use DreamPlace to place a netlist synthesized by DC, the following error is reported:
> 
> <img alt="截屏2024-12-10 18 33 42" width="1036" src="https://private-user-images.githubusercontent.com/45511263/394234845-0e5aa89e-2f12-47ae-9d9c-9b3e4e05e654.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzM5MjExMzgsIm5iZiI6MTczMzkyMDgzOCwicGF0aCI6Ii80NTUxMTI2My8zOTQyMzQ4NDUtMGU1YWE4OWUtMmYxMi00N2FlLTlkOWMtOWIzZTRlMDVlNjU0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMTElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjExVDEyNDAzOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWM0MGRlMjQ2ZjlkNWU4ZWU3ZWI1NjU1OGNkMGYzYWM4MmYyNDZkNmUzOThlZGVjMzNlODk4ZTBhM2UyYmNkNzAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.o_iqTTAAmHQXnJJk_A4EJBQGzHaHjgJ06liPMm3W78U">
> Verilog file： ![image](https://private-user-images.githubusercontent.com/45511263/394235064-98ad74e5-cddf-4e8b-b410-b2e1b5ccd9c0.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzM5MjExMzgsIm5iZiI6MTczMzkyMDgzOCwicGF0aCI6Ii80NTUxMTI2My8zOTQyMzUwNjQtOThhZDc0ZTUtY2RkZi00ZThiLWI0MTAtYjJlMWI1Y2NkOWMwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMTElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjExVDEyNDAzOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTRjYWI1OWFjYTY1NGNhMDAwYmQzY2U0ZjY2NGE0NWY2YzU4OWQ1YTE1N2RjMzk0NzZiOTc5MzI5M2VkN2JkMjImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.PreiPdBh4sTRXOj2_dB6XPjS_h6ybOhTRJZNGfqBPHs)
> 
> I tried “set flatten_array” from DC, but it doesn't work. How did this error occur? What can I do to solve this problem?

The verilog file you provided does not contain such bus syntax.

c01dwater, 2024-12-11:
Dear Prof. @limbo018. Do you know if this error is related to the def file? Because I noticed that my def file is a previous version.
[aes128.zip](https://github.com/user-attachments/files/18095854/aes128.zip)

chao1577135517, 2025-04-22:
I have met the same question，the input netlist format maybe not right，how to get the right netlist by innovus