limbo018/DREAMPlace #207: Def parser error

gxydbc, 2024-12-28:
Dear Prof. @limbo018 
I used https://github.com/ABKGroup/RosettaStone/tree/main/bench/ng45/ISPD2006_adaptec1 for placement, but encountered a "Segmentation fault (core dumped)" issue. After investigation, I found that the problem might lie in the `bool Driver::parse_file(const std::string &filename)` function in `/DREAMPlace/thirdparty/Limbo/limbo/parsers/def/adapt/DefDriver.cc`. However, due to its complexity, I hope to get your help. Thank you very much!
[ISPD2006_adaptec1.zip](https://github.com/user-attachments/files/18265478/ISPD2006_adaptec1.zip)

[image]

limbo018, 2024-12-28:
Could you share the case?

gxydbc, 2024-12-28:
> Could you share the case?

Dear Prof @limbo018 , I've uploaded a compressed file. Can you download it? Or I can email you. You can also find the case information here:https://github.com/ABKGroup/RosettaStone/tree/main/bench/ng45/ISPD2006_adaptec1. Thanks!
[ISPD2006_adaptec1.zip](https://github.com/user-attachments/files/18265778/ISPD2006_adaptec1.zip)

limbo018, 2024-12-28:
You only uploaded macro lef. I think you missed to give tech lef before macro lef.

gxydbc, 2024-12-28:
> You only uploaded macro lef. I think you missed to give tech lef before macro lef.

I have added the tech LEF-related files and can now run them successfully. Thank you very much!