limbo018/DREAMPlace #208: Compilation error

wenh06, 2024-12-31:
Branch: master
Hash: [ec18f9e](https://github.com/limbo018/DREAMPlace/commit/ec18f9eb234e06b207588f088bfe0ad4afa034f4)
Build environment: docker + GitHub action

Dockerfile:
```docker
FROM pytorch/pytorch:1.7.1-cuda11.0-cudnn8-devel
# FROM pytorch/pytorch:2.0.0-cuda11.7-cudnn8-runtime

ARG PYPI_TUNA=1

LABEL maintainer="Yibo Lin <yibolin@pku.edu.cn>"

# Rotates to the keys used by NVIDIA as of 27-APR-2022.
RUN rm /etc/apt/sources.list.d/cuda.list 2>/dev/null || true
RUN rm /etc/apt/sources.list.d/nvidia-ml.list 2>/dev/null || true

RUN apt-get update && apt-get install -y gnupg

RUN apt-key del 7fa2af80
RUN apt-key adv --fetch-keys https://developer.download.nvidia.com/compute/cuda/repos/ubuntu1804/x86_64/3bf863cc.pub
RUN apt-key adv --fetch-keys https://developer.download.nvidia.com/compute/machine-learning/repos/ubuntu1804/x86_64/7fa2af80.pub

# Installs system dependencies.
RUN apt-get update \
        && apt-get install -y \
            flex \
            libcairo2-dev \
            libboost-all-dev 

# Installs system dependencies from conda.
RUN conda install -y -c conda-forge bison

# Installs cmake.
ADD https://cmake.org/files/v3.21/cmake-3.21.0-linux-x86_64.sh /cmake-3.21.0-linux-x86_64.sh
RUN mkdir /opt/cmake \
        && sh /cmake-3.21.0-linux-x86_64.sh --prefix=/opt/cmake --skip-license \
        && ln -s /opt/cmake/bin/cmake /usr/local/bin/cmake \
        && cmake --version

# Installs python dependencies.
RUN if [ "$PYPI_TUNA" = "1" ]; then \
        pip config set global.index-url https://mirrors.tuna.tsinghua.edu.cn/pypi/web/simple; \
        fi
RUN pip install \
        pyunpack>=0.1.2 \
        patool>=1.12 \
        matplotlib>=2.2.2 \
        cairocffi>=0.9.0 \
        pkgconfig>=1.4.0 \
        setuptools>=39.1.0 \
        scipy>=1.1.0 \
        numpy>=1.15.4 \
        shapely>=1.7.0

# Installs dreamplace.
RUN mkdir -p /DREAMPlace
WORKDIR /DREAMPlace
COPY . /DREAMPlace
RUN mkdir -p build install
WORKDIR /DREAMPlace/build
RUN cmake .. -DCMAKE_INSTALL_PREFIX=../install -DPython_EXECUTABLE=$(which python)
RUN make
RUN make install
# RUN ln -s /DREAMPlace/build/install/bin/dreamplace /usr/local/bin/dreamplace

WORKDIR /DREAMPlace
```

GitHub action config:
[... 94 lines cut ...]
#26 515.6          for (std::size_t j = 0; j < pport.vLayer.size(); ++j) {
#26 515.6                                      ^~~~~
#26 515.6                                      qsort
#26 515.6 /DREAMPlace/dreamplace/ops/place_io/src/PlaceDB.cpp:557:17: error: 'const struct DefParser::Pin' has no member named 'vPinPort'
#26 515.6    } else if (!p.vPinPort.empty()) {
#26 515.6                  ^~~~~~~~
#26 515.6 /DREAMPlace/dreamplace/ops/place_io/src/PlaceDB.cpp:558:35: error: 'const struct DefParser::Pin' has no member named 'vPinPort'
#26 515.6      for (std::size_t i = 0; i < p.vPinPort.size(); ++i) {
#26 515.6                                    ^~~~~~~~
#26 515.6 /DREAMPlace/dreamplace/ops/place_io/src/PlaceDB.cpp:562:18: error: 'PinPort' is not a member of 'DefParser'
#26 515.6        DefParser::PinPort const& pport = p.vPinPort[i];
#26 515.6                   ^~~~~~~
#26 515.6 /DREAMPlace/dreamplace/ops/place_io/src/PlaceDB.cpp:563:35: error: 'pport' was not declared in this scope
#26 515.6        for (std::size_t j = 0; j < pport.vLayer.size(); ++j) {
#26 515.6                                    ^~~~~
#26 515.6 /DREAMPlace/dreamplace/ops/place_io/src/PlaceDB.cpp:563:35: note: suggested alternative: 'qsort'
#26 515.6        for (std::size_t j = 0; j < pport.vLayer.size(); ++j) {
#26 515.6                                    ^~~~~
#26 515.6                                    qsort
#26 516.9 make[2]: *** [dreamplace/ops/place_io/CMakeFiles/place_io_cpp.dir/src/PlaceDB.cpp.o] Error 1
#26 516.9 dreamplace/ops/place_io/CMakeFiles/place_io_cpp.dir/build.make:190: recipe for target 'dreamplace/ops/place_io/CMakeFiles/place_io_cpp.dir/src/PlaceDB.cpp.o' failed
#26 516.9 make[2]: Leaving directory '/DREAMPlace/build'
#26 516.9 CMakeFiles/Makefile2:3863: recipe for target 'dreamplace/ops/place_io/CMakeFiles/place_io_cpp.dir/all' failed
#26 516.9 make[1]: *** [dreamplace/ops/place_io/CMakeFiles/place_io_cpp.dir/all] Error 2
```

limbo018, 2024-12-31:
You need update the submodules like Limbo.

wenh06, 2024-12-31:
> You need to update the submodules like Limbo.

Yes, the submodules are up-to-date. I set
```yml
    - uses: actions/checkout@v4
      with:
        submodules: recursive
```
in the action file.

limbo018, 2024-12-31:
Can you run ```git status``` to check the status of submodules? Sometimes ```git submodule update``` may fail due to unstable Internet connection.

wenh06, 2024-12-31:
The following is the output of fetching submodules of the `checkout` action:

```text
Fetching submodules
  /usr/bin/git submodule sync --recursive
  /usr/bin/git -c protocol.version=2 submodule update --init --force --depth=1 --recursive
  Submodule 'thirdparty/Limbo' (https://github.com/limbo018/Limbo.git) registered for path 'thirdparty/Limbo'
  Submodule 'thirdparty/OpenTimer' (https://github.com/enzoleo/OpenTimer.git) registered for path 'thirdparty/OpenTimer'
  Submodule 'thirdparty/cub' (https://github.com/NVlabs/cub.git) registered for path 'thirdparty/cub'
  Submodule 'thirdparty/munkres-cpp' (https://github.com/saebyn/munkres-cpp.git) registered for path 'thirdparty/munkres-cpp'
  Submodule 'thirdparty/pybind11' (https://github.com/pybind/pybind11.git) registered for path 'thirdparty/pybind11'
  Cloning into '/home/runner/work/dreamplace-docker/dreamplace-docker/thirdparty/Limbo'...
  Cloning into '/home/runner/work/dreamplace-docker/dreamplace-docker/thirdparty/OpenTimer'...
  Cloning into '/home/runner/work/dreamplace-docker/dreamplace-docker/thirdparty/cub'...
  Cloning into '/home/runner/work/dreamplace-docker/dreamplace-docker/thirdparty/munkres-cpp'...
  Cloning into '/home/runner/work/dreamplace-docker/dreamplace-docker/thirdparty/pybind11'...
  Submodule path 'thirdparty/Limbo': checked out 'f66d910cc1ce589185bad3da1d26a10710036302'
  Submodule 'limbo/thirdparty/OpenBLAS' (https://github.com/xianyi/OpenBLAS.git) registered for path 'thirdparty/Limbo/limbo/thirdparty/OpenBLAS'
  Cloning into '/home/runner/work/dreamplace-docker/dreamplace-docker/thirdparty/Limbo/limbo/thirdparty/OpenBLAS'...
  From https://github.com/xianyi/OpenBLAS
   * branch              15cb124012c74e9b1b2a180[69](https://github.com/wenh06/dreamplace-docker/actions/runs/12560479197/job/35017870711#step:2:78)9b2f008b7b99e0c -> FETCH_HEAD
  Submodule path 'thirdparty/Limbo/limbo/thirdparty/OpenBLAS': checked out '15cb124012c74e9b1b2a180699b2f008b7b99e0c'
  Submodule path 'thirdparty/OpenTimer': checked out '10dfd93668c1ed2511cc24daa6b460a1683255fe'
  Submodule path 'thirdparty/cub': checked out '39[73](https://github.com/wenh06/dreamplace-docker/actions/runs/12560479197/job/35017870711#step:2:82)57e9aca61f244b0aed2e77efca25f8a95d74'
  Submodule path 'thirdparty/munkres-cpp': checked out '61086fcf5b3a8ad4bda578cdc2d1dca57b548786'
  Submodule path 'thirdparty/pybind11': checked out 'c5ed9d4b193e0c1563282aebcfad57fecb0b4969'
  /usr/bin/git submodule foreach --recursive git config --local gc.auto 0
  Entering 'thirdparty/Limbo'
  Entering 'thirdparty/Limbo/limbo/thirdparty/OpenBLAS'
  Entering 'thirdparty/OpenTimer'
  Entering 'thirdparty/cub'
  Entering 'thirdparty/munkres-cpp'
  Entering 'thirdparty/pybind11'
```

I add some checking in the Dockerfile:

```docker
RUN git status; \
    git submodule foreach git status; \
    git submodule foreach git checkout master; \
    git submodule foreach git fetch; \
    git submodule foreach git merge origin/master; \
    git submodule foreach git status; \
    git status
```

its output is:

```text
#28 [22/24] RUN git status;     git submodule foreach git status;     git submodule foreach git checkout master;     git submodule foreach git fetch;     git submodule foreach git merge origin/master;     git submodule foreach git status;     git status
#28 1.203 On branch docker-ci
#28 1.203 Your branch is up to date with 'origin/docker-ci'.
#28 1.203 
#28 1.203 nothing to commit, working tree clean
#28 1.239 Entering 'thirdparty/Limbo'
#28 1.300 HEAD detached at f66d910
#28 1.300 nothing to commit, working tree clean
#28 1.308 Entering 'thirdparty/OpenTimer'
#28 1.318 HEAD detached at 10dfd93
#28 1.318 nothing to commit, working tree clean
#28 1.325 Entering 'thirdparty/cub'
#28 1.334 HEAD detached at 397357e
#28 1.334 nothing to commit, working tree clean
#28 1.340 Entering 'thirdparty/munkres-cpp'
#28 1.347 HEAD detached at 61086fc
#28 1.347 nothing to commit, working tree clean
#28 1.354 Entering 'thirdparty/pybind11'
#28 1.362 HEAD detached at c5ed9d4
#28 1.362 nothing to commit, working tree clean
[... 23 lines cut ...]
#28 2.383 Your branch is up to date with 'origin/master'.
#28 2.383 
#28 2.383 nothing to commit, working tree clean
#28 2.391 Entering 'thirdparty/OpenTimer'
#28 2.400 On branch master
#28 2.400 Your branch is up to date with 'origin/master'.
#28 2.400 
#28 2.400 nothing to commit, working tree clean
#28 2.407 Entering 'thirdparty/cub'
#28 2.415 HEAD detached at 397357e
#28 2.415 nothing to commit, working tree clean
#28 2.422 Entering 'thirdparty/munkres-cpp'
#28 2.429 HEAD detached at 61086fc
#28 2.429 nothing to commit, working tree clean
#28 2.436 Entering 'thirdparty/pybind11'
#28 2.444 HEAD detached at c5ed9d4
#28 2.444 nothing to commit, working tree clean
#28 2.526 On branch docker-ci
#28 2.526 Your branch is up to date with 'origin/docker-ci'.
#28 2.526 
#28 2.526 nothing to commit, working tree clean
#28 DONE 2.6s
```

I still get the error `'const struct DefParser::Pin' has no member named 'vPinPort'`

limbo018, 2025-01-01:
I found your commit ID of Limbo is not consistent to DREAMPlace. If you check the following link, you can see [Limbo @ 4cf5d2c](https://github.com/limbo018/Limbo/tree/4cf5d2cd407570b0c39c6fb766b0740423b864bf) instead of f66d910cc1ce589185bad3da1d26a10710036302. 

https://github.com/limbo018/DREAMPlace/tree/master/thirdparty

In general, you do not need to manually pull submodules. Using ```git submodule --init; git submodule update``` should work. It seems the Limbo library under your DREAMPlace repo is not the correct version.

limbo018, 2025-01-01:
The latest commit (https://github.com/limbo018/DREAMPlace/commit/70c814b944020e1ea11a3e980d9427dcf6fc7a24) has updated to track the latest version of Limbo (https://github.com/limbo018/Limbo/commit/6c115bfb1d4dca3a92044d505ee00456aac2bd45). It should fix your issue, if you pull the latest version of DREAMPlace.

wenh06, 2025-01-01:
> The latest commit ([70c814b](https://github.com/limbo018/DREAMPlace/commit/70c814b944020e1ea11a3e980d9427dcf6fc7a24)) has updated to track the latest version of Limbo ([limbo018/Limbo@6c115bf](https://github.com/limbo018/Limbo/commit/6c115bfb1d4dca3a92044d505ee00456aac2bd45)). It should fix your issue, if you pull the latest version of DREAMPlace.

Thank you. This fixes the initial error, and the `make` command succeeds. But the `make install` command failed:

```text
CMake Error at dreamplace/ops/gift_init/cmake_install.cmake:73 (file):
  file INSTALL cannot find "/DREAMPlace/dreamplace/ops/gift_init/utils": No
  such file or directory.
```

I think [this line](https://github.com/limbo018/DREAMPlace/blob/70c814b944020e1ea11a3e980d9427dcf6fc7a24/dreamplace/ops/gift_init/CMakeLists.txt#L26C21-L26C45) causes this error

```
install(DIRECTORY ${CMAKE_CURRENT_SOURCE_DIR}/utils DESTINATION dreamplace/ops/${OP_NAME})
```

We do not have a folder named `utils`. I found that `gift_init` was newly added in version 4.2.0 not long ago.

limbo018, 2025-01-01:
This line can be removed. I have removed it in the latest commit (https://github.com/limbo018/DREAMPlace/commit/c3fe55aaa7fbefc5a0583953ac8f7ebd258bb8b4).

wenh06, 2025-01-01:
All issues resolved. Thank you!

closed 2025-01-01 by wenh06