limbo018/DREAMPlace #209: Pin rotations from DEF/LEF

lelzeiny, 2025-01-06:
Hello!

I'm adding to the DREAMPlace repo, but noticed that the pins in rawdb or pydb object in PlaceDB.py has "unrotated pins" from [PyPlaceDB::convertOrient()](https://github.com/limbo018/DREAMPlace/blob/523e4c3e1da6f51edc734b10f5542d5969450cfd/dreamplace/ops/place_io/src/PyPlaceDB.cpp#L685), but rotated sizes. Why is that?

For example: if a LEF has a pin at (0, 3) when the component is sized 5 x 3 and the DEF defines the orientation as "placed west", I would expect the db report the pin to be at (0, 0). Instead, I've seen that its been rotated east to (3, 5).

limbo018, 2025-01-07:
Hi, I would like to confirm with you about the rotations. Could you check the following orientations correct? 

[image]

lelzeiny, 2025-01-07:
Yes, that is what I expect too but I see this is not the case. I have linked a DEF/LEF of a simpler example. 
[simple00_def.txt](https://github.com/user-attachments/files/18329311/simple00_def.txt)
[simple00_lef.txt](https://github.com/user-attachments/files/18329312/simple00_lef.txt)

Looking at MAS2845 P2, the LEF component is 7 x 3 and the pin is at (5.5, 0) I would expect the pin to be at (3, 5.5) when rotated west. But the database offset reports (0, 1.5) which is consistent with it being rotated east. I think this problem originates here in [PyPlaceDB::convertOrient()](https://github.com/limbo018/DREAMPlace/blob/523e4c3e1da6f51edc734b10f5542d5969450cfd/dreamplace/ops/place_io/src/PyPlaceDB.cpp#L685) because the dest orientation is "N" and the src is the DEF orientation.

I have also run with these flags to maintain my current orientations:

    "global_place_flag": 0,
    "legalize_flag": 0,

limbo018, 2025-01-07:
I figured out the problem. The cases 90 and 270 in ```getRotatedPinOffsets``` should be swapped. Fixed in https://github.com/limbo018/DREAMPlace/commit/44ca569ec9a54fc78d65aa0899c92825fbd201c5

lelzeiny, 2025-01-08:
I don't think that resolves the bug, and it also doesn't address inconsistencies with "S". Although I mentioned it earlier, I also think it originates earlier than convertOrient(), because I see the pin offsets before that function is called isn't directly from the LEF like I expected. I'll take a look at it later today though.

limbo018, 2025-01-09:
I verified the example you gave in simple00. It looks correct. Here is the log after I printed the pin offsets before and after rotation. 

```
[INFO   ] -- Converting Node Orientation --
node a1
[DEBUG  ] original: node 0, orient W, sizes 701x384, pin 0 (559, 0), pin 1 (701, 384), 
[DEBUG  ] src_degree_flip = (90, 0), rot_degree = 270, flip = 0
node a1
[DEBUG  ] after rotation: node 0, orient N, sizes 384x701, pin 0 (384, 559), pin 1 (0, 701), 
node a2
[DEBUG  ] original: node 1, orient S, sizes 101x80, pin 0 (21, -21), 
[DEBUG  ] src_degree_flip = (180, 0), rot_degree = 180, flip = 0
node a2
[DEBUG  ] after rotation: node 1, orient N, sizes 101x80, pin 0 (80, 101), 
```

lelzeiny, 2025-01-09:
Thats my mistake. You are right, west and east are resolved.

But when I change the orientation to "S" and print the original / after rotation, my original changes from the LEF to be (142, 384) and the after rotation is the original (559, 0). Why would the original change? Same thing with a2 in your snippet.

limbo018, 2025-01-09:
To clarify: the ```original``` in the log means before rotation, where all cells have orientation "N". 

I cannot catch your new question. Could you give an example?

lelzeiny, 2025-01-09:
If you look at your output for a2 for example, the LEF states the pin is at (80, 101) meaning the rotated pin is at (21, -21). Your output shows the opposite (21, -21) -> (80, 101). The rotation is correct, but the original point was already rotated.

I was trying to point out in the prev comment the same thing happens to P0 when I change node a1.

limbo018, 2025-01-09:
The target of function ```convertOrient()``` is actually trying to regard a rotated cell as "N" equivalent. In the following example, which corresponds to ```a2```, if you want to regard "S" as "N", then the pin is at (80, 101). 

[image]

lelzeiny, 2025-01-09:
But the LEF places the pin at (80, 101) originally

```
MACRO MAS2822
	CLASS BLOCK ;
	SIZE 1.01 BY 0.8 ;
	ORIGIN 0 0 ;
	SYMMETRY X ;
	SITE core 0 0 N DO 1 BY 1 STEP 0 0 ;
	PIN P1
	DIRECTION INOUT ;
	USE SIGNAL ;
		PORT
			LAYER metal1 ;
				RECT 0.79495 1.006 0.80505 1.014 ;
		END
	END P1
```

lelzeiny, 2025-01-09:
I'll also be on this zoom for the next ~1hr if you have time: https://berkeley.zoom.us/j/94164689207?pwd=KKT07dHlKlaPu8tlCmvjcka58BNzs5.1

limbo018, 2025-01-09:
I see your point. Will take a detailed look later. I have a meeting right now. Will come back to you.

lelzeiny, 2025-01-09:
Looks like PlaceDB.cpp rotated the Pin objects and changes the offset in the **Pin.h object** in [PlaceDB::updateNodePinOffset()](https://github.com/limbo018/DREAMPlace/blob/523e4c3e1da6f51edc734b10f5542d5969450cfd/dreamplace/ops/place_io/src/PlaceDB.cpp#L1569) so the pin offsets are already rotated when accessed by PyPlaceDB.cpp.

limbo018, 2025-01-09:
You are right. Commenting line 1752-1755 in PlaceDB.cpp should resolve the issue. (https://github.com/limbo018/DREAMPlace/blob/523e4c3e1da6f51edc734b10f5542d5969450cfd/dreamplace/ops/place_io/src/PlaceDB.cpp#L1752)

The pins were rotated twice.

limbo018, 2025-01-10:
This commit (https://github.com/limbo018/DREAMPlace/commit/132474b571bc1073aa3a9a86f30d500456a232f0) comments out line 1752-1755 in PlaceDB.cpp. It should resolve the issue.

closed 2025-01-15 by lelzeiny