limbo018/DREAMPlace #210: Assertion `xl1 < xl2 && xh1 < xh2' failed: node 8 (nan, nan, nan, nan) overlaps with node 40 (nan, nan, nan, nan)

Kash6, 2025-01-10:
I'm running the benchmarks from the ispd 2015 dataset usin the command:
 python dreamplace/Placer.py ../test/ispd2015/lefdef/mgc_superblue11_a.json

And I get the following error right after the legality check:
/home/eleven/projects/DREAMPlace/dreamplace/ops/abacus_legalize/../abacus_legalize/src/abacus_legalize_cpu.h:96: void DreamPlace::sortNodesInRow(const T*, const T*, const T*, const T*, int, std::vector<int, std::allocator<int> >&) [with T = float]: Assertion `xl1 < xl2 && xh1 < xh2' failed: node 8 (nan, nan, nan, nan) overlaps with node 40 (nan, nan, nan, nan)
Aborted (core dumped)

I haven't modeified anything and I'm running it on torch 1.10.2+cu113.

limbo018, 2025-01-11:
It works fine on my side. Here's the log. 
```
[INFO   ] Greedy legalization takes 2975.9 ms
[INFO   ] Legality check takes 81.7796 ms
[DEBUG  ] average displace = 10.6866
[INFO   ] Abacus legalization takes 28.9314 ms
[DEBUG  ] average displace = 8.22345
[INFO   ] Abacus legalization takes 27.9994 ms
[DEBUG  ] average displace = 41.4481
[INFO   ] Abacus legalization takes 29.2102 ms
[DEBUG  ] average displace = 10.9823
[INFO   ] Abacus legalization takes 27.8204 ms
[DEBUG  ] average displace = 7.63178
[INFO   ] Abacus legalization takes 171.506 ms
[INFO   ] DREAMPlace - legalization takes 5.713 seconds
[INFO   ] DREAMPlace - iteration  961, wHPWL 3.606356E+08, time 0.355ms
```

I saw ```nan``` in your log. What is the earliest place it appears? Could you check your GPU card? Old GPU may not be supported. You can also try running on CPU to see whether it works.

Kash6, 2025-01-13:
[INFO   ] DREAMPlace - nesterov initialization takes 0.151899 seconds
[INFO   ] DREAMPlace - density_weight = [NAN, NAN, NAN, NAN, NAN]
[INFO   ] DREAMPlace - optimizer step 127.450 ms
[INFO   ] DREAMPlace - iteration    0, (   0,  0,  0), Obj NAN, Density [0.000E+00, 0.000E+00, 0.000E+00, 0.000E+00, 0.000E+00], DensityWeight [0.000E+00, 0.000E+00, 0.000E+00, 0.000E+00, 0.000E+00], wHPWL 0.000000E+00, Overflow [0.000E+00, 0.000E+00, 0.000E+00, 0.000E+00, 0.000E+00], Global Overflow 0.000000E+00, MaxDensity [0.000E+00, 0.000E+00, 0.000E+00, 0.000E+00, 0.000E+00], gamma 4.312266E+03, time 4.193ms
[INFO   ] DREAMPlace - full step 162.298 ms
[INFO   ] DREAMPlace - optimizer step 96.291 ms
[INFO   ] DREAMPlace - iteration    1, (   1,  0,  0), Obj NAN, Density [0.000E+00, 0.000E+00, 0.000E+00, 0.000E+00, 0.000E+00], DensityWeight [NAN, NAN, NAN, NAN, NAN], wHPWL 0.000000E+00, Overflow [0.000E+00, 0.000E+00, 0.000E+00, 0.000E+00, 0.000E+00], Global Overflow 0.000000E+00, MaxDensity [0.000E+00, 0.000E+00, 0.000E+00, 0.000E+00, 0.000E+00], gamma 2.585135E+01, time 3.657ms

This is the first place it appears. I've checked the json file and it has density_weight set to 8e-5 by default. 
[image]

Im using the Nvidia a30 shown above. I tried on CPU and it works with the same json config. Any idea why?

enzoleo, 2025-01-14:
It happens very often when your CUDA toolkit version doesn't match the runtime library supported by your driver. I see you have CUDA Version 12.3 from the output of `nvidia-smi`, so you should probably first check whether `nvcc --version` gives a matched version.

Kash6, 2025-01-14:
I think I have the right toolkit installed:
[image]
Is the torch version(1.10.2+cu113) im using causing the problem? If so, I tried searching for cuda 12.3 versions but couldn't find any.

enzoleo, 2025-01-14:
Not sure whether it caused the problem. Maybe you can have a test on torch versions with `cu121` or `cu124` flag.

Kash6, 2025-01-20:
I had to rebuild it from scratch using torch 2.1.0+cu121 and it worked, thanks

closed 2025-01-20 by Kash6